isofit/isofit #28: Question on error running libRadTran test

serbinsh, 2019-03-15:
```
root@41d87ec8e363:/isofit/examples/20171108_Pasadena# ./run_example_libradtran.sh
Traceback (most recent call last):
  File "../../utils/surfmodel.py", line 30, in <module>
    from common import find_header, expand_path, json_load_ascii
ImportError: cannot import name 'find_header'
Traceback (most recent call last):
  File "../../isofit/isofit.py", line 126, in <module>
    main()
  File "../../isofit/isofit.py", line 57, in main
    fm = ForwardModel(config['forward_model'])
  File "/isofit/isofit/forward.py", line 64, in __init__
    self.RT = getattr(import_module(module), cname)(config[key])
  File "/isofit/isofit/rt_libradtran.py", line 54, in __init__
    self.build_lut()
  File "/isofit/isofit/rt_lut.py", line 124, in build_lut
    cmd = self.rebuild_cmd(point, fn)
  File "/isofit/isofit/rt_libradtran.py", line 137, in rebuild_cmd
    with open(infilepath0, 'w') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000_alb0.inp'
```

root@41d87ec8e363:/isofit# ${LIBRADTRAN_DIR}
bash: /libRadtran-2.0.2/: Is a directory

root@41d87ec8e363:/isofit# $ISOFIT_BASE
bash: /isofit: Is a directory

Any ideas?  Was trying to walk through the code to debug but so far can't tell exactly what the issue is.  Is it perhaps actually related to the import error or missing file or both? 

I pulled down the "Data for the REPTRAN absorption parameterization. Updated December 24, 2017! Please untar these data in the libRadtran directory. Parameterization data for bands as well as for satellite channels is included."

```
root@41d87ec8e363:/libRadtran-2.0.2/data# ls -l
total 56
-rw-r--r--  1 isofit root   1860 Mar 15 12:28 Makefile
-rw-rw-rw-  1 isofit  11074 1884 Dec 18  2017 Makefile.in
drwxrwxrwx  4 isofit  11074 4096 Mar 15 12:28 aerosol
drwxrwxrwx  4 isofit  11074 4096 Mar 15 12:28 albedo
drwxrwxrwx  2 isofit  11074 4096 Mar 15 12:28 altitude
drwxrwxrwx  2 isofit  11074 4096 Mar 15 12:28 atmmod
drwxrwxrwx  3 isofit  11074 4096 Mar 15 12:28 cloudprp
drwxr-xr-x  9 isofit isofit 4096 Dec 20  2017 correlated_k
drwxrwxrwx  2 isofit  11074 4096 Mar 15 12:28 crs
drwxrwxrwx 17 isofit  11074 4096 Mar 15 12:28 filter
drwxrwxrwx 10 isofit  11074 4096 Mar 15 12:28 ic
drwxrwxrwx  2 isofit  11074 4096 Mar 15 12:28 scripts
drwxrwxrwx  2 isofit  11074 4096 Mar 15 12:28 solar_flux
drwxrwxrwx  4 isofit  11074 4096 Mar 15 12:28 wc
```

serbinsh, 2019-03-15:
No LUT folder in there?

serbinsh, 2019-03-15:
```
root@41d87ec8e363:/isofit/examples/20171108_Pasadena# python3 ../../utils/surfmodel.py configs/ang20171108t184227_surface.json
Traceback (most recent call last):
  File "../../utils/surfmodel.py", line 30, in <module>
    from common import find_header, expand_path, json_load_ascii
ImportError: cannot import name 'find_header'
```

serbinsh, 2019-03-15:
I see common is in this package here: https://github.com/isofit/isofit/blob/master/isofit/common.py

serbinsh, 2019-03-15:
I went back again and tried to make sure paths were correct in the Docker

    ## Set Python Path and other environment variables
ENV PYTHONPATH="${PYTHONPATH}:/isofit/isofit:/isofit/utils"
ENV ISOFIT_BASE=/isofit
ENV LIBRADTRAN_DIR=/libRadtran-${LIBRADTRAN}

Now I am getting this error 

```
saving results to /isofit/examples/20171108_Pasadena/configs/../remote/ang20170228_surface_model.mat
Traceback (most recent call last):
  File "../../isofit/isofit.py", line 126, in <module>
    main()
  File "../../isofit/isofit.py", line 57, in main
    fm = ForwardModel(config['forward_model'])
  File "/isofit/isofit/forward.py", line 64, in __init__
    self.RT = getattr(import_module(module), cname)(config[key])
  File "/isofit/isofit/rt_libradtran.py", line 54, in __init__
    self.build_lut()
  File "/isofit/isofit/rt_lut.py", line 124, in build_lut
    cmd = self.rebuild_cmd(point, fn)
  File "/isofit/isofit/rt_libradtran.py", line 137, in rebuild_cmd
    with open(infilepath0, 'w') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000_alb0.inp'
``` so made it farther this time

serbinsh, 2019-03-15:
Otherwise close to running in docker:

```
LEE-158844:isofit_master shawnserbin$ docker run -t -i --user isofit isofit:latest /bin/sh -c 'cd /isofit/examples/20171108_Pasadena/ && ./run_example_libradtran.sh'
saving results to /isofit/examples/20171108_Pasadena/configs/../remote/ang20170228_surface_model.mat
Traceback (most recent call last):
  File "../../isofit/isofit.py", line 126, in <module>
    main()
  File "../../isofit/isofit.py", line 57, in main
    fm = ForwardModel(config['forward_model'])
  File "/isofit/isofit/forward.py", line 64, in __init__
    self.RT = getattr(import_module(module), cname)(config[key])
  File "/isofit/isofit/rt_libradtran.py", line 54, in __init__
    self.build_lut()
  File "/isofit/isofit/rt_lut.py", line 124, in build_lut
    cmd = self.rebuild_cmd(point, fn)
  File "/isofit/isofit/rt_libradtran.py", line 137, in rebuild_cmd
    with open(infilepath0, 'w') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/isofit/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100_alb0.inp'
LEE-158844:isofit_master shawnserbin$
```

serbinsh, 2019-03-15:
I think I sorted it out after exploring the code. It was trying to put files into the "lut" location and I had assumed it made that location for you.  Re-running the docker command as:

```
docker run -t -i --user isofit serbinsh/isofit_containers:isofit_master_latest /bin/sh -c 'cd /isofit/examples/20171108_Pasadena && mkdir -p lut && ./run_example_libradtran.sh'
```

Seems to be working, at least to some extent:

```
saving results to /isofit/examples/20171108_Pasadena/configs/../remote/ang20170228_surface_model.mat
rebuilding
bash /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000.sh
bash /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-1.5000.sh
bash /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-2.0000.sh
bash /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-2.0000.sh
/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-2.0000.sh: line 4:    56 Killed                  ../bin/uvspec < /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-2.0000_alb0.inp > /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-2.0000_alb0.out
/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000.sh: line 4:    55 Killed                  ../bin/uvspec < /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000_alb0.inp > /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000_alb0.out
/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-2.0000.sh: line 4:    57 Killed                  ../bin/uvspec < /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-2.0000_alb0.inp > /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-2.0000_alb0.out
/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-1.5000.sh: line 4:    58 Killed                  ../bin/uvspec < /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-1.5000_alb0.inp > /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-1.5000_alb0.out
/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-2.0000.sh: line 5:    59 Killed                  ../bin/uvspec < /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-2.0000_alb05.inp > /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-2.0000_alb05.out
/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000.sh: line 5:    60 Killed                  ../bin/uvspec < /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000_alb05.inp > /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000_alb05.out
/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-1.5000.sh: line 5:    62 Killed                  ../bin/uvspec < /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-1.5000_alb05.inp > /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-1.5000_alb05.out
/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-2.0000.sh: line 5:    61 Killed                  ../bin/uvspec < /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-2.0000_alb05.inp > /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-2.0000_alb05.out
/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-2.0000.sh: line 6:    63 Killed                  ../bin/uvspec < /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-2.0000_alb025.inp > /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-2.0000_alb025.out
/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000.sh: line 6:    64 Killed                  ../bin/uvspec < /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000_alb025.inp > /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000_alb025.out
...
```

Not sure what those "killed" comments mean

serbinsh, 2019-03-15:
I actually increased the available memory to Docker and may have solved that issue too:

```
docker run -t -i --user isofit -v ~/scratch/isofit_output:/isofit/examples/20171108_Pasadena/output -v ~/scratch/isofit_output/images:/isofit/examples/20171108_Pasadena/images -v ~/scratch/isofit_output/lut:/isofit/examples/20171108_Pasadena/lut serbinsh/isofit_containers:isofit_master_latest /bin/sh -c 'cd /isofit/examples/20171108_Pasadena && mkdir -p lut && ./run_example_libradtran.sh'
saving results to /isofit/examples/20171108_Pasadena/configs/../remote/ang20170228_surface_model.mat
rebuilding
bash /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000.sh
bash /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-2.0000.sh
bash /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-1.5000.sh
bash /isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-2.0000.sh

```

Here is an image of the lut directory in the Docker container mapped to my host drive

[image]

serbinsh, 2019-03-15:
[image]

So the file path issue was the folder "lut" was missing from "/isofit/examples/20171108_Pasadena/" and the "killed" issues had to do with limited memory to the docker container.  After upping that from 2GBs to 8GBs (may not need that much) the code ran!  This is pretty awesome stuff!  

I will leave this open(?) to flag whether or not it should create that LUT folder?  Maybe I missed the docs and if that is supposed to be created by the user

davidraythompson, 2019-03-16:
Hi Shawn,
Great to hear you have this running.  Feel free to submit a pull request to
add LUT directory creation; it sounds like a fine feature!  I'll do it
anyhow in my next updates but it may be a little while before I'm in a
position to submit those to master...
David

pgbrodrick, 2020-03-25:
Resolved in release-2.0.0, through a directory check and creation in look_up_tables.py.

closed 2020-03-25 by pgbrodrick