isofit/isofit #32: conda-forge package for isofit

lewismc, 2019-07-26:
I am proposing that we create an isofit recipe for [conda-forge](https://conda-forge.org/).
This involves [adding a recipie](https://conda-forge.org/#add_recipe) and means that isofit will be available directly from within the Conda packaging environment.

lewismc, 2019-07-26:
I will begin work on this and capture documentation as to how things are done.

davidraythompson, 2019-07-26:
Cool.

lewismc, 2019-07-27:
Initial pull request for the conda-forge recipe is available at https://github.com/conda-forge/staged-recipes/pull/8940

@davidraythompson I see that your clone of isofit under your own github repository is ahead of this repository. Can you clarify which repository is the canonical source? Thanks

davidraythompson, 2019-07-28:
This repository is the canonical stable version.  I use my own clone for development and testing purposes.

lewismc, 2019-07-28:
@davidraythompson ack, I've update the conda-forge recipe to reflect that.

adam-erickson, 2019-10-21:
Just a heads up that we currently have a recipe awaiting approval on conda-forge.

davidraythompson, 2019-10-21:
Thanks Adam (who next takes action on that?)

adam-erickson, 2019-10-21:
I believe we are waiting for approval from you and @winstonolson as the other package maintainers. Let me know if you didn't get a notification and I can send info by email.

davidraythompson, 2019-10-21:
I did not get a notification so an email would be great, sorry about that.  
Since you are facile with that process, and will likely be working with the code for a while, we should consider simply making you a peer "maintainer" for efficiency in the future if you're up for it :)

adam-erickson, 2019-10-21:
Discussion on the pull request can be found here:

https://github.com/conda-forge/staged-recipes/pull/9903

I'll wait until we have a new release binary hosted and will then update the pull request and re-run their integration tests.

lewismc, 2019-10-24:
hi @adam-erickson I commented over on conda-forge/staged-recipes#9903 I'm slightly puzzled as to why the canonical source url if your personal GH account. 
Hopefully the efforts can align... duplication is pointless.
Thanks

adam-erickson, 2019-10-25:
Hi @lewismc, as I mentioned there that was only a temporary solution to get things moving while we continued to make changes to the source for official release.

pgbrodrick, 2023-01-10:
The conda-forge recipe now exists and automatically updates with package changes, starting with #292 and with subtle adjustments since.  Closing.

closed 2023-01-10 by pgbrodrick