isofit/isofit #44: Release location

adam-erickson, 2019-10-21:
If I may make a suggestion, we may want to host official releases here as GitHub has a page/link dedicated to this:

https://github.com/isofit/isofit/releases

Folks that use GitHub a lot may go looking for releases there. We can then point the conda recipe directly to this URL, as it is not buried within the repository.

davidraythompson, 2019-10-21:
Great idea let's do it. Winston, if you agree let's tag, bag & ship :)

winstonolson, 2019-10-21:
Is there anything that needs to be done other than tag a specific version for it to show up on https://github.com/isofit/isofit/releases?  

One of the things I did in the last merge was include the gzipped tarball (isofit/dist/isofit-1.1.6.tar.gz) so that the recipe could point to that file in the source.url field.

adam-erickson, 2019-10-21:
@winstonolson Nothing else needs to be done, just have to use their GUI interface and a consistent naming/versioning scheme. This also has nice functionality in terms of providing users a list of major changes with each new version.

winstonolson, 2019-10-21:
Okay, cool.  I'm reading about it now.  I'll create a tag and release for this latest version shortly.

winstonolson, 2019-10-21:
I created the release: https://github.com/isofit/isofit/releases/tag/v1.1.6. 

It looks like this now gives us a link to the release binary here, https://github.com/isofit/isofit/archive/v1.1.6.tar.gz.  I assume is what conda needs for the recipe.

@adam-erickson, the meta.yaml file in this particular tag does not have this url.  It has a link to the binary in the "dist" folder.  Do we now update the meta.yaml to include a link into "isofit/isofit/archive/" instead?  Still just trying to understand the new process...

Thanks!

adam-erickson, 2019-10-25:
Hi @winstonolson, as mentioned at https://github.com/conda-forge/staged-recipes/pull/9903 we could also point the source in `meta.yaml` to PyPi. However, this seems a bit circular as code goes from GitHub -> PyPi. Also, I don't know that PyPi offers nice release notes and versioning? Either way should work on conda-forge.

adam-erickson, 2019-10-26:
Note that @chrisburr has made the changes to the conda recipe for us and it now points to the isofit/isofit release on GitHub.

davidraythompson, 2020-08-24:
We have a new versioning strategy with periodic stable releases.  Any objections to my closing this issue?

pgbrodrick, 2020-08-24:
None, I believe the essences of this is being done, and is documented.

closed 2020-08-24 by pgbrodrick