isofit/isofit #48: Reduce, simplify code overall

davidraythompson, 2019-11-07:
I'm getting concerned about code bloat as we continue to add functionality.  My vision would be to keep the repository simple, elegant, testable, verifiable, and agile as a research tool.  To that end, it needn't be  monolithic all-singing & dancing repository.  :) 

You all may have different ideas though.  How can we tackle this? If we decide to reduce, perhaps we could start by banishing all ancillary utilities (e.g. utils.py) to a separate repository?  Not every user will have the expertise or interest to - for example - build surface models.  We could also cut out some lesser-used RT options.  if we bring in RTTOV we could abandon LibRadTran, e.g.

pgbrodrick, 2019-11-15:
I'm obviously new to the repo, but as I look through and start to use it I think most of the current code base can remain, but some additional structure would be quite useful to a) know where to look for things and b) isolate out sections that are more or less 'core'.

I would advocate against breaking the repo apart, as many of the tools (utils, surface models, etc.) really are quite useful to have here even if it's for a minority of users.  Relatively speaking, this is also not a large repo yet (not saying we shouldn't stay trim, just that we're still very containable with some organization).

Proposal -Introduce submodules:

**core**
isofit.py
inverse.py
inverse_mcmc.py
inverse_simple.py
geometry.py
sunposition.py
common.py

**applicaiton**
fileio.py
util.py

**surface**
surf.py
surf_emissive.py
surf_glint.py
surf_iop.py
surf_multicomp.py

**radiative_transfer**
rt_6s.py
rt_libradtran.py
rt_lut.py
rt_modtran.py

If folks are willing to give this a try, I can restructure the code (including the install) in a PR so we can take a look, but I'm also the new guy here, so happy to get shot down.

davidraythompson, 2019-11-16:
This sounds great to me.  Wholly agree with this organization. Per utils.py discussion, it might make sense to split that into varied files, all within the application module.

winstonolson, 2019-11-18:
Adding my two cents, I think this added structure helps break down the code into more accessible units.  I might suggest creating a "utils" submodule instead of "application" and then I guess I'd move the fileio.py script back to "core".  But that's just getting picky.  I think think this is a great idea!

pgbrodrick, 2019-11-18:
Yup, I think this makes sense, given the call from isofit.  Will adapt to that and get a PR up.

closed 2019-11-20 by pgbrodrick