isofit/isofit #50: A convenient utils.py solution?

davidraythompson, 2019-11-12:
I'd like to revisit the decision of putting utils.py into the common repository.  Running it as a python function rather than as a standalone turns out to be making it more difficult me to perform quick agile experiments - not only by calling from within shell scripts, as in pbs or other batch worflow systems (which get less readable), but also for running individual spot analysis from the command line itself where I find myself having to reconstruct things in a file rather than tab-completing, etc.  Since utils.py is not part of the critical isofit functionality, and they are not often chained together, can we split utils.py off somehow to shrink the core codebase while still adhering to Adam's preferred distribution ruleset?  Or simply revert? I'm open to all ideas.

pgbrodrick, 2019-11-15:
Would it help to, using the proposed restructure ( #48 ), to break the components of utils.py into individual files and put them each separately in the application submodule?  If not, can you be a little more specific of a command-line use-case that's giving you trouble so I can rec a better structure?

davidraythompson, 2019-11-16:
Your solution sounds great to me. Each could come with its own main routine, and be a command line application - or importable.

closed 2019-11-20 by pgbrodrick