isofit/isofit #51: use of project board

pgbrodrick, 2019-11-15:
Hey All,

Anyone mind if I start a project board for organization of checked out issues?  Would be nice to be able to see what folks are working on / assign/be-assigned tasks to minimize redundant work.  Also would be good to post some of the roadmap items on there.

winstonolson, 2019-11-18:
Sounds like a good idea to me, and a chance to revisit older issues that may have gone stale or become obsolete.  I say go for it!

pgbrodrick, 2019-11-18:
Great - can you add me as a collaborator so I can build one?

winstonolson, 2019-11-18:
I just realized I don't have access to the Settings on this repo.  @davidraythompson, can you add Phil as a collaborator when you have a chance?  Thanks!

isofit, 2019-11-18:
Hi Winston, @davidraythompson is also a lowly collaborator without the ability to change settings.  But Behold! I, the omniscient entity ISOFIT, have now invited Phil to the official repo to wield the sacred push/pull access.

pgbrodrick, 2019-11-18:
Thanks omiscient entity!  Will start up a some projects, feel free to edit at your leisure.

closed 2019-11-18 by pgbrodrick