isofit/isofit #56: bound violation after simple_inverse

pgbrodrick, 2019-12-08:
Currently, depending on the configuration, x0 can end up outside of optimization bounds following the simple inversion x0 seeding, given that the simple inversion uses a heuristic.  This is easily solved by checking and enforcing bounds post preseed.

Flagging here as a known issue, which will be fixed in my next PR.

davidraythompson, 2020-02-10:
I had a similar issue, and just placed a PR that introduces out-of-bounds extrapolation by the RegularGridInterpolator object in the lookup table.  I'm not sure if this will solve your original case, but it seems likely.

pgbrodrick, 2020-03-25:
Resolved in speedups a while back, definitely included in release-2.0.0.

closed 2020-03-25 by pgbrodrick