isofit/isofit #57: Forward model with datacube throws error from interp1d

adam-erickson, 2020-01-03:
The `resample_reflectance()` function in `isofit.core.surface` is raising an error that originates from its `interp1d()` function:

`ValueError: x and y arrays must be equal in length along interpolation axis.`

The error appears attributable to the parsing of the file `data/stonewall_rfl.txt` by the function `load_spectrum()`. The two vectors output by the function appear unequal in length per the error. Looking into this now...

adam-erickson, 2020-01-03:
It appears that the first time that `resample_reflectance()` is called the dimensions of `self.rfl` and `self.rwl` match, but they do not the second time:

```bash
(425,) (425,)
(425,) (211,)
```

pgbrodrick, 2020-06-12:
Solved as of PR #115.

closed 2020-06-12 by pgbrodrick