isofit/isofit #58: Feature request: Isofit() arguments to define function behavior

adam-erickson, 2020-01-06:
Hi Everyone, looking forward, it would be great to modify the `Isofit` class so that arguments are passed to it that define the action to perform. This will make `isofit` library behaviour more transparent and Pythonic (i.e., idiomatic), reducing training time to attract more users and contributors. Even if we are worried about keeping the model parameters self-contained in JSON files for portability, the JSON dictionaries could be loaded to `dict` and passed as `kwargs` rather than a JSON path. This would allow supporting both behaviours - JSON path or `kwargs` - at minimal cost to the programmer. This would also make it easier to use the library dynamically, or in other words, generate parameters. Ideally, each model action would have its own member function of the `Isofit` class for clarity, debugging, and portability.

Thank you all for providing an excellent tool that I am very happy to help advance.

adam-erickson, 2020-01-08:
Also, I'm not sure about your feeling regarding documentation, but we may want to prefer vanilla `docstrings` to Google, etc., as the former is more common, descriptive, and is supported by 'default' autoformatters like `autopep8`.

Here is an example of vanilla `docstrings`:

```python
def some_function():
        """A description of a function.

        Parameters
        ----------
        func : object
            a function to return with or without a decorator

        Returns
        -------
        object
            original function without decorator if condition is unmet
        """
```

And the same example in Google style `docstrings`:

```
def some_function():
        """A description of a function.

        Args:
            func: A function to return with or without a decorator.

        Returns:
            Original function without decorator if condition is unmet.
        """
```

If you want to go the Google route to save LOC, their complete style guide for Python can be found here ([link](http://google.github.io/styleguide/pyguide.html)) and is supported by newer autoformatters such as YAPF and Black that let you select the style to apply similar to `clang-format`. I've heard great things about Black, but it appears to still be in beta. Curious to hear your thoughts on this.

davidraythompson, 2020-01-09:
Hi Adam,

Great ideas - as one voice on the committee, my advocacy is for simplicity (and criticalaly reducing the lines of code!) wherever possible.  I'm happy to clean up the configuration strategy. 
 That, and fileio.py, seem a bit like blemishes on an otherwise elegant codebase :)

One thing we should consider is that dictionaries provide a nice hierarchical structure.  Would a **kwargs interface lock us in to a flat structure, or would we end up passing nested dictionaries in the end, anyway?  

Finally note that there are internal dependencies - i.e. different surface or radiative transfer model options require various different field definitions.  

I propose we discuss documentation as a separate issue- in brief I too love docstrings and agree we need a good consistent self-contained strategy for documentation in the code.

adam-erickson, 2020-01-10:
Thank you, @davidraythompson! I agree, it's a lovely codebase to use. Here is a simple example showing how `**kwags` can enable both styles of passing arguments:

```python

def mult(x, y):
    return x * y

mult(2, 3)

p = {'x': 2, 'y': 3}
mult(**p)
```

So, if the parameters are stored in a JSON file, we would simply load them to `dict` and pass them to functions using `**kwargs`. This will automatically expand the parameters. This would require specialized functions for processing each type of JSON parameter file, rather than walking the JSON structure and conditionally choosing a process to perform implicitly.

pgbrodrick, 2020-03-28:
Hey All,

Hopping in to comment here, as I think I started working on this from a different angle.

My voice would be to push for object-oriented configs, letting us hold cleanly to hierarchical structure in a transparent way (that's traceable through the code).  The core disadvantage here is it requires a bit of boilerplate code to assign dictionary properties into an object.  This is can certainly be handled in more elegant ways, but doing it in the config-object way makes all properties explicit and easy to trace....so that would be my vote.  [Example from another project](https://github.com/pgbrodrick/bfg-nets/blob/master/bfgn/configuration/configs.py).

This is the only way that I've found to pass around complicated configurations in a way that's traceable at all.  If we go the kwargs route, everything does indeed look more pythonic, but it gets very difficult very quickly to figure out what parameters are attached, kindof like it is now.

While the boilerplate here will be a bit annoying, it's also a good way that we can add checking and enforcement of file parameters *early*.  E.g., this is where we check all file paths and throw structured (and combined!!!!) errors, so during use we don't have to run through a long series of detecting missing files one after another.  It's far from perfect, but you can get a sense of what this looks like.  Happy to hear better methods if there are any, but I'd really like to avoid dictionaries moving forward, they caused me some pretty significant headache during #73 .  Thoughts?

Re docstrings, I'm gonna put a strong vote in towards Google-style, largely as that's what I've gotten used to but also because I've had better luck coordinating those with external packages to automate read-the-docs style documentation.  @adam-erickson, let me know if you object.

adam-erickson, 2020-03-28:
Hi @pgbrodrick, I fully understand what you mean. I've used configuration classes in other projects for the very reasons you mentioned; it provides a nice traceable way of encapsulating configuration data to be passed around to other objects. It made sense for that project and it may for ISOFIT's configurations as well. In that case, I wrote a class method that writes out the parameters to a single JSON file for portability. Also for portability, I suggest maintaining the ability to load parameters from JSON configuration files, as this opens up a world of web applications should the need arise in the future (e.g., returning JSON parameters from a REST/GraphQL API query over HTTP). That too can be accomplished with a helper class method such as `load_json`. Meanwhile, the parameters are still a class instance internally with the niceties that you mentioned. They could also be loaded into the parameters class from **kwargs if so inclined.

About the Pythonic bit, I was mostly commenting on how internally it is really hard to see which mode of operation is being run within `IO.__next__()`. For this, it would be great to have simple if-else statements for LUT file generation, forward modeling, and inverse modeling modes. The configuration file itself would benefit from an explicit `mode` parameter. Also internally, it would make it easier on us to have the OE bits as isolated from the other code as much as possible, perhaps within an OE class. This will make testing different optimization methods easier. I have some specific thoughts related to this that I will send in a follow-up e-mail... :+1: 

For docstrings, I've always been partial to the NumPy style for how clean it is to read, but the Google styles are nice as well and are available in one place. Regarding the implementation, there is a nice tutorial from the author of the `requests` library here: https://docs.python-guide.org/writing/documentation/. Everything I've mentioned is just a suggestion/opinion, so please feel free to do whatever is necessary to advance the project. I'm very excited that you are taking on this work and am happy to help however I can.

davidraythompson, 2020-03-28:
I absolutely agree with many of @adam-erickson and @pgbrodrick's statements above, including: (1) the value of explicit parameters/member data rather than implicit dictionary entries in our interfaces, and (2) having an explicit switch to activate "lut_only" vs. "forward simulation" vs. "inversion."  Right now the mode is inferred in non-obvious ways from the configuration at various places throughout fileio.py.  We might also consider a different switch for "ASCII single-spectrum" vs. "Binary multiple-spectrum" behavior.  

I'm agnostic about docstring style, and @adam-erickson seems flexible, so does @pgbrodrick's "strong vote" decide our course?

pgbrodrick, 2020-06-12:
Marking this as closed - documentation is substantially updated in PR #119 , using google doc strings.  Several of the other discussion points have been (partially) attempted during the 2.0 -> 3.0 pathway.  Specific points can be re-issued as needed!

closed 2020-06-12 by pgbrodrick

adam-erickson, 2020-06-13:
Excellent! I think this will be a valuable contribution to ISOFIT.