isofit/isofit #69: Interpolation error output from forward model in gridded case

adam-erickson, 2020-02-14:
Hi all, when I run gridded 50x50 hyperspectral datacubes, I get the following error in forward mode:

```python
Traceback (most recent call last):
  File "/Users/username/anaconda3/lib/python3.7/site-packages/isofit-1.4.3-py3.7.egg/isofit/core/isofit.py", line 88, in __init__
    self.__call__()
  File "/Users/username/anaconda3/lib/python3.7/site-packages/isofit-1.4.3-py3.7.egg/isofit/core/isofit.py", line 119, in __call__
    self.fm.reconfigure(*configs)
  File "/Users/username/anaconda3/lib/python3.7/site-packages/isofit-1.4.3-py3.7.egg/isofit/core/forward.py", line 317, in reconfigure
    self.surface.reconfigure(config_surface)
  File "/Users/username/anaconda3/lib/python3.7/site-packages/isofit-1.4.3-py3.7.egg/isofit/surface/surface.py", line 61, in reconfigure
    self.resample_reflectance()
  File "/Users/username/anaconda3/lib/python3.7/site-packages/isofit-1.4.3-py3.7.egg/isofit/surface/surface.py", line 67, in resample_reflectance
    p = interp1d(self.rwl, self.rfl, fill_value='extrapolate')
  File "/Users/username/anaconda3/lib/python3.7/site-packages/scipy/interpolate/interpolate.py", line 431, in __init__
    _Interpolator1D.__init__(self, x, y, axis=axis)
  File "/Users/username/anaconda3/lib/python3.7/site-packages/scipy/interpolate/polyint.py", line 60, in __init__
    self._set_yi(yi, xi=xi, axis=axis)
  File "/Users/username/anaconda3/lib/python3.7/site-packages/scipy/interpolate/polyint.py", line 125, in _set_yi
    raise ValueError("x and y arrays must be equal in length along "
ValueError: x and y arrays must be equal in length along interpolation axis.
```

It appears that the IO() object is iterated over each pixel and fed into `self.fm.reconfigure(*configs)`. While this runs file the first iteration, on the second iteration it appears to pull the reflectances from the `reference_reflectance` ENVI file and try to interpolate these with the sensor band positions from the `data/wavelengths.txt` file. These are different lengths and it errors (the reference data is 211 bands while the target is 213). Where this gets quite weird is that it works correctly for the standard 1x15 bare-earth `reference_reflectance` ENVI file.

I'm running a trace now to try to dig to the source of this problem. A major rewrite of the IO class is recommended for the future, as the way it selects the operation mode from the JSON configs is extremely opaque. I would recommend a standard argument like 'mode' that executes specialized subroutines conditional upon the value of this switch - then it will be easier to understand the call graph and where the data goes.

adam-erickson, 2020-02-14:
So, I modified the surface `wavelength_file` to match the surface reflectance bands and not the `wavelength_file` used in `instrument` and `radiative_transfer`, which are the sensor wavelengths I believe. I don't know if this is correct, but it works. Please be sure to correct me if I am doing this wrong, as it is not described in the wiki ([link](https://github.com/isofit/isofit/wiki/Code-and-File-Formats#explanation-of-python-files)).

pgbrodrick, 2020-07-10:
Stale issue, resolved through some of the systematic refactors a while back, but this got left behind.  Closing!

closed 2020-07-10 by pgbrodrick