isofit/isofit #70: Complex square root inverse covariance

jfahlen, 2020-02-22:
In line 187 in common.py, I'm getting a tiny negative eigenvalue that's causing the square root on the next line to generate complex square root inverse covariance matrices and real-valued but complex-typed inverse covariances. I assume this is a round off issue as a covariance matrix should always have positive eigenvalues. I'm not sure what the fix to this should be. Should we clip tiny eigenvalues? Something else?

It's possible that my matrix is singular or at least ill conditioned, but the code should be robustified against tiny negative eigenvalues.

jfahlen, 2020-02-22:
This seems to be a consequence of pull request #67.

davidraythompson, 2020-02-26:
Hi Jay,
Jouni (resident statistician) believes the new inversion approach should actually be more stable - and faster. But if you're getting a negative eigenvalue we certainly should robustify it.  
I wonder if we could just floor the eigenvalues at some infinitesimally-small real positive value?
David

pgbrodrick, 2020-03-06:
Just reporting that I'm seeing the same issue as @jfahlen, only after updating to the latest decomposition.  Doing some tests, but some form of robustification (or utilization of an alternate decomposition package/approach) seems necessary.

davidraythompson, 2020-03-06:
I'm consulting with Jouni on how he would deal with this (preconditioning the covariance matrix, replacing negative eigenvalues with some small positive floor, etc.)

pgbrodrick, 2020-03-06:
Great!  I'm seeing poorly conditioned input matrices in my particular example, order e5, which is likely the root problem.   Currently digging backwards to try and find out why, and to see if there's something we can manage on the physical side of things to get things better conditioned (though we still want to get things numerically stable - or at least flagged - no matter what).

pgbrodrick, 2020-03-14:
There's an attempted fix in play, using diagonal offsets as needed.  If the inversion fails, a value error will now be thrown.

closed 2020-03-14 by pgbrodrick