isofit/isofit #74: forward model code simplification

pgbrodrick, 2020-03-17:
Reduce some code layers, so that look_up_tables, which is currently the true parent to modtran/6s/libradtran, is actually that - radiative_transfer.  This means merging in some of the radiative_transfer functionality.  Some of radiative_transfer functionality should migrate upwards, into forward model.

General idea is that we should have:
- an abstract forward model class, that handles things like making specific calls to support the inversion
- a radiative transfer parent class, that holds generic radiative-transfer modeling components, and
- specific radiative transfer classes (like modtran/libradtran) that inherit from radiative_transfer but have some specific attributes.

adam-erickson, 2020-04-18:
I have some thoughts about this : ) Before doing any reorganization, I would draw out the logical relationships between classes on paper or a white board (or LaTeX, Keynote, etc.), including classes that may be needed in the future but do not exist today. Currently, the class hierarchy logic is very jumbled.

Since everything in ISOFIT is radiative transfer after all, I would avoid naming classes things like `RadiativeTransfer` which are quite ambiguous. Instead, I would break down classes into descriptive names such as, for example:

- Surface
- Atmosphere
- Architecture
- Solvers/Estimators
- OperationMode
- Isofit (this would be unchanged as the main interface)

The OperationMode class would contain Forward and Inverse subclasses, or better yet, functions within the same class. An if-then switch (Python doesn't have real switches) using a `mode` parameter equal to `'forward'` or `'inverse'` would determine the class behaviour. The Solvers (or Estimators) class would include OE, MCMC, and Algebraic, isolating these codes so that we may properly optimize them. A Surface parent class would store the surface state and any surface model interfaces. An Atmosphere parent class would do the same for the atmosphere, abandoning referring to it as 'radiative transfer.'

I'm guessing that eventually, things like orbital models may be desired, so I would make that a child class of the Architecture class along with the Instrument class. The ObservationGeometry class (more descriptive than simply Geometry) would go in there as well. Surface classes could do the same by allowing space for classes that interface models like PROSAIL in addition to simple Lambertian models, as the former is widely popular and already has a nice Python interface. This will help future-proof the class hierarchy.

As an aside, another class with a non-descriptive name is `VectorInterpolator` (doesn't that just mean 1-D interpolation?). Its initialization function also does a lot more than initialization. Calling it does... it's hard to say because it isn't documented. If it's just an interpolator, a function would be sufficient, but clearly its doing more than interpolation.

davidraythompson, 2020-04-19:
Hi Adam,
Excellent ideas as always. The core hierarchy of function calls is:

Inversion Algorithm
      | ---Forward Model
                  |--- Surface (only knows about surface reflectance and emission properties)
                  |--- Instrument (rather than "architecture," because geometry and instrument 
                  |                            is all we need) 
                  |--- Radiative Transfer (more general than "atmosphere" because it knows 
                                                about the sun, and all surface/atmosphere coupling.  Also, 
                                                in the future it may also calculate photon transport for 
                                                planetary spectra from airless bodies)

As we move toward operations on EMIT, the team will be focusing mostly on consolidation, documentation, simplification & testing.  There's a lot we could do to simplify the top level interface and I/O.  I particularly like your idea of making operational "modes" explicit in the configuration.  I can envision two switches: "simulation" vs. "retrieval," and "single-spectrum text" vs. "multi-spectrum binary."  Is that what you had in mind?

Are you planning to take on modal behavior on as a project, and is this something we could do by changing only the top-level isofit.py and fileio.py?

adam-erickson, 2020-04-20:
Hi David, thank you for the kind words! : ) I was thinking that the proposed class hierarchy could look something like this (omitted items will be sent via email):

```
Isofit
      |--- Architecture
            |--- Instrument
            |--- ...
            |--- ...
            |--- Orbit
                  |--- ObservationGeometry
            |--- ...
      |--- Surface
            |--- Lambertian
            |--- MultiComponent
            |--- PROSAIL
            |--- SCOPE
            |--- ...
      |--- Atmosphere
            |--- Modtran
            |--- LibRadTran
            |--- SixS
            |--- ...
      |--- OperationMode
            |--- Forward
            |--- Inverse
      |--- Solvers/Estimators
            |--- OE
            |--- MCMC
                  |--- MetropolisHastings
                  |--- HamiltonianNUTS
            |--- Algebraic
      |--- Retrieval
            |--- Model
            |--- Observation
      |--- Statistics
      |--- Plotting
```

The idea behind making the hierarchy flatter, or less hierarchical, would be to make the modules more portable. _Ideally_, users would be able to do `from isofit.solvers import OE` and apply it to, for example, a full instrument-to-surface inversion or a standalone surface model with a dictionary or class of parameters. I'm not sure if that would be feasible in a first-pass refactoring, but it may be a good design target to have. Additionally, we may want to have Retrieval, Statistics, and Plotting classes down the road, some of which could borrow from our other library. The optional Plotting functions would call the Retrieval and Statistics functions on the model outputs and plot the results. The cool thing is that this would allow users to pre-define a domain-specific model retrieval as well as a 'ground-truth' observation retrieval for generating statistics and plots, also super useful for our other library. These things would greatly simplify writing wrapper libraries around ISOFIT.

The main switch I had in mind is the `forward` or `inverse` mode, but `simulation` or `retrieval` would be another internal switch based on a config parameter. Switching between text and binary mode should be possible just by checking the file name and/or data and thus may not need an explicit parameter. Nevertheless, the logic should be explicit and human-readable.

This will be a big undertaking with much of the challenge being fully comprehending the existing codebase. I'm happy to tackle this if it needs to be done, which it sounds like it does, but I cannot do so immediately as I have other commitments at the moment. What I will be able to do is chip away at a major refactoring as a side project, if that would be sufficient.

brentwilder, 2026-09-08:
Marking this as a closed since mostly this conversation reflects refactoring discussions from ISOFIT-v2 to ISOFIT-v3 (we are now on V4 after a large refactoring effort earlier this year).

Additionally, all of the items highlighted by Phil in the initial issue post have largely been addressed. 1. There is an abstract forward model class. One may use this flexibility in a variety of ways, including to support different types of inversion methods. 2. We now currently have `./atmosphere/atmosphere.py` that handles general atmospheric radiative transfer portions of the codebase, as well as, 3. individual engines in `./atmosphere/engines/` (e.g., MODTRAN, libRadtran).

Another component we addressed ("forward model code simplification") from our latest refactoring effort was the removal of `radiative_transfer_engine.py` (in ISOFIT-v3), which contained components of the actual forward model as we traditionally have it written. We worked most of the code from `radiative_transfer_engine.py` into either `atmosphere.py` or `forward.py`

closed 2026-09-08 by brentwilder