isofit/isofit #77: Path issue on linux cluster

serbinsh, 2020-03-26:
I have typically been running ISOFIT on my MacOSX laptop/desktop (and in the past Docker) but have been migrating my efforts to my Linux cluster.  I built a new conda environment on the cluster and installed the master branch of isofit

```
commit dfbfef7ad3cafac2ff6842728eee380232335860
Author: David R Thompson <davidraythompson@gmail.com>
Date:   Tue Feb 11 15:55:58 2020 -0800

    Add sha256 hash value to conda-forge recipe

```

I have also built LibRadTran on the system (2.0.2) and tested that is functions properly using built in examples

```
(/data2/sserbin/conda_envs/isofit) [sserbin@modex isofit]$ $LIBRADTRAN_DIR
-bash: /data/software/libradtran/2.0.2: is a directory
```

```
(/data2/sserbin/conda_envs/isofit) [sserbin@modex isofit]$ ls $LIBRADTRAN_DIR
bin  include  lib  share
```

However when I run the example in ISOFIT I get a string of errors:

```
[sserbin@modex 20171108_Pasadena]$ conda activate /data2/sserbin/conda_envs/isofit
(/data2/sserbin/conda_envs/isofit) [sserbin@modex 20171108_Pasadena]$ ./run_example_libradtran.sh
Loading  /data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../../../data/reflectance/surface_model_ucsb
saving results to /data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../remote/ang20170228_surface_model.mat
rebuilding
bash /data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh
/data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh: line 3: cd: /data/software/libradtran/2.0.2/test: No such file or directory
/data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh: line 4: ../bin/uvspec: No such file or directory
/data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh: line 5: ../bin/uvspec: No such file or directory
/data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh: line 6: ../bin/uvspec: No such file or directory
/data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh: line 7: ../bin/zenith: No such file or directory
/data2/sserbin/conda_envs/isofit/lib/python3.7/site-packages/scipy/_lib/deprecation.py:20: UserWarning: loadtxt: Empty input file: "/data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../lut//LUT_H2OSTR-1.5000_AOT550-0.0100_alb0.out"
  return fun(*args, **kwargs)
Traceback (most recent call last):
  File "/data2/sserbin/conda_envs/isofit/bin/isofit", line 54, in <module>
    main()
  File "/data2/sserbin/conda_envs/isofit/bin/isofit", line 49, in main
    return Isofit(config_file=config, level=level, row_column=row_column, profile=profile)
  File "/data2/sserbin/conda_envs/isofit/lib/python3.7/site-packages/isofit/core/isofit.py", line 59, in __init__
    self.fm = ForwardModel(self.config['forward_model'])
  File "/data2/sserbin/conda_envs/isofit/lib/python3.7/site-packages/isofit/core/forward.py", line 72, in __init__
    self.RT = getattr(import_module(module), cname)(config[key])
  File "/data2/sserbin/conda_envs/isofit/lib/python3.7/site-packages/isofit/radiative_transfer/libradtran.py", line 44, in __init__
    self.build_lut()
  File "/data2/sserbin/conda_envs/isofit/lib/python3.7/site-packages/isofit/radiative_transfer/look_up_tables.py", line 164, in build_lut
    self.load_rt(point, fn)
  File "/data2/sserbin/conda_envs/isofit/lib/python3.7/site-packages/isofit/radiative_transfer/libradtran.py", line 172, in load_rt
    wl, rdn0,   irr = s.loadtxt(self.lut_dir+'/LUT_'+fn+'_alb0.out').T
ValueError: not enough values to unpack (expected 3, got 0)
```

However, if I go into the LUT folder and edit on of the LibRadTran inputs to point it to the location of the data folder (i.e. provide the full path):

```
atmosphere_file midlatitude_winter
source solar /data/software/libradtran/2.0.2/share/libRadtran/data/solar_flux/kurudz_0.1nm.dat
albedo 0.25
rte_solver disort
pseudospherical
umu 1.0
phi 0.0
phi0 180.0
zout 2.3
altitude 0.24
time 2017 11 08 18 42 27
mol_modify O3 300 DU
latitude N 34.139247
longitude W 118.127521
output_user lambda uu eglo
mol_abs_param reptran fine
mol_modify H2O 15.0 MM
crs_model rayleigh bodhaine
wavelength 350.0 2520.0
aerosol_default
aerosol_visibility 181.19606324354544
quiet
```

It then runs fine

```
/data/software/libradtran/2.0.2/bin/uvspec < LUT_H2OSTR-1.5000_AOT550-0.0100_alb025.inp

...

2517.900   2.004355920e-04  3.566952e+00
 2518.000   1.949800208e-04  3.469864e+00
 2518.100   1.963903487e-04  3.494962e+00
 2518.200   1.921122894e-04  3.418830e+00
 2518.300   1.700568711e-03  7.597309e+00
 2518.400   1.680064714e-03  7.505707e+00
 2518.500   1.688385033e-03  7.542878e+00
 2518.600   1.701516565e-03  7.601543e+00
 2518.700   1.697550179e-03  7.583824e+00
 2518.800   1.666747150e-03  7.446210e+00
 2518.900   1.137489962e-04  2.462983e+00
 2519.000   1.114159313e-04  2.412466e+00
 2519.100   1.139678279e-04  2.467721e+00
 2519.200   1.139016676e-04  2.466288e+00
 2519.300   1.091709491e-04  2.363855e+00
 2519.400   1.138594816e-04  2.465375e+00
 2519.500   1.138837033e-04  2.465899e+00
 2519.600   6.134965224e-04  4.921408e+00
 2519.700   6.188083789e-04  4.964019e+00
 2519.800   6.076458958e-04  4.874475e+00
 2519.900   6.257934147e-04  5.020052e+00
 2520.000   6.196629256e-04  4.970874e+00
(/data2/sserbin/conda_envs/isofit) [sserbin@modex lut]$
```

Thus, in a nutshell I had to provide the full path to the uvspec bin and the data folder but I provided my base LibRadTran dir via the env variable?  I am using the location where i ran "make install" for LibRadTran not where I compiled it.  I tried that as well and seemed to have similar issues.  I havent had these issues on other machines or my docker container.

serbinsh, 2020-03-26:
I recompiled LibRadTran just to confirm (though i know it was working previously) and tried re-running the example

```
[sserbin@modex 20171108_Pasadena]$ conda activate /data2/sserbin/conda_envs/isofit
(/data2/sserbin/conda_envs/isofit) [sserbin@modex 20171108_Pasadena]$ ./run_example_libradtran.sh
Loading  /data2/sserbin/NASA_SBG/isofit_back/examples/20171108_Pasadena/configs/../../../data/reflectance/surface_model_ucsb
saving results to /data2/sserbin/NASA_SBG/isofit_back/examples/20171108_Pasadena/configs/../remote/ang20170228_surface_model.mat
rebuilding
bash /data2/sserbin/NASA_SBG/isofit_back/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh
/data2/sserbin/NASA_SBG/isofit_back/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh: line 3: cd: /data/software/libradtran/2.0.2/test: No such file or directory
/data2/sserbin/NASA_SBG/isofit_back/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh: line 4: ../bin/uvspec: No such file or directory
/data2/sserbin/NASA_SBG/isofit_back/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh: line 5: ../bin/uvspec: No such file or directory
/data2/sserbin/NASA_SBG/isofit_back/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh: line 6: ../bin/uvspec: No such file or directory
/data2/sserbin/NASA_SBG/isofit_back/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh: line 7: ../bin/zenith: No such file or directory
/data2/sserbin/conda_envs/isofit/lib/python3.7/site-packages/scipy/_lib/deprecation.py:20: UserWarning: loadtxt: Empty input file: "/data2/sserbin/NASA_SBG/isofit_back/examples/20171108_Pasadena/configs/../lut//LUT_H2OSTR-1.5000_AOT550-0.0100_alb0.out"
  return fun(*args, **kwargs)
Traceback (most recent call last):
  File "/data2/sserbin/conda_envs/isofit/bin/isofit", line 54, in <module>
    main()
  File "/data2/sserbin/conda_envs/isofit/bin/isofit", line 49, in main
    return Isofit(config_file=config, level=level, row_column=row_column, profile=profile)
  File "/data2/sserbin/conda_envs/isofit/lib/python3.7/site-packages/isofit/core/isofit.py", line 59, in __init__
    self.fm = ForwardModel(self.config['forward_model'])
  File "/data2/sserbin/conda_envs/isofit/lib/python3.7/site-packages/isofit/core/forward.py", line 72, in __init__
    self.RT = getattr(import_module(module), cname)(config[key])
  File "/data2/sserbin/conda_envs/isofit/lib/python3.7/site-packages/isofit/radiative_transfer/libradtran.py", line 44, in __init__
    self.build_lut()
  File "/data2/sserbin/conda_envs/isofit/lib/python3.7/site-packages/isofit/radiative_transfer/look_up_tables.py", line 164, in build_lut
    self.load_rt(point, fn)
  File "/data2/sserbin/conda_envs/isofit/lib/python3.7/site-packages/isofit/radiative_transfer/libradtran.py", line 172, in load_rt
    wl, rdn0,   irr = s.loadtxt(self.lut_dir+'/LUT_'+fn+'_alb0.out').T
ValueError: not enough values to unpack (expected 3, got 0)
```

Now I see that it was looking in a "test" folder that comes from make check I believe 

```
bash /data2/sserbin/NASA_SBG/isofit_back/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh
/data2/sserbin/NASA_SBG/isofit_back/examples/20171108_Pasadena/configs/../lut/LUT_H2OSTR-1.5000_AOT550-0.0100.sh: line 3: cd: /data/software/libradtran/2.0.2/test: No such file or directory
```

serbinsh, 2020-03-26:
And interestingly after re-compiling I get this running the input files

```
(/data2/sserbin/conda_envs/isofit) [sserbin@modex lut]$ /data/software/libradtran/2.0.2/bin/uvspec < LUT_H2OSTR-2.0000_AOT550-0.1000_alb025.inp
**********************************************************************************
*Error: Data files for REPTRAN not found in directory data/correlated_k/reptran. *'
*       Please check whether you have downloaded the required REPTRAN data files *
*       from http://www.libradtran.org/doku.php?id=download and unzipped the data*
*       in the libRadtran folder.                                                *
**********************************************************************************
Error 2 while opening the representative wavelengths file.
Error 2 returned by function set_transmittance_wl_grid_reptran in (line 292, function 'setup_wlgrid', file 'ancillary.c')
Error 2 setting up transmittance wavelength grid in uvspec (uvspec.c)
Error 2 during execution of uvspec

```

But I can run the LRT examples just fine.....not sure why LRT is now having trouble finding the data folder?  Its there...

serbinsh, 2020-03-26:
I added the LibRadTran data dir to my path as expressed in the INSTALL (did this before but didnt seem to help)

and now get 

```
(/data2/sserbin/conda_envs/isofit) [sserbin@modex lut]$ /data/software/libradtran/2.0.2/bin/uvspec < LUT_H2OSTR-1.5000_AOT550-0.0100_alb025.inp
Error, file '../data/solar_flux/kurudz_0.1nm.dat' not found!
Error -1 reading ../data/solar_flux/kurudz_0.1nm.dat
Error -1 reading ../data/solar_flux/kurudz_0.1nm.dat
Error -1 setting up extraterrestrial irradiance in uvspec (uvspec.c)
Error -1 during execution of uvspec
```

It starts running but fails to find the data it needs to fully execute.  It must be some assumptions of paths or other that is tripping this up?  Seems I must be close but just missing something

serbinsh, 2020-03-26:
Strange....I re-edited the LUT file and tried again and this time it worked after point to the lib rad tran data dir on my path

```
source solar /data/software/libradtran/2.0.2/data/solar_flux/kurudz_0.1nm.dat
```

```
...
 2518.600   1.701516565e-03  7.601543e+00
 2518.700   1.697550179e-03  7.583824e+00
 2518.800   1.666747150e-03  7.446210e+00
 2518.900   1.137489962e-04  2.462983e+00
 2519.000   1.114159313e-04  2.412466e+00
 2519.100   1.139678279e-04  2.467721e+00
 2519.200   1.139016676e-04  2.466288e+00
 2519.300   1.091709491e-04  2.363855e+00
 2519.400   1.138594816e-04  2.465375e+00
 2519.500   1.138837033e-04  2.465899e+00
 2519.600   6.134965224e-04  4.921408e+00
 2519.700   6.188083789e-04  4.964019e+00
 2519.800   6.076458958e-04  4.874475e+00
 2519.900   6.257934147e-04  5.020052e+00
 2520.000   6.196629256e-04  4.970874e+00
```

But if I log into another node and try I get the error about not finding the data again despite it being on my path.....

BUT

```
(/data2/sserbin/conda_envs/isofit) [sserbin@node03 lut]$ LIBRADTRAN_DATA_FILES=/data/software/libradtran/2.0.2/share/libRadtran/data /data/software/libradtran/2.0.2/bin/uvspec < LUT_H2OSTR-1.5000_AOT550-0.0100_alb025.inp
```

seems to run on that node. It seems to be related to finding the LRT data folder but I'm not sure how I should be defining this for ISOFIT so that the paths match expectations and can resolve properly when running isofit or hypertrace, for that matter.  In this case I had again to edit the data path in the *inp to point directly to the file...

```
...
 2518.000   1.949800208e-04  3.469864e+00
 2518.100   1.963903487e-04  3.494962e+00
 2518.200   1.921122894e-04  3.418830e+00
 2518.300   1.700568711e-03  7.597309e+00
 2518.400   1.680064714e-03  7.505707e+00
 2518.500   1.688385033e-03  7.542878e+00
 2518.600   1.701516565e-03  7.601543e+00
 2518.700   1.697550179e-03  7.583824e+00
 2518.800   1.666747150e-03  7.446210e+00
 2518.900   1.137489962e-04  2.462983e+00
 2519.000   1.114159313e-04  2.412466e+00
 2519.100   1.139678279e-04  2.467721e+00
 2519.200   1.139016676e-04  2.466288e+00
 2519.300   1.091709491e-04  2.363855e+00
 2519.400   1.138594816e-04  2.465375e+00
 2519.500   1.138837033e-04  2.465899e+00
 2519.600   6.134965224e-04  4.921408e+00
 2519.700   6.188083789e-04  4.964019e+00
 2519.800   6.076458958e-04  4.874475e+00
 2519.900   6.257934147e-04  5.020052e+00
 2520.000   6.196629256e-04  4.970874e+00
(/data2/sserbin/conda_envs/isofit) [sserbin@node03 lut]$
```

davidraythompson, 2020-03-26:
Hi Shawn,

Thanks for the detailed forensic info here.  LibRadTran paths are always tricky.  In my installation, I believe it references the reptran absorption data file location relative to the current working directory where uvspec is run, so you may have to be in the top-level or test/ subdirectories for it to work. 

To double-check, is your $LIBRADTRAN_DIR consistent with your data directory location? In a typcial installation, data/ is at the top level directory of the libRadTran installation, but it looks like yours might be several layers deeper.

As an aside, I infer you're doing this for the SBG UQ exercise.  For that purpose, I'd HIGHLY recommend MODTRAN instead; LibRadTran is provided but we always get much better results with MODTRAN due - I believe - to the finer resolution band model.  Adam's distribution comes with precalculated lookup tables, so you should not need a license unless you want a different atmosphere than the ~8 canonical types provided.  In that case, I can work with you to provide the LUTs you'd need.

David

serbinsh, 2020-03-26:
@davidraythompson Thanks!  Yeah, apologies for the long thread....

I will answer the second part first - yes, I was working on getting everything running on the cluster to work on things related to SBG. I do plan to use the MODTRAN canonical atmospheres with the UQ runs for Ben; my scenes are in Alaska and I think there is a high-latitude summer atmosphere correct?  However I was also interested in playing around with running my own LUTs via LibRadTran as well, mostly for my own interests and understanding.

Ok, as for the install - I followed the INSTALL instructions for LRT.  When you run make install it creates the dirs I showed above, including putting the data in share/libRadtran/data which is much deeper than you say ISOFIT is looking for.  However I did copy that data dir up to the top of the LIBRADTRAN_DIR; perhaps the disconnect is the use of the compile lib dir vs using make install dir?  

Let me try again with the /data in the top of the dir but I did try earlier and had similar issues :/. its possible I was doing things out of order.  I will let you know what I find.

serbinsh, 2020-03-26:
OK well I have tried running with data in the top level of the installed LIBRADTRAN_DIR

```
(/data2/sserbin/conda_envs/isofit) [sserbin@node03 2.0.2]$ ll
total 8
drwxrwxr-x  2 sserbin sserbin 4096 Mar 26 08:49 bin
drwxr-xr-x 14 sserbin sserbin 4096 Mar 26 09:20 data
drwxrwxr-x  3 sserbin sserbin   31 Mar 26 08:49 include
drwxrwxr-x  2 sserbin sserbin   60 Mar 26 08:49 lib
drwxrwxr-x  3 sserbin sserbin   31 Mar 26 08:49 share
```

Still failing even if I explicitly tell it where the /data folder is.  I am really perplexed by this as I should have everything where it needs to be yet it still seems to fail finding what it needs.  

You mentioned the "test" folder do you think it may be looking test/data??

pgbrodrick, 2020-07-10:
Based on recent comments in other threads, it seems like @serbinsh you have examples working now.  Is this stale and/or do you have any concluding thoughts to post for future users?

serbinsh, 2020-07-10:
I think this was mostly solved by using LD_LIB paths and the "engine_base_dir" option in the configs.  I think my issue was mostly related to making sure IsoFit could find the lib

```
"radiative_transfer_engines": {
  "engine_name": "libradtran",
  "engine_base_dir": "/path/to/LibRadTran-2.0.3",  # or whatever
  ...
}
```

One other thing that was likely a factor was that IsoFit uses the LibRadTran-x.x.x "/test" folder during execution and when I did "make install" it placed everything in the install dir except /test, not sure why. Anyway the easiest thing I found was to just make the library but not "install" it and point IsoFit to the build dir.

closed 2020-07-10 by serbinsh