isofit/isofit #81: Topographic correction

davidraythompson, 2020-04-02:
In principle, a topographic correction depends on surface reflectance assumptions (Lambertian vs. BRDF) so I think it would make sense to incorporate this functionality in a new surface object.  Should surface slope come in via the geometry() object?

pgbrodrick, 2020-04-03:
The slope could be calculated based on the 3rd band of the IGM, but this lead to issues if the surface resolution is too high, so bringing it in as a new file read probably makes sense?

davidraythompson, 2020-04-19:
I've reviewed the literature and options on this one a bit more. There are elegant solutions to both topographic variability and BRDF that run as postprocessing, e.g. [https://github.com/EnSpec/HyTools-sandbox](url) .  Current techniques rely on case-specific assumptions and scene statistics.  So I propose we define topography and BRDF as outside our jurisdiction for now., and instead concentrate on getting good flat-lambertian-equivalent reflectance inversions from individual spectra.  That's hard enough :)  Departures from the flat Lambertian assumptions can be handled as postprocessing, following current convention.

closed 2020-04-19 by davidraythompson