isofit/isofit #87: Parametric noise files with leading spaces cannot be read

ashiklom, 2020-04-08:
Parametric noise files whose first line looks like this:

```
 #wave, a, b, c, ...
^ Leading space here.
```

...cannot be read because this table reading code...

https://github.com/isofit/isofit/blob/83055d48f55726a56867e74be2a28dabced518cb/isofit/core/instrument.py#L92-L93

...does not recognize that lines with a leading space followed by a `#` should also be skipped.

To my knowledge, `numpy.loadtext` can't take regular expressions for the `comments` argument—otherwise, that would be the ideal fix. Perhaps you could just catch file-reading/parsing errors here and re-throw them with a more informative error message? Currently, the error thrown is a bit cryptic—something about being unable to parse certain fields. Alternatively, maybe just add a note to the documentation about the format of these files, with a brief caution about leading spaces and other characters?

adam-erickson, 2020-04-18:
Just to reiterate, my personal recommendation is to fix the function that is generating the erroneous parametric noise files, as ISOFIT cannot handle every edge case and structured text formats are intentionally strict. What ISOFIT can do though is provide a parametric noise file generator importable by end-users, though that may not address this issue in the short-term. Why not write a simple temporary function to pre-process the noise files on the user end?

ashiklom, 2020-05-14:
I'm gonna go ahead and close this because it probably falls more under "user error" than something libradtran should handle.

closed 2020-05-14 by ashiklom