isofit/isofit #108: ?Multiple wavelength grids

davidraythompson, 2020-05-19:
So when I originally wrote the first version of this code, I made each component (surface, radiative transfer, and instrument) use its own wavelength grid.  The idea was to permit modularity, e.g. so that one surface model could be used for many different instrument calibrations, or for RTM calculations could be performed at high spectral resolution and then downsampled.  It would also permit instrument wavelength calibration to be floated as a free parameter.  However, we seem to have lost this functionality along the way; when I actually try to use different reflectance wavelength grids  there are errors in file I/O.   

It seems like we should either go "all in" with separate grids, and restore the original modularity, or go "all out" and force the user to rebuild surface models for every flightline using a constant wavelength grid.  The latter might cut out complexity, which  is my primary objective right now.

pgbrodrick, 2020-05-19:
It was likely me that lost some of this functionality during #92 - specifically, I dropped the requirement of the RTM wavelengths, since in all examples I could find we used the same wavelengths as the instrument.  However, we can easily restore that, as the config parameter was lost but the infrastructure is still there.  Is this the IO error you're getting, or is there another?

I'm torn here in general - I'm with you on general complexity decrease, but I also see the utility of allowing the greater flexibility of multiple wavelengths.  We do have the infrastructure to support both operating modes in place though.  In general, my thought process at such a junction is to set defaults (now possible) and examples to orient towards the simplified case, but leave the option open for the more complicated - though we need tests in play in order to do that, so that I (or someone else) doesn't accidentally drop out functionality again.

Thoughts?

davidraythompson, 2020-05-19:
Yeah, now I'm questioning whether the flexibility of multiple grids is worth the complexity.    Imagine how simple life would be if we enforce a single, uniform wavelength grid everywhere, and the 'wl' variable means the same thing in every instance....  many of the resampling operations would go away also.

pgbrodrick, 2020-05-19:
According to @jfahlen , some of the resampling operations can also be rather expensive to do on-the-fly.

I can't see much utility in not redoing the surface model for each case, given the relative ease and speed of doing so.  Which mostly leaves alignment with the RTMs.  What if we nixed the internal alignment, with the thought that alignment could be handled with a utility (external or in utils) that  is run once a priori?  Probably where your head's at already.

Simulation is the only case that I don't know much about - would this hurt that?  Otherwise, I agree that it would be nice to drop out the extra config parameters (already part-done, as noted), along with the not insubstantial amount of code that checks the grids.  We could put it on the 3.0 project board, it's quite tractable.  Might also make IO easier, which is another (separate) hurdle.

davidraythompson, 2020-05-19:
> What if we nixed the internal [resampling], with the thought that 
> alignment could be handled with a utility (external or in utils) that is run once a priori?

Agreed, we can push the complexity to supporting utilities.  

> We could put it on the 3.0 project board, it's quite tractable. 

Will do.

evan-greenbrg, 2026-09-08:
@pgbrodrick @davidraythompson If either of you has a chance, please confirm that this can be depreciated. We still support different wavelength grids.

pgbrodrick, 2026-09-08:
Yes, this has been substantially overtaken by events - we continue to have multiple wavelength grids, but now include efficient and well thought out mechanisms to handle those interfaces, thanks to significant effort and many PRs.  Closing!

closed 2026-09-08 by pgbrodrick