isofit/isofit #109: modtran h2o-check fails on some systems

pgbrodrick, 2020-05-21:
As reported by @NimrodCarmon , on some systems the 10s time limit of the modtran run to check the H2O upper limit leads to a corrupted tp6 file, which then can't be read.

Possible fixes: wrap line reads in try catch, and hope the line we want is clean.

swap out "BMNAME": "p1_2013" for "BMNAME": "05_2013", which runs at coarse spectral resolution, and check results.  A quick test on a real setup has that throwing an error, but after where we care about for this test, e.g.:

##############################################################
Warning from routine SCLCOL:  Input water column,  50.00000 gm/cm2, is above maximum allowed.
                              The water column is being set to the maximum,   4.59827 gm/cm2.
Errors were encountered during runtime.  Please check the configuration.
Errors for case 0 :
------------------------------------------------------------
Error in routine CHKRES:  Specified spectral resolution (FWHM on CARD4 is too fine for the chosen band model.
                          It must exceed     3.17520 Nanometers.
driver_mod/CD4
driver_mod/DRIVER
    1    1 (Root name and run number)
------------------------------------------------------------
##############################################################

This should be fine in general, but with a little more testing, this could get easily swapped out into the code, and would be a bit more elegant.

NimrodCarmon, 2020-05-22:
With more careful look we found out that the UTC band in the OBS file in some cases might have different time format than expected, which led modtran to crash before the 10s and the tp6 file from that run couldn't be read. Changing the format of "GMTIME" in the .json input to modtran stopped modtran from crashing.

pgbrodrick, 2020-05-27:
Resolved by @NimrodCarmon in #112.

closed 2020-05-27 by pgbrodrick