isofit/isofit #114: How to forward-model an ENVI image?

ashiklom, 2020-06-08:
It used to be possible to run a forward model (i.e. given a known surface reflectance and atmosphere, simulate the TOA radiance) on an ENVI binary image file by setting the `"input": "reflectance_file"` configuration. However, after the many recent refactors, it seems that `input.reflectance_file` is effectively superceded by `forward_model.surface.reflectance_file`.

When I try to run with a config like the following (`reference_reflectance` has 15 different pixels), I get an array of 15 identical TOA radiances. And when I subsequently run the inversion on those radiances, I get 15 flat reflectances, exactly matching the `constant-refl.txt`. This suggests that the dimensions from `reference_reflectance` are read in, but the reflectance is entirely superceded by `reference_reflectance`. Omitting `forward_model.surface.reflectance_file` or even `forward_model.surface` altogether results in an error.

```json
{
    "forward_model": {
        "instrument": {
            "SNR": 300,
            "integrations": 1,
            "wavelength_file": "data/wavelengths.txt"
        },
        "radiative_transfer": {
            "lut_grid": {
                "AOT550": [0.001, 0.123, 0.6],
                "H2OSTR": [2.216, 2.716, 3.216]
            },
            "radiative_transfer_engines": {
                "vswir": {
                    "engine_name": "libradtran",
                    "lut_names": [
                        "H2OSTR",
                        "AOT550"
                    ],
                    "lut_path": "./lut",
                    "statevector_names": [
                        "H2OSTR",
                        "AOT550"
                    ],
                    "template_file": "lut/lrt_template.inp",
                    "wavelength_file": "data/wavelengths.txt"
                }
            },
            "statevector": {
                "AOT550": {
                    "bounds": [0.001, 0.6],
                    "init": 0.123,
                    "prior_mean": 0.123,
                    "prior_sigma": 1.0,
                    "scale": 0.1
                },
                "H2OSTR": {
                    "bounds": [2.216, 3.216],
                    "init": 2.716,
                    "prior_mean": 2.716,
                    "prior_sigma": 5.0,
                    "scale": 0.1
                }
            },
            "unknowns": {
                "H2O_ABSCO": 0.01
            }
        },
        "surface": {
            "reflectance_file": "data/constant-refl.txt",
            "surface_category": "surface"
        }
    },
    "implementation": {
        "inversion": {
            "simulation_mode": true,
            "windows": [
                [400, 1300],
                [1450, 1780],
                [1950, 2450]
            ]
        }
    },
    "input": {
        "reflectance_file": "data/reference_reflectance"
    },
    "output": {
        "simulated_measurement_file": "output/simulated_toa_radiance"
    }
}
```

Passing an image to `forward_model.surface.reflectance_file` fails because that config option is read by `load_spectrum` (`core/common.py`), which can only read plan text files.

I can run this workflow just fine by omitting `input.reference_reflectance` altogether and just passing one reflectance pixel at a time to `forward_model.surface.reflectance_file`, but that quickly becomes challenging for real images with hundreds to thousands of pixels.

Is there a right way to forward-model image reflectance files? Or has this feature been deliberately dropped from `isofit`?

pgbrodrick, 2020-06-08:
Hi @ashiklom,

Thanks for reporting the issue here.  This is a known issue that I've failed to post about, and is in the midst of being resolved.  As you indicate, I had to rework a bit of the IO to handle the reference_reflectance properly, but that should be done at https://github.com/pgbrodrick/isofit/tree/simulation_fix .  Some colleagues are running some tests before I push a PR, but you can check that out now (I'll try to get the PR up today though).

Under that branch, the correct behavior should be to specify the input reflectance_file, but *not* the reflectance inside of the surface config (as that should be reserved for references to priors).

ashiklom, 2020-06-08:
@pgbrodrick Thanks for the quick reply! Really nice to work with such a great dev team 😄 

I'll check out this branch, see if I can get it to do what I want, and report back.

ashiklom, 2020-06-08:
OK, reporting back: That branch resolves the issue! For documentation purposes, I had to make the following changes to get the forward run to work:

- I had to add the `forward_model.surface.wavelength_file` with the wavelengths for which I wanted to do the simulation.
- I had to specify `implementation.mode = "simulation"` (this used to be `implementation.inversion.simulation_mode = true`).

However, now I'm not sure how to configure the `surface` to perform an inversion. Running with `forward_model.surface.reflectance_file` set to a constant surface reflectance causes the inversion to return that exact surface reflectance (in my case, flat reflectance). Running with no `forward_model.surface.reflectance_file` (e.g. only the `forward_model.surface.wavelength_file` set) returns the error `AttributeError: 'Surface' object has no attribute 'rfl'`. Here's the full traceback:

```
multiprocessing.pool.RemoteTraceback:
"""
Traceback (most recent call last):
  File "/Users/ashiklom/opt/anaconda3/envs/isofit/lib/python3.7/multiprocessing/pool.py", line 121, in worker
    result = (True, func(*args, **kwds))
  File "/Users/ashiklom/projects/sbg-uncertainty/isofit/isofit/core/isofit.py", line 119, in _run_set_of_spectra
    self.states = self.iv.invert(meas, geom)
  File "/Users/ashiklom/projects/sbg-uncertainty/isofit/isofit/inversion/inverse.py", line 232, in invert
    Seps_inv, Seps_inv_sqrt = self.calc_Seps(x, meas, geom)
  File "/Users/ashiklom/projects/sbg-uncertainty/isofit/isofit/inversion/inverse.py", line 164, in calc_Seps
    Seps = self.fm.Seps(x, meas, geom)
  File "/Users/ashiklom/projects/sbg-uncertainty/isofit/isofit/core/forward.py", line 206, in Seps
    Kb = self.Kb(x, geom)
  File "/Users/ashiklom/projects/sbg-uncertainty/isofit/isofit/core/forward.py", line 262, in Kb
    rfl = self.surface.calc_rfl(x_surface, geom)
  File "/Users/ashiklom/projects/sbg-uncertainty/isofit/isofit/surface/surface.py", line 90, in calc_rfl
    return self.rfl
AttributeError: 'Surface' object has no attribute 'rfl'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/ashiklom/opt/anaconda3/envs/isofit/bin/isofit", line 7, in <module>
    exec(compile(f.read(), __file__, 'exec'))
  File "/Users/ashiklom/projects/sbg-uncertainty/isofit/bin/isofit", line 54, in <module>
    main()
  File "/Users/ashiklom/projects/sbg-uncertainty/isofit/bin/isofit", line 49, in main
    return isofit.run()
  File "/Users/ashiklom/projects/sbg-uncertainty/isofit/isofit/core/isofit.py", line 164, in run
    results = [p.get() for p in results]
  File "/Users/ashiklom/projects/sbg-uncertainty/isofit/isofit/core/isofit.py", line 164, in <listcomp>
    results = [p.get() for p in results]
  File "/Users/ashiklom/opt/anaconda3/envs/isofit/lib/python3.7/multiprocessing/pool.py", line 657, in get
    raise self._value
AttributeError: 'Surface' object has no attribute 'rfl'
```

Is that the expected behavior?

pgbrodrick, 2020-06-08:
Thanks for reporting back, and documenting the required config parameters used.

The above is the expected behavior for the config you specified, though this highlights that the config checks should be updated to provide better feedback (duly noted, and I've added it to my list of config-checks to update in the documentation improvements I'm working up this week).

In order to run the inversion, you'll need to specify some form of surface reflectance priors, which can be provided in one of many ways.  Currently, we support providing a single reflectance value via either the surface->reflectance_file parameter or the surface->reflectance parameters (former for a file, latter for an actually array).

Or, you can use the surface_model.py file in the utils folder to generate a collection of priors based on a the radiance cube you generate for use with the multicomponent_surface style of the surface.  This is a common execution method, and you can see demos of it in several of the examples - e.g. https://github.com/isofit/isofit/blob/master/examples/20171108_Pasadena/run_example_libradtran.py

Is there a behavior different from above that you would want/expect?

ashiklom, 2020-06-08:
Thanks! That's all good to know. In general, I think the behavior you describe above makes a lot of sense! I like the ability to control the surface reflectance prior, including the ability to create a multicomponent prior if desired.

I think the main thing I'm after here is a way to specify an uninformative or weakly informative prior that allows the inversion to be primarily driven by the data (e.g. like what you would use for a large-scale mission where you only have a vague sense of the surfaces you'll be looking at). It seems like the default assumptions associated with a flat, plain-text prior give it way too much weight: The estimated surface reflectance spectra I'm getting are all identical to the prior and retain none of the variability in the radiance (see image).

[image]

Is there something else I'm setting incorrectly here?

If it's useful, I also noticed that the Isofit residuals seem astronomically high (see below). This makes me think that the algorithm is correctly trying to optimize against the observed radiance, but is stuck in a local minima caused by the excessively weighted prior.

```
Iteration: 03  Residual:    831135.41  Atmosphere: AOT550: 0.001 H2OSTR: 2.216
Iteration: 03  Residual:    956209.41  Atmosphere: AOT550: 0.599 H2OSTR: 2.216
Iteration: 03  Residual:    567457.21  Atmosphere: AOT550: 0.002 H2OSTR: 2.216
Iteration: 03  Residual:    818220.92  Atmosphere: AOT550: 0.574 H2OSTR: 2.216
Iteration: 03  Residual:   1042280.63  Atmosphere: AOT550: 0.600 H2OSTR: 2.216
Iteration: 03  Residual:    567457.21  Atmosphere: AOT550: 0.002 H2OSTR: 2.216
Iteration: 03  Residual:   1016161.23  Atmosphere: AOT550: 0.598 H2OSTR: 2.216
Iteration: 04  Residual:    831106.62  Atmosphere: AOT550: 0.001 H2OSTR: 2.216
Iteration: 04  Residual:    956182.28  Atmosphere: AOT550: 0.600 H2OSTR: 2.216
Iteration: 04  Residual:    567381.21  Atmosphere: AOT550: 0.001 H2OSTR: 2.216
Iteration: 04  Residual:    817370.72  Atmosphere: AOT550: 0.599 H2OSTR: 2.216
Iteration: 04  Residual:   1042260.75  Atmosphere: AOT550: 0.600 H2OSTR: 2.216
Iteration: 04  Residual:   1016074.20  Atmosphere: AOT550: 0.600 H2OSTR: 2.216
Iteration: 04  Residual:    567381.21  Atmosphere: AOT550: 0.001 H2OSTR: 2.216
```

<details>
<summary>Config (click to expand)</summary>

```json
{
    "forward_model": {
        "instrument": {
            "SNR": 300,
            "integrations": 1,
            "wavelength_file": "../data/wavelengths.txt"
        },
        "radiative_transfer": {
            "lut_grid": {
                "AOT550": [0.001, 0.123, 0.6],
                "H2OSTR": [2.216, 2.716, 3.216]
            },
            "radiative_transfer_engines": {
                "vswir": {
                    "engine_name": "libradtran",
                    "lut_names": [
                        "H2OSTR",
                        "AOT550"
                    ],
                    "lut_path": "./lut",
                    "statevector_names": [
                        "H2OSTR",
                        "AOT550"
                    ],
                    "template_file": "./lut/lrt_template.inp",
                    "wavelength_file": "../data/wavelengths.txt"
                }
            },
            "statevector": {
                "AOT550": {
                    "bounds": [0.001, 0.6],
                    "init": 0.123,
                    "prior_mean": 0.123,
                    "prior_sigma": 1.0,
                    "scale": 0.1
                },
                "H2OSTR": {
                    "bounds": [2.216, 3.216],
                    "init": 2.716,
                    "prior_mean": 2.716,
                    "prior_sigma": 5.0,
                    "scale": 0.1
                }
            },
            "unknowns": {
                "H2O_ABSCO": 0.01
            }
        },
        "surface": {
            "surface_category": "surface",
            "reflectance_file": "../data/constant-refl-10nm.txt"
        }
    },
    "implementation": {
        "mode": "inversion",
        "inversion": {
            "windows": [
                [400, 1300],
                [1450, 1780],
                [1950, 2450]
            ]
        }
    },
    "input": {
        "measured_radiance_file": "./output/simulated_toa_radiance"
    },
    "output": {
        "estimated_reflectance_file": "./output/estimated_reflectance",
        "estimated_state_file": "./output/estimated_state"
    }
}
```

</details>

davidraythompson, 2020-06-08:
Hi Alexey,

I don't think this is an issue with the prior, because it appears that, based on your configuration file, there is no prior :)  For the simple "surface" object, the surface reflectance is held to your reflectance file and not treated as a free parameter in the retrieval. I'd suggest defining a multicomponent surface object following the example in the 20171108_Pasadena subdirectory.  

Alternatively, you might consider looking at tutorial modules 11-14 at:
https://github.com/davidraythompson/istutor

Tutorial #14 is all about defining priors on surface reflectances.   We've recently published a manuscript - https://authors.elsevier.com/a/1b9IP7qzSr6nU (link active until mid-July) - which describes a very unconstrained surface prior for a general purpose L2 product.

I'm happy to field followup questions on any of this!

davidraythompson, 2020-06-09:
I believe this is now fixed in latest merge.   Thanks @pgbrodrick!  
I'll wait to close, though, until we hear for sure.

ashiklom, 2020-06-09:
Thanks @davidraythompson! Those resources look very useful, and are exactly what I was looking for. I'll work through them and let you know if I have questions.

And thanks @pgbrodrick for the fix!

closed 2020-06-09 by davidraythompson