isofit/isofit #120: MODTRAN h2o upper bound on broad LUT grid

pgbrodrick, 2020-06-15:
Our current handling of the H2O upper bound is detrimentally prohibitive in the case of scenarios where another LUT dimensions changes the H2O bound substantially.  Take for example the case where you have an elevation gradient that runs from 1-3 km.  At the top (3km), the max water value is 3 g/cm2, at the bottom, it's 5 g/cm2 (I'm making up pseudo-realistic values here).  The LUT table would be confined to:

elevation: [1,2,3]
h2o: [0.5,1,2,3]

because an h2o value of 4 would violate **some** of the elevation grid, and we're using a RegularGrid interpolator.  But if we accept that, than any (very reasonable) high water condition at low altitude is outside of our LUT.

There are a few potential fixes available:

1) Loosen the H2O upper bound constraint, calculate the per-pixel max h2o value, and adjust the upper bound.
2) Loosen the H2O upper bound constraint, and induce a (mild) slope for water values that are too high, so that the Jacobian is not 0.

Neither of these solutions is perfect.  The first requires (lots) of h2o-max value checking, basically per pixel.  That's prohibitively expensive, and even if checks were limited to something like one per elevation band (though there are other LUT dimensions that could pose issues), this could lead to failure cases.

The second has the drawback of a) excessive MODTRAN runs and b) effective (if not actual) induced degeneracy in the atmospheric solution state.  b) is not important if our goal is surface reflectance exclusively, but could be substantial for efforts to estimate the atmospheric state.

Other potential solutions include replacing the meshgrid interpreter with a linear interpolator, but then we risk interpolating through highly-unknown portions of the state space.  Ultimately, I'll probably implement 2) as the least-bad solution, but posting here for a while in case anyone else has a better idea!

Note - this seems like a bit of a corner-case discussion until trying to implement isofit on any line that spans any significant elevation gradient, which is not particularly uncommon.

davidraythompson, 2020-06-15:
Two other ideas spring to mind.  One would be to treat all H2O parameters as scaling factors, and use the MODTRAN H2OUNIT="+" option to exceed 100% RH.  

The second would be to turn up the atmospheric temperature to "Venus."  But that would mess with the TIR, and seems a bit TOO kludgey.  

Of these, I think I still prefer your #2 if it is easy to implement.

pgbrodrick, 2020-06-16:
Good thoughts, thanks!

Exceeding 100% RH may be the way to go ultimately, particularly given that in several examples the isofit resolved H2O solution is close to the upper bound (despite the conditions being highly unlikely to be near 100% RH).

I'm in agreement that Venus might cause more problems than it's worth :-).  Will give a bit more thought / time for review, and then pick a patch.

brentwilder, 2026-08-25:
This is now addressed by using derived MODTRAN-based polynomials (https://github.com/isofit/isofit/pull/689) to do a per-pixel check prior to OE based on the elevation (https://github.com/isofit/isofit/pull/982 , https://github.com/isofit/isofit/pull/976). It thereby addresses #198 as well.

closed 2026-08-25 by brentwilder