isofit/isofit #126: Use yaml for the input file specification instead of json

jfahlen, 2020-06-30:
I don't like the finicky json format: you can't comment a line or add a comment, and the curly braces are out of control. 

Yaml allows comments, uses tabs instead of curly braces (like python!), and, conveniently, can parse json files so there's no need to force people to change all their old input files.

As an example, here's the Pasadena darklot config converted to yaml (I added a few nonsense comments as examples in the first input section):

```
ISOFIT_BASE: "../../.."
input:
  # List input files
  measured_radiance_file: "../remote/ang20171108t184829_rdn_v2p11_darklot.txt" # Here's an input file
  reference_reflectance_file: "../insitu/DarkTarget_Trial1.txt"
output:
  estimated_reflectance_file: "../output/ang20171108t184829_rfl_v2p11_darklot.txt"
  algebraic_inverse_file: "../output/ang20171108t184829_alg_v2p11_darklot.txt"
  modeled_radiance_file: "../output/ang20171108t184829_mdl_v2p11_darklot.txt"
  data_dump_file: "../output/ang20171108t184829_data_v2p11_darklot.mat"
  posterior_errors_file: "../output/ang20171108t184829_post_v2p11_darklot.mat"
  plot_directory: "../images/"
forward_model:
  instrument:
    wavelength_file: "../remote/20170320_ang20170228_wavelength_fit.txt"
    parametric_noise_file: "../../../data/avirisng_noise.txt"
    integrations: 10
    unknowns:
      channelized_radiometric_uncertainty_file: "../../../data/avirisng_systematic_error.txt"
      uncorrelated_radiometric_uncertainty: 0.01
  surface:
    surface_category: multicomponent_surface
    surface_file: "../remote/ang20170228_surface_model.mat"
  radiative_transfer:
    statevector:
      H2OSTR:
        bounds: [1.5, 2]
        scale: 0.01
        prior_mean: 1.75
        prior_sigma: 0.5
        init: 1.75
      AOT550:
        bounds: [0.01, 0.1]
        scale: 0.01
        prior_mean: 0.05
        prior_sigma: 0.2
        init: 0.05
    lut_grid:
      H2OSTR: [1.5, 2]
      AOT550: [0.01, 0.1]
    unknowns:
      H2O_ABSCO: 0.01
    radiative_transfer_engines:
      vswir:
        engine_name: modtran
        wavelength_range: [370, 2505]
        lut_path: "../lut/"
        template_file: ang20171108t184227_modtran.json
        lut_names: [H2OSTR, AOT550]
implementation:
  inversion:
    windows: [[380, 1300], [1450, 1780], [1950, 2450]]
```

The can be read with:
```
f = open(filename, 'r')
y = yaml.load(f, Loader = yaml.FullLoader)
```

`y` will be a dict just like the json output. Conveniently, the same code can also read and parse the original json.

I haven't scoped out the code change, but I guess is that it will be pretty simple.

Thoughts?

pgbrodrick, 2020-06-30:
I agree, this is a good idea!  I don't want to discontinue the use of JSON though, both because of legacy and because there's no real need to, but I'm happy to add in the yaml support.

In fact, I'm so supportive of this that I already did when I updated the configs :-).  Check out create_new_config call in isofit/configs/configs.py.

pyyaml is also already enforced in the setup.py.

That said, I haven't actually tested a yaml config out yet, and historically I've found small issues with what it casts objects in during the read.....so if you run across any issues, respond here and I'm happy to take a look.

I'll leave it to you @jfahlen to close this out if your tests succeed.

pgbrodrick, 2020-07-06:
PR #127 accepted, extending the included via safe loading.

closed 2020-07-06 by pgbrodrick