isofit/isofit #128: "Assignment destination is read-only" in instrument.py - Sy

ashiklom, 2020-07-06:
Trying to perform an inversion interactively (following along with `istutor`, but with some different values) produces the following error when running the inversion (`iv.invert`):

```
Error in py_call_impl(callable, dots$args, dots$keywords) : 
  ValueError: assignment destination is read-only

Detailed traceback: 
  File "/Users/ashiklom/opt/anaconda3/envs/r-isofit/lib/python3.8/site-packages/isofit/inversion/inverse.py", line 328, in invert
    Seps_inv, Seps_inv_sqrt = self.calc_Seps(x, meas, geom)
  File "/Users/ashiklom/opt/anaconda3/envs/r-isofit/lib/python3.8/site-packages/isofit/inversion/inverse.py", line 181, in calc_Seps
    Seps = self.fm.Seps(x, meas, geom)
  File "/Users/ashiklom/opt/anaconda3/envs/r-isofit/lib/python3.8/site-packages/isofit/core/forward.py", line 208, in Seps
    Sy = self.instrument.Sy(meas, geom)
  File "/Users/ashiklom/opt/anaconda3/envs/r-isofit/lib/python3.8/site-packages/isofit/core/instrument.py", line 177, in Sy
    meas[bad] = 1e-5
```

As far as I can tell, this is because [Python passes objects by reference](https://stackoverflow.com/questions/845110/emulating-pass-by-value-behaviour-in-python) rather than by value by default, and [this line in instrument.Py - Sy](https://github.com/isofit/isofit/blob/499da437c3b021654e82ca3a8166d32891416977/isofit/core/instrument.py#L177) tries to modify a read-only function argument.

I've been able to work around this error by dropping the low radiance values manually prior to calling the inversion. I don't know Python well enough to authoritatively suggest a proper solution, but I _think_ the most straightforward solution is just to create a (shallow?) copy of `meas` early on in the function (e.g. `meas2 = meas.copy()`) and then use that copy throughout the rest of the function.

Note: I hit this error under a somewhat non-standard configuration: Python 3.8.3, Numpy v.1.18.5, and, most importantly, calling Python from R via [`reticulate`](https://rstudio.github.io/reticulate/). There may be some additional memory-protection stuff going on in how reticulate passes objects around that causes this error. I haven't tested it in pure Python yet, so mine might be a special case.

pgbrodrick, 2020-07-06:
Hey Alexey, thanks for the detailed message.

I don't think the issue is inside of instrument.py, as we do actually want the referenced value to get modified.  Rather, the read in measurement value must still be a numpy memmap or the like, rather than a full array, and opened via read-only.

Can you post your full config (I only need the input, but the whole is fine).  My guess is that you're using a measured_radiance_file without a radiometry_correction_file, since looking through the .copy() is on the wrong line on L389 of [fileio](https://github.com/isofit/isofit/blob/master/isofit/core/fileio.py#L389)....but worth confirming this is the issue (as your hunch about Python through reticulate may be an issue as well).  Anyway, if you show your input config I can confirm and post a patch.

Thanks for the catch!

ashiklom, 2020-07-06:
Thanks for the quick reply Phil! You're right -- I don't have a `radiometry_correction_file` set, so that may well be an issue.

I don't actually have an `input` config to show you because I'm mostly following David's [tutorial from `istutor`](https://github.com/davidraythompson/istutor/blob/master/12_OptimalEstimationB/12_OE_Inverse.ipynb) (12_OptimalEstimationB) where all of the inputs are passed around in memory. The best I can do on short notice is provide my R code -- it's not beautiful, but hopefully it will give you a sense of the configuration:

```r
library(reticulate)
library(here)
library(fs)
isofit <- import("isofit")
ray <- import("ray")
isofit_forward <- import("isofit.core.forward")
isofit_configs <- import("isofit.configs.configs")
isofit_inverse <- import("isofit.inversion.inverse")
Config <- isofit_configs$Config
Inversion <- isofit_inverse$Inversion
ForwardModel <- isofit_forward$ForwardModel

wavelength_file <- here("data/wavelengths.txt")
waves_df <- read.table(wavelength_file, col.names = c("id", "wl_um", "fwhm"))
waves_nm <- waves_df$wl_um * 1000
lrt_dir <- "~/projects/models/libradtran/libRadtran-2.0.3/"

instrument_settings <- dict(
  wavelength_file = wavelength_file,
  SNR = 300
)

rtm_settings <- dict(
  lut_grid = dict(AOT550 = c(0.001, 0.123, 0.6),
                  H2OSTR = c(2.216, 2.716, 3.216)),
  radiative_transfer_engines = dict(
    vswir = dict(
      engine_name = "libradtran",
      engine_base_dir = lrt_dir,
      lut_names = c("H2OSTR", "AOT550"),
      lut_path = dir_create(here("r-interface", "output", "lut")),
      statevector_names = c("H2OSTR", "AOT550"),
      template_file = here("test-isofit", "lut", "lrt_template.inp"),
      wavelength_file = wavelength_file
    )
  ),
  statevector = dict(
    AOT550 = dict(
      bounds = c(0.001, 0.6),
      init = 0.123,
      prior_mean = 0.123,
      prior_sigma = 1.0,
      scale = 0.1
    ),
    H2OSTR = dict(
      bounds = c(2.216, 3.216),
      init = 2.716,
      prior_mean = 2.716,
      prior_sigma = 5.0,
      scale = 0.1
    )
  ),
  unknowns = dict(H2O_ABSCO = 0.01)
)

surface_settings <- dict(
  surface_category = "multicomponent_surface",
  wavelength_file = wavelength_file,
  surface_file = here("data/uninformative-prior.mat")
)

fm_config <- Config(dict(
  forward_model = dict(instrument = instrument_settings,
                       surface = surface_settings,
                       radiative_transfer = rtm_settings)
))

if (!ray$is_initialized()) ray_server <- ray$init()
fm <- ForwardModel(fm_config)

isofit_geometry <- import("isofit.core.geometry")
geom <- isofit_geometry$Geometry()

# Calculate radiance from reflectance, using PRO4SAIL
true_refl <- rrtm::pro4sail_4(1.4, 40, 0.01, 0.01, 3, 0.5)$bdr
true_h20 <- 2.716
true_aot <- 0.123
obsrad <- fm$calc_rdn(np_array(c(true_refl, true_h20, true_aot)), geom)

inv_windows <- list(c(410, 1200), c(1500, 1700), c(2050, 2300))
iwindows <-
  (waves_nm >= inv_windows[[1]][1] & waves_nm <= inv_windows[[1]][2]) |
  (waves_nm >= inv_windows[[2]][1] & waves_nm <= inv_windows[[2]][2]) |
  (waves_nm >= inv_windows[[3]][1] & waves_nm <= inv_windows[[3]][2])
inversion_settings <- dict(implementation = dict(
  mode = "inversion",
  inversion = dict(
    windows = do.call(rbind, inv_windows)
  )
))
inverse_config <- Config(inversion_settings)
iv <- Inversion(inverse_config, fm)
state_trajectory <- iv$invert(obsrad, geom)
```

pgbrodrick, 2020-07-06:
Hi Alexey,

Thanks for this - in-memory is just as good.  Unfortunately though, the error ID'd above, though still probably needs addressing, is not the cause of your issue, since you're not actually running your calculations through the fileio, but rather calling the inversion directly.

Ultimately, the error here looks like it's a result of reticulate here, but we can throw in some safeguards anyway (also addressing the copy I found above as a result).

@ashiklom , can you check if PR #129 resolves the problem?

ashiklom, 2020-07-07:
Sorry for the delay. Yes, I just tested that PR and I think it resolved my issue, thanks!

closed 2020-07-07 by pgbrodrick