isofit/isofit #130: New ray requirement not supported on older operating systems

serbinsh, 2020-07-07:
Yesterday I was working on getting a new IsoFit conda env running on my BNL RHEL 6.10 linux cluster.  Yes its an older LTS version and has been on the list of machines to update, but is used a fair amount and then COVID....so it may be awhile now.

I learned yesterday that ray needs glibc >=2.14 (or so) but RHEL 6.10 has glibc 2.12.  I built two new versions (2.18 and 2.19) yesterday but they both seem to crash my module system and all around created some strange behavior.  It wasnt clear if I could easily run either of those when using IsoFit conda.  It wasnt wasn't clear how to point ray to a build folder that it can link with without actually adding it to my LD_LIBRARY_PATH.

The original error

```
(/data2/sserbin/conda_envs/isofit) [sserbin@modex 20171108_Pasadena]$ python
Python 3.8.3 (default, Jul  2 2020, 16:21:59)
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ray
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data2/sserbin/conda_envs/isofit/lib/python3.8/site-packages/ray/__init__.py", line 48, in <module>
    CDLL(so_path, ctypes.RTLD_GLOBAL)
  File "/data2/sserbin/conda_envs/isofit/lib/python3.8/ctypes/__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /lib64/libc.so.6: version `GLIBC_2.14' not found (required by /data2/sserbin/conda_envs/isofit/lib/python3.8/site-packages/ray/_raylet.so)
>>>
```

I tried one conda glibc https://anaconda.org/asmeurer/glibc but that broke conda for me

```
[sserbin@modex ~]$ [sserbin@modex ~]$ conda
-bash: [sserbin@modex: command not found
[sserbin@modex ~]$ Segmentation fault
-bash: Segmentation: command not found
[sserbin@modex ~]$ conda
Segmentation fault
```

My current work around is that @ashiklom created a ray-optional branch for testing the dependency on ray here https://github.com/ashiklom/isofit/tree/optional-ray.  I was running a test yesterday with this version which made it farther, including running of LibRadTran, but seems to have crashed during the inversion step. This may be unrelated to the ray issue

```
Configuration check: inversion_grid_as_preseed is not an object, skipping
Configuration check: n_cores is not an object, skipping
Configuration check: ip_head is not an object, skipping
Configuration check: redis_password is not an object, skipping
Configuration check: rte_configure_and_exit is not an object, skipping
Configuration check: rte_auto_rebuild is not an object, skipping
Configuration check: ray_temp_dir is not an object, skipping
Configuration file checks complete, no errors found.
Rebuilding radiative transfer look up table
Inferred ASCII file format for /data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/remote/ang20171108t184227_rdn_v2p11_BeckmanLawn.txt
Inferred ASCII file format for /data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/insitu/BeckmanLawn.txt
Inferred ASCII file format for /data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/output/ang20171108t184227_alg_v2p11_BeckmanLawn-libRadTran.txt
Inferred ASCII file format for /data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/output/ang20171108t184227_rfl_v2p11_BeckmanLawn-libRadTran.txt
Inferred ASCII file format for /data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/output/ang20171108t184227_mdl_v2p11_BeckmanLawn-libRadTran.txt
Beginning inversions using 1 cores
bash /data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-1.5000.sh
bash /data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-1.5000.sh
bash /data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.0100_H2OSTR-2.0000.sh
bash /data2/sserbin/NASA_SBG/isofit/examples/20171108_Pasadena/configs/../lut/LUT_AOT550-0.1000_H2OSTR-2.0000.sh
Traceback (most recent call last):
  File "/data2/sserbin/conda_envs/isofit/bin/isofit", line 54, in <module>
    main()
  File "/data2/sserbin/conda_envs/isofit/bin/isofit", line 49, in main
    return isofit.run()
  File "/data2/sserbin/conda_envs/isofit/lib/python3.8/site-packages/isofit/core/isofit.py", line 175, in run
    results = [self._run_set_of_spectra(self, index_sets[l], index_sets[l + 1])
  File "/data2/sserbin/conda_envs/isofit/lib/python3.8/site-packages/isofit/core/isofit.py", line 175, in <listcomp>
    results = [self._run_set_of_spectra(self, index_sets[l], index_sets[l + 1])
TypeError: _run_set_of_spectra() takes 3 positional arguments but 4 were given
```

This was the 20171108_Pasadena example.

So my question is how ray is being used for parallelization and whether it would be possible to make the ray lib dependency optional in situations like this?  Then users could use another approach to parallelize the LUT creation, etc.

As for why that example crashed, I will need to check why there were too many arguments

serbinsh, 2020-07-07:
Also FYI - I tried temporarily linking the newer glibc and installing ray so it would find the new lib dir

e.g. 

```
export LD_LIBRARY_PATH="/data/software/glibc/2.19/lib${LD_LIBRARY_PATH:+:$LD_LIBRARY_PATH}"
```
and then tried to install ray - but not surprisingly I get a seg fault 

```
(/data2/sserbin/conda_envs/ray-isofit) [sserbin@modex 2.19]$ export LD_LIBRARY_PATH="/data/software/glibc/2.19/lib${LD_LIBRARY_PATH:+:$LD_LIBRARY_PATH}"
(/data2/sserbin/conda_envs/ray-isofit) [sserbin@modex 2.19]$ $LD_LIBRARY_PATH
-bash: /data/software/glibc/2.19/lib:/data/software/maui/3.3.1/lib:/data/software/jags/4.3.0/lib64:/data/software/libxml2/2.9.7/lib:/data/software/gdal/2.4.2/lib:/data/software/geos/3.6.3/lib:/data/software/libkml/devel/lib:/data/software/geotiff/1.4.2/lib:/data/software/libtiff/4.0.8/lib:/data/software/proj/5.2.0/lib:/data/software/udunits/2.2.25/lib:/data/software/python/3.6.2/lib:/data/software/git/2.10.0-rc0/lib64:/data/software/perl/5.22.1/lib:/data/software/json-c/main/lib:/data/software/gcc/gcc-5.4.0/share:/data/software/gcc/gcc-5.4.0/lto-plugin:/data/software/gcc/gcc-5.4.0/zlib:/data/software/gcc/gcc-5.4.0/libiberty:/data/software/gcc/gcc-5.4.0/libexec:/data/software/gcc/gcc-5.4.0/libdecnumber:/data/software/gcc/gcc-5.4.0/libcpp:/data/software/gcc/gcc-5.4.0/libbacktrace:/data/software/gcc/gcc-5.4.0/lib64:/data/software/gcc/gcc-5.4.0/lib:/data/software/gcc/gcc-5.4.0/x86_64-unknown-linux-gnu: No such file or directory
(/data2/sserbin/conda_envs/ray-isofit) [sserbin@modex 2.19]$ pip install ray
Segmentation fault
```

serbinsh, 2020-07-07:
Also there is a similar thread here: https://github.com/ray-project/ray/issues/6451

serbinsh, 2020-07-07:
Just realized that the seg faults come from conda using glibc 2.12, which was probably obvious to others

```
(/data2/sserbin/conda_envs/isofit) [sserbin@modex 20171108_Pasadena]$ conda info

     active environment : /data2/sserbin/conda_envs/isofit
    active env location : /data2/sserbin/conda_envs/isofit
            shell level : 1
       user config file : /home/sserbin/.condarc
 populated config files : /home/sserbin/.condarc
          conda version : 4.8.3
    conda-build version : not installed
         python version : 3.7.7.final.0
       virtual packages : __glibc=2.12
       base environment : /home/sserbin/miniconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/sserbin/miniconda3/pkgs
                          /home/sserbin/.conda/pkgs
       envs directories : /home/sserbin/miniconda3/envs
                          /home/sserbin/.conda/envs
               platform : linux-64
             user-agent : conda/4.8.3 requests/2.24.0 CPython/3.7.7 Linux/2.6.32-696.3.2.el6.x86_64 scientific/6.10 glibc/2.12
                UID:GID : 32248:32248
             netrc file : None
           offline mode : False
```

pgbrodrick, 2020-07-07:
Hi Shawn,

There'a lot here, but I'll try to get to your core question first, which I believe is: **So my question is how ray is being used for parallelization and whether it would be possible to make the ray lib dependency optional in situations like this?**

Currently, ray is used for parallelization at two different points - the first is during LUT generation, and the second is during inversions.  I brought Ray onboard here mostly because of persistent challenges with using native python multiprocessing - despite substantial effort, we were unable to get full core utilization on modern systems, particularly ones using multi-socket cpus.  Ray addressed this issue, and also facilitates multi-node work, which helps us process very large flightlines or finely spaced LUTs.  Ultimately, the choice here comes down to whether we want to be tied to poor performance with multiprocessing, or go with Ray - at the time, I had not seen reports of challenges with older machines, but even if I had I think I would still lean to have gone this way.  That said, if you have suggestions for better approaches, we could certainly reconsider.

Re: turning off ray, we currently have multiple options supported that help with turning off parallelization.  The first is to use the [implementation setting](https://isofit.readthedocs.io/en/latest/isofit.configs.sections.implementation_config.html) rte_configure_and_exit, which will set up the LUT settings files but not execute the code, allowing you to run your own parallelization.  Additionally, if you set n_cores to 1, ray will run in a 'serial' mode, which also helps with debugging.

However, neither of these will address your problem, since ray won't even import.  Forking and removing ray is certainly an option, which appears to be largely what @ashiklom has done on his branch - though I don't see an immediate cause to the error you reference above....I wonder if you have multiple isofit instances in your conda env somehow?

I'm glad to see you posted your question to the Ray repo as well, which I think is probably a more appropriately place to raise the dependency issues.  Resolving the conda environment on this particularly machine is going to be very challenging - if you really want to pursue it here, I'd suggest posting all details regarding what is installed on your particular environment, all paths, etc. - but given then nature of your machine it will likely still be challenging to track down.  I'm sorry I can't offer anything more concrete than that, but perhaps one of the other maintainers or users has some ideas.

serbinsh, 2020-07-08:
@pgbrodrick Thanks for the detailed reply!  My general response is, I completely understand your comments and choice to use ray for the updated parallelization support.  I think we are running into the common issue if lib support so, as you say, its not really an isofit problem but a pyhton/system lib support issue

For now, what I have done is to get LibRadTran running on my Mac Desktop and built isofit in conda.  I am using this as an alternative to my cluster at work, for now, to be able to work on new examples and make progress.  I had to mess around with gcc versions and compiler options on my mac but eventually got 2.0.3 up and running and am running tests.  Everything seems to be working and ray is successfully parallelizing using my conda env.

In addition, I am meeting with my IT folks on Monday to discuss updating modex (the cluster) to a much newer LTS OS.  That will solve the glibc issue, at least for awhile.  Hopefully ray dev is smart enough to maintain support for LTS systems.

Also as I have access to NASA ADAPT/NCCS I will likely use IsoFit there.  Seems @ashiklom got libradtran running there so we can both share that build for running isofit sims. 

So I guess we can close this but maybe add a note to IsoFit about support issues on RHEL 6.10 and less?  given that glibc is 2.12 and you need at least 2.14 for ray

serbinsh, 2020-07-08:
Also, on my mac I did not run into the 

```
TypeError: _run_set_of_spectra() takes 3 positional arguments but 4 were given
```
Issue which now that I have been messing around with all of this is clearly related to the fact that ray spawning failed to generate LUTs.  basically it was a ray issue.

serbinsh, 2020-07-08:
Sorry and one more, evidence that the Pasadena

[image]
 run finished on my mac

davidraythompson, 2020-07-08:
Hi Shawn,
Great!  Do you have time to work with me on your retrievals?  I'd like to see what we can do about those residuals.  
David

serbinsh, 2020-07-08:
@davidraythompson Yes, of course!  This was just running your basic example to make sure my environment was working.

Simultaneously, @ashiklom (as you know) is working on a reduced complexity wrapper to run IsoFit workflows and so I expect to be adopting that for running science retrievals.  I think when i have my full workflow env setup we can talk about / discuss RT issues/residuals.  I am also working on getting access to MODTRAN at BNL to generate LUTs as well so I can compare the two.  

But perhaps the next step for me is to run through the Hypertrace examples and LUTs you generated for SBG to confirm I can re-run my Alaskan endmembers through and get similar results as before.  

We can ping you when we are ready to start running real runs (not test runs) again!

davidraythompson, 2020-07-08:
@serbinsh Excellent.  This sounds like a good strategy.

ashiklom, 2020-07-09:
Posting here because it's related to Ray, but might be worth its own issue:

The `isofit.core.Isofit` class doesn't seem to do a great job of cleaning up after itself. Specifically, trying to run `Isofit(config)` multiple times in the same Python session throws the following error:

```
RuntimeError: Maybe you called ray.init twice by accident? This error can be suppressed by passing in 'ignore_reinit_error=True' or by calling 'ray.shutdown()' prior to 'ray.init()'.
```

Meanwhile, running it in different sessions running in parallel seems to create multiple Ray servers that conflict with each other.

One way to get around this is to start ray in a separate Python session (or from the command line, `ray start --head`) and then use the Ray address `"auto"`. This will cause each subsequent Isofit process to hook into the existing Ray cluster rather than trying to create a new one. At least that's the theory -- I haven't quite gotten this to work on my local machine.

Another thing that should probably happen is a call to `ray.shutdown()` at the end of the `Isofit.run()` function, as recommended by the error message. Or similarly, only initializing ray if it isn't already initialized, e.g.:

```python
if not ray.is_initialized():
  ray.init(**rayargs)
```

serbinsh, 2020-07-09:
@ashiklom is part of this referencing this error stack I get sometimes?

```Done!
*** Aborted at 1594321483 (unix time) try "date -d @1594321483" if you are using GNU date ***
PC: @                0x0 (unknown)
*** SIGABRT (@0x2d9135480000521c) received by PID 21020 (TID 0x7fbd502437c0) from PID 21020; stack trace: ***
    @     0x7fbd4f45b0e0 (unknown)
    @     0x7fbd4f45b067 gsignal
    @     0x7fbd4f45c448 abort
    @     0x7fbd32d76051 google::LogMessage::Flush()
    @     0x7fbd32d76121 google::LogMessage::~LogMessage()
    @     0x7fbd32d539f9 ray::RayLog::~RayLog()
    @     0x7fbd32aba3d5 ray::CoreWorkerProcess::~CoreWorkerProcess()
    @     0x7fbd32aba42a std::unique_ptr<>::~unique_ptr()
    @     0x7fbd4f45db29 (unknown)
    @     0x7fbd4f45db75 exit
    @     0x7fbd4fc98ebe Rstd_CleanUp
    @     0x7fbd4fc999cf R_CleanUp
    @     0x7fbd4fb8d51b end_Rmainloop
    @     0x7fbd4fb8d58e run_Rmainloop
    @     0x7fbd4fb8d59e Rf_mainloop
    @           0x4008b8 main
    @     0x7fbd4f447b45 __libc_start_main
    @           0x4007b9 (unknown)
Aborted```

ray and the forward LUT gen and inversion works but when complete i get this, but using from within reticulate in R.  Or maybe this is also a different issue ;)

davidraythompson, 2020-07-09:
RE: re-initialization, I managed to solve it in my case by adding 'ignore_reinit_error':True to rayargs in isofit.py

serbinsh, 2020-07-09:
If we wanted to close this too (ray not compatible on older machines with glibc < 2.14) @pgbrodrick perhaps a simple solution for this is to provide a minimum requirements file or comment in the readme?

pgbrodrick, 2020-07-09:
I'll add the above line to the readme documentation (outside of a PR, since it's text update).

Re: the ray initialization, it looks like in a previous PR, @jfahlen had recommended specific cleanup (https://github.com/isofit/isofit/commit/489ac533d18c7f349c4816b1f67146a7075aad34), which was accepted.  During a refactor, this must have fallen back out, but I'll submit into a PR and resubmit - will hold this issue in place until I confirm that resolves the issues raised by @ashiklom above.

pgbrodrick, 2020-07-10:
Tested, and #135 does solve this.  Just be sure to clean up the object:

```
model = Isofit('my_config.json')
model.run()
del model
model2 = Isofit('my_config.json')
model2.run()
del model
```

Support for multiple, independent models would require a few extra flags for ray, and could be done in the future if needed.

@davidraythompson , I think #135 is ready for you, and then we can close this out.

closed 2020-07-17 by pgbrodrick