isofit/isofit #132: LibRadTran enhancements

ashiklom, 2020-07-09:
Three closely related issues to optimizing LUT generation using LibRadTran (LRT from here on to save typing).

---

(1) LRT has the ability to simulate radiance across multiple zenith angles at the same time, with basically no hit to performance. In my tests with `uvspec` (the LRT executable), I found that setting `umu` (cosine output zenith angle) to less than 1 (off nadir), LRT runs ~10x slower. However, it takes almost exactly the same length of time to produce output for multiple `umu`s as one, which suggests that it does the same (more complicated) math once for all of the angles and then can quickly spit out as many as we want. Some benchmarks:

```
umu 1.0. # ~1.5 minutes on my MacBook pro
umu 0.9  # ~10 minutes on my MacBook pro
umu 0.7 0.8 0.9 1.0  # ~10 minutes on my MacBook pro
```

Haven't tried this for azimuth angles, but I wouldn't be surprised if the behavior was similar.

For hypertrace-related activities where we're interested in looking across a range of sun-sensor geometries, I think there's a real opportunity to use this to cut down on the number of LUTs that need to be generated. However, the Isofit libradtran code would need to be adjusted accordingly.

---

(2) Closely related, is it possible to include angles in the LUTs? In other words, can we include geometric information in the LUT grids and use that for interpolation, as we do with atmospheric conditions? This would allow a single set of LUTs to cover a wide range of not only atmospheric compositions but also sun-sensor geometries. Skimming the relevant code, it _seems_ like it might already do this (by adding more stuff to the `lut_grid` and `lut_names`), but when I experimented with modifying the Isofit LRT code to be able to do this, but I got some weird results.

----

(3) LRT is able to calculate the spherical albedo, but requires a separate solver to do it. To do this, one needs to add `disort_spherical_albedo` on its own line and add `spher_alb` to the list of outputs. In my tests, this runs relatively quickly --- about as long as a standard run at nadir -- and the speed is largely independent of `umu`. However, in this mode, LRT _only_ calculates the spherical albedo and nothing else. I wonder if it would make sense to replace Isofit's current mode of calculating `sphalb` with this direct calculation. AFAICT, we would still need to run LRT twice to get `rhoatm` and `rho05`, which are necessary to calculate `transm`, but we could replace the middle albedo calculation (albedo = 0.25) with the potentially faster `sphalb` calculation. I think this would also avoid some of the current numerical issues with that equation (#78).

davidraythompson, 2020-07-09:
Hi @ashiklom 
(1) Good idea. 
(2) We are certainly wired to include angles in the LUTs, for MODTRAN at least.  Note that the atmospheric state vector elements are a subset of the LUT dimensions.  We can also include view angles, and have before (Thompson et al. RSE 231, 2019).  If you pass in an *obs format file containing view angles, and the view zenith parameters are included in the LUT, it should do the appropriate interpolation.  We might need a couple more lines of code if you want to do this for LibRadTran though.
(3) I'm ambivalent here.  I'm told that calculating algebraically from multiple runs can be more accurate, because it calculates an "effective spherical albedo" that is consistent with the transmission and path radiance terms.  This might be different than a "first-principles" sphalb value since it accounts for instrument sampling effects. I don't know whether we'd see a difference in practice though.  I think you and Shawn are the biggest LibRadTran users right now so I'd trust your judgement on whether this is an important feature.

jammont, 2024-09-16:
LibRadTran support has been deprecated.

closed 2024-09-16 by jammont