isofit/isofit #133: SciKit dependency on NASA ADAPT/NCCS

serbinsh, 2020-07-09:
While getting IsoFit to run on ADAPT/NCCS in conda I noticed an issue when running the IsoFit Pasadena example. This occurred at the start of the example but didnt initially stop the run as it still built the new LUTs (have LibRadTran installed) but it failed at the start of the model inversion step/. 

Here is the error

```
>>> from isofit.utils import surface_model
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/sserbin/.conda/envs/isofit_rwrapper/lib/python3.8/site-packages/isofit/utils/__init__.py", line 7, in <module>
    from .surface_model import surface_model
  File "/home/sserbin/.conda/envs/isofit_rwrapper/lib/python3.8/site-packages/isofit/utils/surface_model.py", line 25, in <module>
    from sklearn.cluster import KMeans
  File "/home/sserbin/.conda/envs/isofit_rwrapper/lib/python3.8/site-packages/sklearn/__init__.py", line 79, in <module>
    from . import __check_build  # noqa: F401
ImportError: cannot import name '__check_build' from partially initialized module 'sklearn' (most likely due to a circular import) (/home/sserbin/.conda/envs/isofit_rwrapper/lib/python3.8/site-packages/sklearn/__init__.py)
>>> surface_model("configs/ang20171108t184227_surface.json")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'surface_model' is not defined
```

Even just trying to load that module throws the error.  

I got around this by rolling back SkiKit

```
(isofit_rwrapper) [sserbin@adaptlogin102 sserbin]$ pip install scikit-learn==0.22.2

Collecting scikit-learn==0.22.2

  Downloading scikit_learn-0.22.2-cp38-cp38-manylinux1_x86_64.whl (7.0 MB)

     |████████████████████████████████| 7.0 MB 81 kB/s

Requirement already satisfied: scipy>=0.17.0 in ./.conda/envs/isofit_rwrapper/lib/python3.8/site-packages (from scikit-learn==0.22.2) (1.5.1)

Requirement already satisfied: joblib>=0.11 in ./.conda/envs/isofit_rwrapper/lib/python3.8/site-packages (from scikit-learn==0.22.2) (0.16.0)

Requirement already satisfied: numpy>=1.11.0 in ./.conda/envs/isofit_rwrapper/lib/python3.8/site-packages (from scikit-learn==0.22.2) (1.19.0)

Installing collected packages: scikit-learn

  Attempting uninstall: scikit-learn

    Found existing installation: scikit-learn 0.23.1

    Uninstalling scikit-learn-0.23.1:

      Successfully uninstalled scikit-learn-0.23.1

Successfully installed scikit-learn-0.22.2

(isofit_rwrapper) [sserbin@adaptlogin102 sserbin]$ python

Python 3.8.3 (default, Jul  2 2020, 16:21:59)

[GCC 7.3.0] :: Anaconda, Inc. on linux

Type "help", "copyright", "credits" or "license" for more information.

>>> from isofit.utils import surface_model

>>>
```

Re-running that example and it hasnt failed in the same spot.  next I need to optimize core usage as I think its only running on 1.

SO I guess this serves as an FYI?  Not sure if anyone else ran into this issue elsewhere?  Also ADAPT/NCCS has some other quirks so it really could be machine-specific.

pgbrodrick, 2020-07-09:
Hi Shawn,

The above seems to refer to a circular import error - do you already have sklearn imported in your local session? I'm not able to reproduce locally:

```
(isofit) brodrick@MT-206693:~/repos/isofit$ pip install scikit-learn==0.23.1
Collecting scikit-learn==0.23.1
  Using cached https://files.pythonhosted.org/packages/32/37/51bf33e269be272255ab6a1c172e15b930bcf33316616efc82e0759ca4ff/scikit_learn-0.23.1-cp37-cp37m-macosx_10_9_x86_64.whl
Requirement already satisfied: scipy>=0.19.1 in /Users/brodrick/miniconda3/envs/isofit/lib/python3.7/site-packages (from scikit-learn==0.23.1) (1.3.1)
Requirement already satisfied: numpy>=1.13.3 in /Users/brodrick/miniconda3/envs/isofit/lib/python3.7/site-packages (from scikit-learn==0.23.1) (1.17.3)
Requirement already satisfied: threadpoolctl>=2.0.0 in /Users/brodrick/miniconda3/envs/isofit/lib/python3.7/site-packages (from scikit-learn==0.23.1) (2.1.0)
Requirement already satisfied: joblib>=0.11 in /Users/brodrick/miniconda3/envs/isofit/lib/python3.7/site-packages (from scikit-learn==0.23.1) (0.14.0)
Installing collected packages: scikit-learn
  Found existing installation: scikit-learn 0.23.0
    Uninstalling scikit-learn-0.23.0:
      Successfully uninstalled scikit-learn-0.23.0
Successfully installed scikit-learn-0.23.1
(isofit) brodrick@MT-206693:~/repos/isofit$ python
Python 3.7.5 (default, Oct 25 2019, 10:52:18) 
[Clang 4.0.1 (tags/RELEASE_401/final)] :: Anaconda, Inc. on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from isofit.utils import surface_model
>>> import sklearn
>>> sklearn.__version__
'0.23.1'
>>> 
```

The surface_model util is certainly not optimized, but the typical us is on fairly sparse scenes, or extracted libraries - definitely feel free to upgrade it though :-).

serbinsh, 2020-07-09:
@pgbrodrick not surprised you couldnt replicate.  The issue likely comes from some strangeness on that machine.  basically sometimes when installing python libs you get a "resource not available" error. Usually I can just re-run and eventually it works. this time it would not stop giving me the error.  I expected this might be related so I tried rolling back which worked.  So I dont really know what the issue was specifically but unlikely isofit.

pgbrodrick, 2020-07-09:
Got it - hopefully something you can resolve.  Closing this issue then.

closed 2020-07-09 by pgbrodrick

serbinsh, 2020-07-09:
Well the NASA admins, but yeah....