isofit/isofit #136: Example files not ignored by git

serbinsh, 2020-07-10:
I wanted to post this here as I'm not sure if this, again, is only my issue but I have noticed that .gitignore is not successfully ignoring dynamic test files on my machines

e.g. 
```
	examples/20171108_Pasadena/images/frame_0.png
	examples/20171108_Pasadena/images/frame_1.png
	examples/20171108_Pasadena/images/frame_10.png
	examples/20171108_Pasadena/images/frame_11.png
	examples/20171108_Pasadena/images/frame_12.png
	examples/20171108_Pasadena/images/frame_13.png
	examples/20171108_Pasadena/images/frame_14.png
	examples/20171108_Pasadena/images/frame_15.png
	examples/20171108_Pasadena/images/frame_16.png
	examples/20171108_Pasadena/images/frame_17.png
	examples/20171108_Pasadena/images/frame_18.png
	examples/20171108_Pasadena/images/frame_19.png
	examples/20171108_Pasadena/images/frame_2.png
	examples/20171108_Pasadena/images/frame_20.png
	examples/20171108_Pasadena/images/frame_3.png
	examples/20171108_Pasadena/images/frame_4.png
	examples/20171108_Pasadena/images/frame_5.png
	examples/20171108_Pasadena/images/frame_6.png
	examples/20171108_Pasadena/images/frame_7.png
	examples/20171108_Pasadena/images/frame_8.png
	examples/20171108_Pasadena/images/frame_9.png
	examples/20171108_Pasadena/output/ang20171108t184227_alg_v2p11_BeckmanLawn-libRadTran.txt
	examples/20171108_Pasadena/output/ang20171108t184227_data_v2p11_BeckmanLawn-libRadTran.mat
	examples/20171108_Pasadena/output/ang20171108t184227_mdl_v2p11_BeckmanLawn-libRadTran.txt
	examples/20171108_Pasadena/output/ang20171108t184227_rfl_v2p11_BeckmanLawn-libRadTran.txt
	examples/20171108_Pasadena/remote/ang20170228_surface_model.mat
```

I can probably come up with something that would work for me but clearly .gitignore was designed to ignore these files. Has anyone else noticed this?

pgbrodrick, 2020-07-10:
Hi @serbinsh ,

This is a standing (undocumented) issue, and needs to be addressed.  In that spirit, the examples also need some cleaning to reduce the total number of files, remove different instrument channelizations, etc.  Any contributions to these issues would be welcome!

Phil

serbinsh, 2020-07-10:
@pgbrodrick OK let me at least get an update to can address the example files. Not sure I followed what you mean by "remove different instrument channelizations"?  

Did you perhaps have a general high-level list of the changes you had in mind? RE number of files, I suppose I could review any that seem to be unnecessary to maintain following a test run and have them deleted.  My only issue will be that I can really only test LibRadTran examples right now given I dont currently have access to MODTRAN so my edits will be based on that.

pgbrodrick, 2020-07-11:
Sorry for the vaguery, and for conflating the issue - I don't think this is aforementioned cleanup really pertains to the LibRadTran examples.  Let's keep the issue contained with the .gitignore cleanup!

closed 2020-08-24 by pgbrodrick