isofit/isofit #143: Invalid state vector in forward simulation mode for multicomponent surface?

ashiklom, 2020-07-29:
When trying to run a forward simulation using a reference mineral reflectance file, I kept getting extremely high TOA radiance values (>100), which in turn produced nonsensical (>1) reflectance values. As best I can tell, the root cause of the problem is this block:

https://github.com/isofit/isofit/blob/f7861332f6099ac0ab271715db514046cc540ed3/isofit/inversion/inverse.py#L282-L284

When I added some print statements, I discovered that my `self.fm.init` here looked like this:

```
[1.5, 1.5, 1.5, 1.5, <etc...>, 0.1, 1.2]
```

As best I can tell, the `1.5`s originate from here:

https://github.com/isofit/isofit/blob/f7861332f6099ac0ab271715db514046cc540ed3/isofit/surface/surface_multicomp.py#L87-L92

If I understand correctly, the vector here should actually be the known surface reflectance (i.e. `self.fm.surface.rfl`) + the state of the atmosphere. On my fork of Isofit, this seems to have been resolved with this hacky commit (at least, after adding this, I once again get reasonable TOA radiance values and reflectance estimates):

https://github.com/ashiklom/isofit/commit/32469b57218330be42afea829768764288f29d82

Am I doing something wrong here, or is this an actual bug?

My full config is as follows (click to expand).

<details>

```xml
{
  "forward_model": {
    "instrument": {
      "SNR": 300,
      "integrations": 1,
      "wavelength_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/data/wavelengths.txt"
    },
    "surface": {
      "surface_category": "multicomponent_surface",
      "surface_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/data/prior.mat"
    },
    "radiative_transfer": {
      "lut_grid": {
        "AOT550": [
          0.001,
          0.123,
          0.6
        ],
        "H2OSTR": [
          1.0,
          2.5,
          3.25
        ]
      },
      "statevector": {
        "AOT550": {
          "bounds": [
            0.01,
            0.6
          ],
          "scale": 0.01,
          "prior_mean": 0.05,
          "prior_sigma": 0.2,
          "init": 0.06
        },
        "H2OSTR": {
          "bounds": [
            1.0,
            3.25
          ],
          "scale": 0.01,
          "prior_mean": 1.75,
          "prior_sigma": 1.0,
          "init": 1.5
        }
      },
      "unknowns": {
        "H2O_ABSCO": 0.01
      },
      "radiative_transfer_engines": {
        "vswir": {
          "engine_name": "libradtran",
          "engine_base_dir": "/Users/ashiklom/projects/models/libradtran/libRadtran-2.0.3",
          "environment": "",
          "lut_names": [
            "H2OSTR",
            "AOT550"
          ],
          "statevector_names": [
            "H2OSTR",
            "AOT550"
          ],
          "lut_path": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/luts/atm_midlatitude_winter__szen_0__ozen_0__saz_0__oaz_0",
          "template_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/luts/atm_midlatitude_winter__szen_0__ozen_0__saz_0__oaz_0/lrt-template.inp"
        }
      }
    }
  },
  "implementation": {
    "inversion": {
      "windows": [
        [
          400.0,
          1300.0
        ],
        [
          1450.0,
          1780.0
        ],
        [
          1950.0,
          2450.0
        ]
      ],
      "simulation_mode": true
    },
    "mode": "simulation"
  },
  "input": {
    "reflectance_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/data/reference_reflectance"
  },
  "output": {
    "simulated_measurement_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/output-quick/atm_midlatitude_winter__szen_0__ozen_0__saz_0__oaz_0/snr_300/aod_0.06__h2o_1.5/toa-radiance"
  }
}
```

</details>

pgbrodrick, 2020-07-29:
My immediate guess is that this is due to a conflation of different config inputs that I did not safeguard against, though it's also possible that you're intended use is different from what I expected when I hooked up the IO here, so don't hesitate to correct me.  But I wouldn't imagine that you'd want a multicomponent surface if you're doing simulation mode.

If you don't object to it, try just setting the surface_category to 'surface', and see if that resolves the issue (I don't think we'll ever get to the initialization you specify above if you do that).  If this works, I'll add in a config check to not allow multicomponent_surfaces to be used if the reflectance_file is specified as an input.

davidraythompson, 2020-07-29:
I  think that typically the vanilla parent reflectance object (not multicomponent version) is used for simulations, because one can exactly specify the reflectance.

ashiklom, 2020-07-30:
> If you don't object to it, try just setting the surface_category to 'surface', and see if that resolves the issue (I don't think we'll ever get to the initialization you specify above if you do that).

@pgbrodrick You were right! This seems to have resolved the problem, so I'll adjust my workflow accordingly. Thanks as always for the quick and helpful replies!

closed 2020-07-30 by ashiklom