isofit/isofit #144: Pixels getting skipped

jfahlen, 2020-08-03:
In ipython, the following command

```
In [1]: from isofit.core.isofit import Isofit                                                                                    
In [2]: Isofit("config/config_file.yaml", row_column = "670, 740, 25, 60", level = 'DEBUG', logfile = './ipython_debug_output_2.txt').run()
```
appears to drop pixels from the retrieval. Grepping the log file for how many lines were processed with
```
$ grep Row ipython_debug_output_2.txt | wc
   1367    5468   32808
```
indicates that 33 pixels were not processed (70*35 = 1400, not 1367). The output state file contains zeros for the pixels that were apparently not processed. (See the blue speckles in the image.) 
[image]
This issue is not exactly reproducible. Repeated runs may have a different number of pixels that are not processed and the location of those pixels can vary.

I don't seem to have the issue when running from the bash command line calling python directly (without ipython), so this is not a pressing issue.

My best guess is that somehow ray and ipython don't like each other.

jfahlen, 2020-08-03:
Update:

One pixel with a zero is 713, 51. The following command
```
$ grep Row ipython_debug_output_2.txt | grep 713 | sort
DEBUG:Row 713 Column 25
DEBUG:Row 713 Column 26
DEBUG:Row 713 Column 27
DEBUG:Row 713 Column 28
DEBUG:Row 713 Column 29
DEBUG:Row 713 Column 30
DEBUG:Row 713 Column 31
DEBUG:Row 713 Column 32
DEBUG:Row 713 Column 33
DEBUG:Row 713 Column 34
DEBUG:Row 713 Column 35
DEBUG:Row 713 Column 36
DEBUG:Row 713 Column 37
DEBUG:Row 713 Column 38
DEBUG:Row 713 Column 51
DEBUG:Row 713 Column 52
DEBUG:Row 713 Column 53
DEBUG:Row 713 Column 54
DEBUG:Row 713 Column 55
DEBUG:Row 713 Column 56
DEBUG:Row 713 Column 57
DEBUG:Row 713 Column 58
DEBUG:Row 713 Column 59
```
shows that a bunch of columns were skipped, but that 51 was processed. I don't understand what's happening.

pgbrodrick, 2020-08-03:
Hi Jay,

Does the image you're displaying correspond with the debug output shown above?  I do not see a blank pixel at (713, 51) [row, column].

jfahlen, 2020-08-03:
I thought it did, I'm confused now. I'm rerunning to make sure.

jfahlen, 2020-08-03:
Many of my previous statements are incorrect. The problem just occurred running the current master from bash (using python, not ipython) and the number of lines in the log doesn't seem to be relevant. In this case, the wc command shown above yields 2450 lines, which is 50 too many.

Here's a new picture showing the different pixels:
[image]

I'm going to try to find a work around for now until I can tie this down a bit more.

pgbrodrick, 2020-08-03:
Alright, this tells us the code is at least trying each pixel, but not that each pixel succeeds.  If possible, to help isolate the problem, it would be great to modify the log at:

https://github.com/isofit/isofit/blob/master/isofit/core/isofit.py#L169-L172

to report results for every pixel (remove the modulus reporting check), and to print the row, column pair instead of the index.  This will help tell us isolate if the inversion is actually being completed at each pixel, or if for some reason occasional inversion instances die for some reason (unlikely, but I'd like to rule it out).

pgbrodrick, 2020-08-24:
@jfahlen , checking in on the state of this - I believe based on personal communication that this was determined to be a system-specific issue that was resolved.  Can you confirm?

jfahlen, 2020-08-24:
I think this is system specific, but I do not know the specific cause and I have no fix for it yet.

pgbrodrick, 2021-03-18:
Current updates have better error reporting for any pixel that gets zero'd out.  Between that and ray updates, hopefully this is mitigated.  If it pops up again, please report!

closed 2021-03-18 by pgbrodrick