isofit/isofit #152: Add bibliography to documentation

davidraythompson, 2020-08-24:
References with links to all papers using this code.

pgbrodrick, 2020-08-24:
Agreed!  Let's do it with an .rst and get it hooked up to the documentation as well.  I'll take getting this started as an action item, and I'm sure you'll know of a few manuscripts I miss!

davidraythompson, 2020-08-24:
My list so far:

N. Carmon, D. R. Thompson, J. Susiluoto, N. Bohn, D. S. Connelly, M. J. Turmon, K. Cawse-Nicholson, R. O Green, M. R Gunson. Uncertainty Quantification for Global Remote Spectroscopy of Surface Composition.  Remote Sensing of Environment, in review, 2020.

D. R. Thompson, Alberto Candela, Nimrod Carmon, Roger N. Clark, David Connelly, Bethany Ehlmann, Robert O. Green, Raymond Kokaly, Nathalie Mahowald, Gregory Okin, Gregg A. Swayze, Michael Turmon, and David S. Wettergreen, “Quantifying Uncertainty for Remote Spectroscopy of Surface Composition,” Remote Sensing of Environment. In press, 2020.

D. R. Thompson, K. N. Babu, A. Braverman, M. Eastwood, R. O. Green, J. Hobbs, J. B. Jewell, M. Mishra, A. Mathur, V. Natraj, F. C. Seidel, P. Townsend, M. Turmon. “Optimal Estimation of Spectral Surface Reflectance in Challenging Atmospheres,” Remote Sensing of Environment, 232, 111258, 2019.

R. Frouin, B.  Franz, A. Ibrahim, K. Knobelspiesse, Z. Ahmad, B. Cairns, J. Chowdhary, H. M. Dierssen, O. Dubovik, J. Tan, A. B. Davis, D. J. Diner, O. Kalashnikova, D. R. Thompson, L. R. Remer, E. Boss, O. Coddington, B-C Gao, L. Gross, O. Hasekamp, A. Omar, B. Pelletier, D. Ramon, P-W. Zhai., Atmospheric correction of satellite ocean-color imagery during the PACE era. Frontiers in Marine Science 7: 145, 2019.

D. R. Thompson, K. Cawse-Nicholson, Z. Erickson, C. Fichot, C. Frankenberg, B-C. Gao, M. M. Gierach, R. O. Green, D. Jensen, V. Natraj, A. Thompson. “A unified approach to estimate land and water reflectances with uncertainties for coastal imaging spectroscopy,” Remote Sensing of Environment 231, 111198, 2019.  

B. D. Bue, D. R. Thompson, S. Deshpande, M. Eastwood, C. Fichot, R. O. Green, T. Mullen, V. Natraj, and M. Parente, “Neural Network Radiative Transfer for Imaging Spectroscopy.” Atmospheric Measurement Techniques 12, 2567–2578, 2019 https://doi.org/10.5194/amt-12-2567-2019

D. R. Thompson, V. Natraj, R. O. Green, M. Helmlinger, B.-C. Gao, and M. L. Eastwood, “Optimal Estimation for Imaging Spectrometer Atmospheric Correction.” Remote Sensing of Environment 216, p. 355-373, 2018.

davidraythompson, 2020-08-28:
Thanks Phil!

closed 2020-08-28 by davidraythompson