isofit/isofit #153: Question: Running IsoFit w/ Ray across multiple nodes

serbinsh, 2020-08-24:
I am experimenting with running IsoFit and the new py-hypertrace on Adapt/NCCS and was wondering if there are any docs/readme's or examples of running IsoFit across multiple nodes in parallel with Ray?  So far I am unsuccessful in this and instead it just runs on a single node, even in an interactive session where I have multiple nodes available. Its unclear how to expose this to Ray.  This may require some research.

pgbrodrick, 2020-08-24:
Hey Sean,

There's pretty extensive ray documentation available [here](https://docs.ray.io/en/master/cluster/index.html).  Once you have ray set up and distributing appropriately, on the isofit side it should be very easy - just transfer in the relevant parameters (redis_password, ip_head, and maybe n_cores) into the [implementation config](https://isofit.readthedocs.io/en/latest/isofit.configs.sections.implementation_config.html).  I can post examples, but I suspect they'd be explicit to my setup, and redundant/less informative to the ray examples....

serbinsh, 2020-08-24:
@pgbrodrick Perhaps though I have been looking at ray examples and I havent been able to make it work successfully.  Perhaps one of your examples with syntax would be helpful for me. 

An example here with SLURM: https://github.com/ray-project/ray/issues/826.  IF you had a "basic" example that would be good.

It does seem like the approach may be to create a slurm session then create a ray cluster and use this with IsoFit? I also wonder if its possible to create a slurm submission script that can incorporate all of this.

serbinsh, 2020-08-24:
OK, I did locate that example now in the docs: https://docs.ray.io/en/master/cluster/slurm.html#ray-slurm-deploy

pgbrodrick, 2020-08-24:
Great, you found it!  For SLURM, I built off that and use the following a lot of the time:

```
#!/bin/bash

#SBATCH --job-name=isofit-broadcast
#SBATCH --cpus-per-task=40
#SBATCH --mem-per-cpu=4GB
#SBATCH --nodes=5
#SBATCH --tasks-per-node=1

nodes=$(scontrol show hostnames $SLURM_JOB_NODELIST) # Getting the node names
nodes_array=( $nodes )

# Must be one less that the total number of nodes
worker_num=$(( ${#nodes_array[@]} - 1 )) 

node1=${nodes_array[0]}

# Start the main node, and get the redis_password
ip_prefix=$(srun --nodes=1 --ntasks=1 -w $node1 hostname --ip-address) # Making address
suffix=':6379'
ip_head=$ip_prefix$suffix
redis_password=$(uuidgen)

# Start the head
srun --nodes=1 --ntasks=1 -w $node1 ray start --block --head --redis-port=6379 --redis-password=$redis_password & 
echo "Head started on $node1"

# Give the head time to start before connecting workers
sleep 2

# Start the worker
for ((  i=1; i<=$worker_num; i++ ))
do
  node2=${nodes_array[$i]}
  echo "starting on node $node2"
  srun --nodes=1 --ntasks=1 -w $node2 ray start --num-cpus=$SLURM_CPUS_PER_TASK --block --address=$ip_head --redis-password=$redis_password & 
  # Flag --block will keep ray process alive on each compute node.
  sleep 2
done

total_cores=$(( SLURM_JOB_NUM_NODES * SLURM_CPUS_PER_TASK ))

python ~/bin/set_ray_params.py ${1} $redis_password $ip_head $total_cores
python -u -c "from isofit.core.isofit import Isofit; model = Isofit(\"${1}\", logfile=\"${2}\"); model.run()"
```

coupled with

```
import argparse
import json

parser = argparse.ArgumentParser(description="Update config with ray parameters.")
parser.add_argument('config_path',  type=str)
parser.add_argument('redis_password',  type=str)
parser.add_argument('ip_head',  type=str)
parser.add_argument('n_cores',  type=int)
args = parser.parse_args()

with open(args.config_path, 'r') as fin:
    config = json.load(fin)

config['implementation']['redis_password'] = args.redis_password
config['implementation']['ip_head'] = args.ip_head
config['implementation']['n_cores'] = args.n_cores

with open(args.config_path, 'w') as fout:
    fout.write(json.dumps(config, indent=4, sort_keys=True))
```

serbinsh, 2020-08-24:
Awesome thanks!

closed 2020-08-24 by pgbrodrick

serbinsh, 2020-08-24:
@pgbrodrick what is

```python ~/bin/set_ray_params.py ${1} $redis_password $ip_head $total_cores```
does that file contain the second part you posted?

pgbrodrick, 2020-08-24:
Yes, exactly - updated the above post to make that more clear for future readers.

serbinsh, 2020-08-24:
This is really helpful, thanks @pgbrodrick!  I was able to get the Ray example to run on Adapt/NCCS OK.  So next step is to develop a simple IsoFit example to make sure its running across nodes. If that works I should be good to run @ashiklom's wrapper on images across nodes as well, which will help to run larger areas.