isofit/isofit #158: I/O for other image formats

ashiklom, 2020-09-21:
Probably the most common ones are:

- [ ] GeoTIFF (including cloud-optimized?)
- [ ] HDF5 (e.g. NEON AOP)
- [ ] NetCDF (we get a clunky version of this "for free" from HDF5 support)

Similarly/alternatively, what do you think about reading files via GDAL? That gives a _lot_ of formats (including TIF and specifically-formatted HDF5 and NetCDF), though we may want to give some more flexibility for HDF5 and NetCDF.

ashiklom, 2020-09-21:
One way to work around this somewhat efficiently might be to allow (more of the) high-level utilities to work directly Numpy memmaps or similar. Though that could get problematic for large images.

serbinsh, 2020-09-21:
I would second including GeoTIFF, HDF5 (lots of RS data use this), and netCDF.  Xarray is very useful for netCDF in python as well, but we may not want another dependency.

pgbrodrick, 2020-09-21:
Hi Alexey,

I think it depends on whether you really mean IO or just I.  The output side is complicated, and we wouldn't actually get much benefit (e.g. compression, tiling, etc.) without a *major* refactor, given that these formats are all oriented towards BSQ style analysis.  The exception is HDF5, which is plausible, but again we wouldn't get too much compression benefit without a post-hoc adjustment, which you can always do now with a simple gdal call (which is what I currently tend to do).  It's also very tricky (or impossible) to do this style of write in parallel.  

On the inputs side, it's certainly plausible, though there are again challenges.  In particular, gdal-based reads from most formats are going to induce a huge cost if the scene is large.  HDF5 is again the exception, and would be reasonable to handle, but given the format flexibility it's going to require a decent amount of boilerplate to safeguard and manage.  I'm not totally convinced that it's not just worth making the gdal call a priori.

The other major danger is system lockup from parallel execution, which can happen in both I and O here depending on the system.

We could always consider gdal calls on the front and back ends to support users without much experience in that realm.

Interested to hear others' thoughts (please consider parallel I/O though - it's an absolute necessity for functional use of this codebase currently).

Phil

pgbrodrick, 2020-09-21:
> One way to work around this somewhat efficiently might be to allow (more of the) high-level utilities to work directly Numpy memmaps or similar. Though that could get problematic for large images.

Can you explain a bit more what you mean by this?

ashiklom, 2020-09-22:
Thanks Phil.

> The output side is complicated, and we wouldn't actually get much benefit (e.g. compression, tiling, etc.) without a major refactor, given that these formats are all oriented towards BSQ style analysis.

Point taken. Thinking through this a bit more, I do think you're right that doing the conversion externally to the main ISOFIT workflow is much more straightforward and probably more efficient. I _do_ think it might be worth adding a bit of documentation and possibly some small utilities/scripts for some common conversions (e.g., for the weird NEON AOP HDF5 format). I can help put this together based on my own use cases.

> One way to work around this somewhat efficiently might be to allow (more of the) high-level utilities to work directly Numpy memmaps or similar. Though that could get problematic for large images.

Sorry, this wasn't super clear. Looking more carefully at the ISOFIT `fileio.py`, I think I was actually describing [what you already do](https://github.com/isofit/isofit/blob/abd3a1dc10a17728b57d59ae191b0b5c384058fc/isofit/core/fileio.py#L182-L183). That actually makes me think that HDF5 support may not be that hard to implement -- I _think_ you should just be able to do something like:

```python
# Elsewhere...
rwflag = "r"
 if self.write:
   rwflag = "rw"
self.file = h5py.File(self.file, rwflag)
# Example h5_dataset_path for NEON AOP:
# "SERC/Reflectance/Reflectance_Data"
self.memmap = self.file[self.h5_dataset_path]
```

In principle, HDF5 [supports parallel IO](https://docs.h5py.org/en/stable/mpi.html#using-parallel-hdf5-from-h5py), though I honestly don't have a lot of experience with this (especially in Python) so I can't speak to how well it works.

> HDF5 is again the exception, and would be reasonable to handle, but given the format flexibility it's going to require a decent amount of boilerplate to safeguard and manage. I'm not totally convinced that it's not just worth making the gdal call a priori.

I'm probably doing something wrong, but I've actually yet to get GDAL to work with NEON AOP HDF5 files. I've found it much easier to work directly with the HDF5 datasets. Regardless, my inclination would be to start with a strict, well-documented, standard format (for both inputs and outputs; probably NEON AOP would be a good starting point) and then gradually relax it as requested by users.

pgbrodrick, 2020-09-22:
Hi Alexey,

Thanks for the clarifications.  A few points:

- I see you found the memmaps, great.  Agreed that in theory HDF5 does support parallel writes (with a bit of work), though I've had implementations problems on some systems previously...but I'm open to digging in again if it's a high demand item. 

> I'm probably doing something wrong, but I've actually yet to get GDAL to work with NEON AOP HDF5 files. I've found it much easier to work directly with the HDF5 datasets. 

Let's take this offline as it's out of scope, but I've been able to work this - let me know if you need a hand with it.

> Regardless, my inclination would be to start with a strict, well-documented, standard format (for both inputs and outputs; probably NEON AOP would be a good starting point) and then gradually relax it as requested by users.

The biggest problem here is on nailing the internal heirarchical paths.  We can simply require these in a configuration file though, and handle things that way.  Last I checked they were unfortunately not standardized.  I recognize that NEON is another large provider of relevant data, so I'm not entirely against adding some features, but we want to be careful here as the intent of ISOFIT is not to become an I/O package.  

I'll offer an alternative suggestion that's been on my radar for some time, which would be to increase the separation between the isofit core and fileio.  E.G., the isofit main class is responsible for delivering the forward results or inversions, and fileio is external to it, and responsible for reading, passing data to isofit, excepting results, and writing.  This would make it easier to remove the onus from ISOFIT for handling all IO cases, as the user could write a little wrapper for their custom IO workflow. It would also make it easier to integrate ISOFIT into other software. Thoughts?

ashiklom, 2020-09-22:
> The biggest problem here is on nailing the internal hierarchical paths. We can simply require these in a configuration file though, and handle things that way.

Yup, this is what I had in mind -- something like `h5_data_path`, `h5_wavelength_path`, `h5_fwhm_path`, etc. (or a nested `h5_paths` dict with a similar structure to the current ENVI file metadata.

> I'll offer an alternative suggestion that's been on my radar for some time, which would be to increase the separation between the isofit core and fileio.

IMHO, this is an excellent idea! I would actually much prefer this to support buried in the class hierarchy or if-else statements.

pgbrodrick, 2020-09-22:
> IMHO, this is an excellent idea! I would actually much prefer this to support buried in the class hierarchy or if-else statements.

Great - then barring objections this is the direction I think we should head.  There are a few outstanding PRs to resolve before digging in, but I've added this to planned development on the [project board.](https://github.com/isofit/isofit/projects/4).