isofit/isofit #160: Make `loc` files optional in `empirical_line`

ashiklom, 2020-09-29:
I'm working on incorporating some of the tools used in `apply_oe` into the current Hypertrace wrapper (particular the segmentation + empirical line, to accelerate the workflow for large images). However, the current `empirical_line` function has an ENVI location file as a required input, but I typically haven't used them in my workflows (in fact, I'm not totally sure where to get them -- are they typically distributed with AVIRIS radiance data?).

Is there a way to modify this function to work without location files? I realize the results will not be as good since the function uses information about vertical and lateral distances between points in the empirical line algorithm...but even so. Alternatively, is there a quick way to generate a "null model" location file for a given image? Would such a file just be a 1-band file with the same dimensions and metadata as the radiance file, with every value set to a constant elevation?

pgbrodrick, 2020-09-29:
Are you working in raw or map space Alexey?  If your analysis is in raw-space, then the loc file is just the IGM....if it's in ortho'd (map) space, then the loc file is an orthorectified IGM.  For ortho'd images, it would be straightforward to simply use the geospatial metadata of the scene to do the same interpolation, and we could support that.  In fact, I could make that the default if things are ortho'd, unless a loc file is provided....and then the loc file is not require unless you're in raw space (which is common, but it's also the standard IGM that way).

I'll note that the spatial info is also used to set up the radiative transfer models depending on the LUT configuration, so if you're neglecting to pass it into isofit, things can get tricky depending on the setup.  We could simply make a config error for this though.

ashiklom, 2020-09-30:
So the workflow starts from an orthorectified surface reflectance file and then uses Isofit's default forward model (without any additional files) to generate what I assume is an orthorectified radiance file, which in turn is what gets passed through the empirical line workflow. Based on that...

> For ortho'd images, it would be straightforward to simply use the geospatial metadata of the scene to do the same interpolation, and we could support that.

...I think this sounds like it would work for my workflow, and more generally seems like a sensible default. I'll take a look at some example IGM files and see if I can come up with something.

The other use case (related to mine) to consider here is of simulated reflectance scenes as produced by RTMs. One idea from our SBG group was to generate such reflectance from land surface models coupled to canopy RTMs at regional to global scales. Kerry and other folks are also working on simulating reflectance at specific locations using complex 3D RTMs (Kerry is using DIRSIG; here at GSFC, we have folks using DART). It would be useful to know what needs to provided with these model outputs for them to work with Hypertrace. From the above, sounds like just georeferenced rasters should be sufficient (once implemented).

davidraythompson, 2020-09-30:
FYI, for the purposes of the empirical line stage, the location values simply serve to identify nearby solutions (in horizontal and elevation coordinates) that define the local empirical line solution.  Those coordinates wouldn't necessarily have to be consistent with the geographic/observing geometry assumptions in the RTM for the workflow to give the correct answers.

ashiklom, 2020-09-30:
Thanks David! Yeah, I thought that might be the case, and was thinking that it might be nice to have a way to automatically generate a "null" location file/matrix with a prescribed spatial resolution and a flat elevation.

pgbrodrick, 2020-10-02:
Hi Alexey,

This 'null' setup sounds doable.  I took a look at the code in detail this morning, and the key thing I think we *don't* want to do is to bring in some additional potential elevation file, and append it to geospatial based location data.  No one explicitly suggested that, but I did consider it seems to just end to the endless possible configurations here.

For now, unless there are objections, I'm going to set up an option called '--null_location_elevation' as a float option, which if not None, and if the loc_file location is not populated, will fill that location with the spatial info from the georeferenced image if available, coordinate info if not georeferenced, and a flat elevation baed on the --nul_location_elevation option.

While not super clean, this is the easiest way I can think of on short notice to facilitate different location options, which I imagine would be useful from the forward/inverse cycle of your use case Alexey.  

Any objections?

FYI, when our new emulator comes online (very soon now), I imagine cleaning this implementation (apply_oe) up substantially, hopefully reducing the number of input options, as they're starting to get out of hand (no shock there!).

ashiklom, 2020-10-02:
> I'm going to set up an option called '--null_location_elevation' as a float option, which if not None, and if the loc_file location is not populated, will fill that location with the spatial info [...]

This sounds great, thanks!

FYI, in my [current implementation](https://github.com/ashiklom/isofit/blob/061c46f8edc1a2b4d60694967dd82787edb0af7d/examples/py-hypertrace/hypertrace.py#L234-L273), I'm not calling the `apply_oe.py` script directly but rather calling the `segment`, `extractions`, and `empirical_line` functions directly, so that they are consistent with the rest of the workflow.

pgbrodrick, 2020-10-02:
Cool - I'll do the synthetic loc_file generation as a function inside of apply_oe, so you can import and apply it consistent with your workflow.

brentwilder, 2026-09-08:
As `empirical_line` is no longer the ideal solution for speeding up super-pixel solutions, as now we use the AOE solution, I am going to mark this issue as stale. But please re-open this if this is still relevant. 

I also do think that this is more of an IO-specific issue and not a segmentation issue. Generating loc files should be fairly straightforward, and now with the inclusion of more topography aware models in ISOFIT I don't think we should try and include too many hooks to make loc optional across different methods. There could be something here though for ISOIO - https://github.com/isofit/isoio

closed 2026-09-08 by brentwilder