isofit/isofit #161: Use ray instead of multiprocess in `empirical_line`

ashiklom, 2020-09-29:
I ran into some very strange interaction between `ray` and `multiprocess` while trying to run the `empirical_line` function (it tried to spawn multiple Python processes, each trying to be a ray master that re-started the full Isofit workflow. It was crazy! No idea what happened...). Regardless, since we seem to have committed to `ray` anyway (which is fine! I've grown to really like it!), I wonder if it would make sense to migrate the `empirical_line` code to use `ray` anyway. Here's what the diff looks like (plus, adding `import ray` at the top of the file and adding a `@ray.remote` decorator to the `_run_chunk` function):

```
@@ -389,24 +392,24 @@ def empirical_line(reference_radiance_file: str, reference_reflectance_file: str
     line_sections = np.linspace(0, n_input_lines, num=n_cores + 1, dtype=int)
 
     # Set up our pool
-    pool = multiprocessing.Pool(processes=n_cores)
     start_time = time.time()
     logging.info('Beginning empirical line inversions using {} cores'.format(n_cores))
+    ic = configs.create_new_config(isofit_config)
+    rayargs = {'ignore_reinit_error': True,
+               'local_mode': n_cores == 1,
+               "address": ic.implementation.ip_head,
+               "redis_password": ic.implementation.redis_password}
+    ray.init(**rayargs)
 
-    # Run the pool (or run serially)
     results = []
     for l in range(len(line_sections) - 1):
         args = (line_sections[l], line_sections[l + 1], reference_radiance_file, reference_reflectance_file,
                 reference_uncertainty_file, reference_locations_file, input_radiance_file,
                 input_locations_file, segmentation_file, isofit_config, output_reflectance_file,
                 output_uncertainty_file, radiance_factors, nneighbors, nodata_value,)
-        if n_cores != 1:
-            results.append(pool.apply_async(_run_chunk, args))
-        else:
-            _run_chunk(*args)
+        results.append(_run_chunk.remote(*args))
 
-    pool.close()
-    pool.join()
+    _ = ray.get(results)
 
     total_time = time.time() - start_time
     logging.info('Parallel empirical line inversions complete.  {} s total, {} spectra/s, {} spectra/s/core'.format(
```

I tested this locally and it seems to work pretty well. Let me know if this is acceptable and I can open a PR.

pgbrodrick, 2020-09-29:
Hi Alexey,

Absolutely no problem to make this switch - the only reason it hasn't been done is that I wrote the parallelism of empirical_line just before the ray switch, and because the relative additional benefit of ray in the empirical_line is low (given the efficiency and stability with the direct binary dumps I used) - but that's just why it hasn't been reached yet, no reason not to do it.

Looking at the diff you provide, I assume the the ic (must be a config?) would be replaced with direct input arguments, but then it generally looks good.  Please also be sure to add an explicit ray.shutdown() call at the end of the function.

If you're feeling particularly inspired, I also haven't gotten to parallelizing the segmentation and extraction codes yet, though it could certainly be done (tad bit tricky because of the chunking involved, as with the empirical line, but it would work).  Just tossing that out there in case you're itching for more parallelization work :-).

ashiklom, 2020-09-30:
Great! I'll try to put together a PR in the next few days.

commit e6260d18 (2020-10-02): Empirical line: Use Ray, not multiprocess

Closes #161
closed 2020-10-02 by pgbrodrick