isofit/isofit #163: Control output interleave, or force BIL interleave in segmentation code

ashiklom, 2020-10-05:
Currently, output interleave is hard-coded to BIP. However, it would be nice to be able to control this via config file. I propose the following small patch to do so:

```patch
diff --git a/isofit/configs/sections/output_config.py b/isofit/configs/sections/output_config.py
index 13829ca..9271aa9 100644
--- a/isofit/configs/sections/output_config.py
+++ b/isofit/configs/sections/output_config.py
@@ -97,6 +97,9 @@ class OutputConfig(BaseConfigSection):
         self._mcmc_samples_file_type = str
         self.mcmc_samples_file = None
 
+        self._interleave_type = str
+        self.interleave = "bip"
+
         self.set_config_options(sub_configdic)
 
     def _check_config_validity(self) -> List[str]:
diff --git a/isofit/core/fileio.py b/isofit/core/fileio.py
index 800b84a..52a592e 100755
--- a/isofit/core/fileio.py
+++ b/isofit/core/fileio.py
@@ -317,7 +317,7 @@ class IO:
                 n_rows=self.n_rows,
                 n_cols=self.n_cols,
                 n_bands=n_bands,
-                interleave='bip',
+                interleave=self.output.interleave,
                 dtype=np.float32,
                 wavelengths=self.meas_wl,
                 fwhm=self.meas_fwhm,
```

Or does this seriously mess up parallel IO?

The motivation for this is actually that the segmentation code demands that the interleave be BIL:

https://github.com/isofit/isofit/blob/d8d297455bc20e041ae9d845a4b6c3382413ee7b/isofit/utils/segment.py#L40-L42

As an alternative, would it make sense to force the interleave here? 

```python
 img_mm = in_img.open_memmap(interleave='bil', writable=False) 
```

In my tests, this seems to work as expected (i.e., the shapes of memmaps retrieved with different interleaves are correct and retrieved pixels from the same locations are identical), but I have no idea whether this incurs a significant performance hit or causes problems in downstream IO.

pgbrodrick, 2020-10-05:
This should work in principle, I need to do some testing in parallel though on different systems to make sure we don't end up with write-blocking.

Regarding the motivation, managing interleaves is something I've got open and will fix here shortly....we can actually force-read all input files as a bip, regardless of their actual format, with no cost penalty relative the current implementation.  This has come up in a few different contexts.  In short, bear with me here and I'll have the core issue fixed soon :-).

ashiklom, 2020-10-05:
> we can actually force-read all input files as a bip, regardless of their actual format, with no cost penalty relative the current implementation

Great! If that's the case, I'll force the interleave read in the segmentation code to BIL for now and stay tuned for an upstream fix. That also makes the output interleave much less urgent and important, so that can definitely wait. Let me know if you need help testing anything!

pgbrodrick, 2020-10-08:
Resolved through PR #166 .

closed 2020-10-08 by pgbrodrick