isofit/isofit #165: apply_oe.py should use multiple initializations

davidraythompson, 2020-10-07:
Maybe the default should be four initialization points at the extrema of water vapor and AOD values?
We should also update our "best practices" document to reflect that this is the preferred approach.

davidraythompson, 2020-10-07:
It looks like the documentation already includes this, so I only need to make apply_oe.py follow the convention

pgbrodrick, 2020-10-08:
Agreed, I think the 4 initialization points should work well.  I'd recommend spacing them _slightly_ inside of the bounds, just to avoid any corner-point weirdness, but with that in place I've found this to be stable in single-point inversions, so I can't imagine any issue with the empirical line!

I suspect the presolve can stay as is, to keep the complexity down, but up to you of course.

closed 2020-10-14 by pgbrodrick