isofit/isofit #174: numpy dependencies w/ tensorflow

serbinsh, 2020-10-26:
More of an FYI - though I suspect others have already seen this

```tensorflow 2.3.1 requires numpy<1.19.0,>=1.16.0, but you'll have numpy 1.19.1 which is incompatible.
Successfully installed PyWavelets-1.1.1 absl-py-0.10.0 astunparse-1.6.3 decorator-4.4.2 gast-0.3.3 google-auth-oauthlib-0.4.1 google-pasta-0.2.0 imageio-2.9.0 iniconfig-1.1.1 isofit joblib-0.17.0 keras-preprocessing-1.1.2 markdown-3.3.3 msgpack-0.6.2 networkx-2.5 oauthlib-3.1.0 opt-einsum-3.3.0 packaging-20.4 pep8-1.7.1 pillow-8.0.1 pluggy-0.13.1 py-1.9.0 pytest-6.1.1 ray-0.8.5 requests-oauthlib-1.3.0 scikit-image-0.17.2 scikit-learn-0.23.2 tensorboard-2.3.0 tensorboard-plugin-wit-1.7.0 tensorflow-2.3.1 tensorflow-estimator-2.3.0 termcolor-1.1.0 threadpoolctl-2.1.0 tifffile-2020.10.1 toml-0.10.1 werkzeug-1.0.1 wrapt-1.12.1
```

May need to update setup to define tighter numpy requirements?  I assume TF is for the new ML MODTRAN inversion?

serbinsh, 2020-10-26:
such as

```
pip install "numpy >=1.11,<1.19.0"
```

or is that error to be ignored?

pgbrodrick, 2020-10-26:
Hi Sean,

I saw this (occurred in the tensorflow addition in #171), and updated the corresponding installation instructions to:

```
pip install --editable /path/to/isofit --use-feature=2020-resolver
```

per the [README](https://github.com/isofit/isofit/blob/master/README.rst).  The above numpy lines should also work, though I figured it may be easier to just rely on the feature resolver - if you prefer what you use above as more explicit, I'm happy to take a PR on it.

closed 2020-10-26 by pgbrodrick