isofit/isofit #179: Cross-band correlations in posterior uncertainties

ashiklom, 2020-11-02:
We've discussed this in the past, and I recognize that this is an active area of research in your group, but I wanted to post it here to elicit more public discussion:

It would be great to have a way to include estimates of the off-diagonal elements of the posterior uncertainty distribution. These are particularly useful for analyses that require drawing reflectance samples from the posterior distribution, for instance for propagating uncertainties in atmospheric correction through to uncertainties in L3+ products. Here are a few ideas (some/all of which I've already heard your team mention):

1. For each pixel, convert the posterior variance-covariance matrix to a correlation matrix. Then, take the mean of these correlation matrices across all pixels within a scene and distribute this overall mean matrix.
2. Same as (1), but instead of 1 matrix per scene, distribute one mean correlation matrix per surface prior class. This would also require capturing (and distributing) which pixels use which priors.
3. Model the correlation structure as a function of wavelength lag, and distribute the parameters of this model for each pixel.

FWIW, in my own work, I'm currently assuming that the posterior uncertainty correlation structure is the same as the correlation structure in the posterior mean reflectance across pixels -- i.e., I'm just calculating the correlation matrix from the posterior mean reflectance estimates from an Isofit run. Since you all have thought a lot more about this, I'd be curious to hear your thoughts on just how wrong this is, and if you have suggestions for a better approach given Isofit's current design.

davidraythompson, 2020-11-02:
Fortunately we don't need to invent a solution! There is an exact closed-form expression for the posterior covariance of the state vector.  It's the \hat{S} matrix in previous publications.  You can get it by calling the forward_uncertainty() function of the inverse object.   </br></br> A different but relevant question is how to make sure the observation noise term captures the actual uncertainty in the observation including model discrepancies. The focus of much current research.

ashiklom, 2020-11-02:
Thanks David! It's great that you are already estimating this quantity -- it's definitely a strong advantage of Isofit! 

My question here is: How do you use the full `S` matrix in Isofit analyses on entire images (and eventually, in operational contexts)? Or, put another way, since storing a separate `S` matrix for every pixel would be impractical because it would result in astronomical file sizes, what is an effective way to summarize all the `S` matrices estimated from a scene? I see that the full matrix (along with many other useful quantities) can be dumped into `data_dump_file`s, but I'm assuming those are only used for single-pixel simulations...? On the other hand, Isofit currently provides the capability to store just the diagonal of this matrix in its `posterior_uncertainty_file`s (https://github.com/isofit/isofit/blob/master/isofit/core/fileio.py#L535), but this neglects all of the off-diagonal information. I guess I was wondering if it would be possible to have a sensible middle ground.

pgbrodrick, 2020-11-02:
Hi Alexey,

You're right that these are typically used for single-band analyses only (to-date!).  While we've discussed a variety of potential ways to reduce the information content down and store it, I haven't yet seen an implementation (though someone in the community may have one worked up).  But in discussions with other colleagues, our intuition aligns with yours, that based on the typical structure of \hat{S}, we should be able to reduce them down considerably while retaining virtually all info - some experimentation is needed to see what the effect is though.  It would be ideal to prove the efficacy of said method over a range of observations before integrating into the codebase, but that sounds tractable.  I'm sure David has more thoughts.

davidraythompson, 2020-11-02:
Great question.  If you find a solution, please submit a PR :)  We should think about about alternative representations that would go beyond a diagonal but not cost too much extra storage.  Fortunately the posterior covariances don't have too many extra degrees of freedom (Thompson et al., RSE 2018).  I think they would be amenable to a "diagonal plus low rank" representation - one example is equation 10 in https://hal.inria.fr/hal-01580733/document

ashiklom, 2020-11-02:
Thanks both!

> "diagonal plus low rank" representation

Yes, this is exactly the kind of thing I had in mind. I wonder how effective something relatively straightforward like only storing the first few principal components of `S_hat` would be -- that seems to be basically the approach suggested here: https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=6871413.

davidraythompson, 2020-11-02:
Yes, that could also work - i.e. just storing the top n eigenvalues of the Cholesky matrix.  Would the codebook of basis vectors be in the header of each scene, or would it be universal?