isofit/isofit #180: implementing adjacency effects

davidraythompson, 2020-11-03:
This is worth discussing because it will involve some cross-dependencies that we do not normally handle - specifically spatial dependencies across pixels.  I can imagine two ways we might do this.

1. Sequential.  Write out atmospheric coefficients to a file after a first pass atmospheric correction, and then apply adjacency effects as a postprocessing step.  Fast & simple.
2. Simultaneous.  The holy grail.  Treat the whole image as an uber-state vector, and each location's conditional likelihood score is defined using the reflectance in nearby pixels.  Solve the whole thing with a couple of passes through the image (essentially a [cyclic coordinate descent optimization](https://en.wikipedia.org/wiki/Coordinate_descent) on the uber-state.  We could also enforce a spatial GP prior on atmospheric terms for smoothness.

I think that #2 might actually  require less additional code than would appear at first glance - mostly just a new inversion method to apply the spatial averaging kernels and feed in appropriate local atmospheric priors during the update steps.  But is this an upgrade we should be pursuing right now?

davidraythompson, 2020-11-03:
Actually I think number 2 could be done by calling ISOFIT multiple times on the same image, each time reading in a spatially blurred state vector image produced by the previous iteration.  The blurred reflectance becomes the background spectrum for the adjacency effect calculation (maybe passed in as a member of the "geom" parameter to avoid interface changes?).  The blurred atmospheric state becomes the GP mean predicted atmospheric state.  I'm pretty sure the math would work out and all the cross-pixel dependencies could be handled outside ISOFIT proper.

pgbrodrick, 2020-11-03:
I'm a strong proponent of number 2 - I'm not sure what number one actually is going to give us, but I think I can imagine a parasitic case where iterative optimizations push you in a systematic direction (perhaps towards the priors?) that is not guaranteed to be towards the simultaneous optimum.  E.G., in the case where in the first solve you get:

g g g g g
g o o o g
g o x o g
g o o o g
g g g g g

In the second, the center pixel (let's call that the target), will get pushed in the 'o' direction, while the 'o' will get pushed in the 'g' direction.  Depending on the strength of the interaction, using 'o' as the prior adjacency for the target might be quite misleading.  But I could be wrong, I'd have to work out an actual example.

One option that I've been mulling to make number 2 tractable is to add in throw-away states to the state vector and cost function.  You could imagine taking some 'blurred' radiance (as the 'adjacent effect radiance') and some target radiance, and solving for the reflectance of each (with each term regularized with a separate prior-fitting penalty), making the assumption that the 'blurred' influences the target but not vice-versa.  Then just discard the adjacent portion of the state.  This has the advantage of using input radiance cleanly, but of course doesn't guarantee that the adjacent effect reflectances are consistent for different target pixels.  But it only involves doubling the statevector, with a corresponding term in the loss function....so it feels like a feasible first-order solution.

re: interface changes....updating the io to do some aggregate radiance/ob/loc grab doesn't actually feel that bad to me, relatively speaking (it's basically a slight IO modification to handle a window, and then just extra terms).  But holding onto a grid of radiance/obs/locs would require some substantial changes.

pgbrodrick, 2020-11-03:
I'm convinced - makes sense to start here no matter where we go afterward, and will be nice to have as a reference case to beat if it turns out complexity is needed (and ideally it just isn't).  Good test scenarios will of course be key!

I'm interested if @NimrodCarmon or others have thoughts.

NimrodCarmon, 2020-11-04:
Option 2 sounds great, although I also like option 1 which gives some additional flexibility for more physical formulations that might be relevant to adjacency (shadows\terrain\etc). Maybe a first run with a fast retrieval approach just to get those env parameters and a first guess to inform the uber state vector?

brentwilder, 2026-08-25:
A first iteration of adjacency effects has been added recently in https://github.com/isofit/isofit/pull/974 . It is currently implemented as something closer to what David was describing in 1) above ^^.  We use the heuristic pre-solve of the atmosphere to run the algebraic line per pixel (or superpixel), and then aggregate with respect to a computed adjacency range (which a function of sensor altitude and surface elevation).

closed 2026-08-25 by brentwilder