isofit/isofit #188: 6S template issues with latest 6S version

ashiklom, 2020-11-25:
I was trying to do some basic runs using the the sRTMnet but ran into the following error during 6S execution:

```
At line 1325 of file main.f (unit = 5, file = 'stdin')
Fortran runtime error: Bad integer for item 1 in list input
```

The issue seems to be that the current 6S input template doesn't work with the most recent (November 2018) version of 6S. After wading through 6S's web of `goto` statements (in the `main.f` file) and a lot of trial and error, I got to a working input file that looks like this:

```
0           (User defined geometry, `igeom`)
102.96113072253456 318.2379244151441 180 0 7 18     (SZA, SAZ, VZA, VAZ, month, day)
8           (User defined H2O, O3, `idatm`)
1.0, 0.3    (Values of H2O, O3)
0           (Aerosol model, `iaer`; 0 = no aerosols)
0.001       (Aerosol optical depth, `v`)
0.00        (target altitude, `xps`)
-705.00     (sensor altitude, `xpp`)
-2          (Select custom wavelength, `iwave`)
0.35 2.5    (Wavelengths for start, `wlinf`, and stop, `wlsup`)
0           (Homogeneous surface, `inhomo`)
0           (no directional effects, `idirec`)
0           (Constant ground reflectance, `igroun`)
0           (Value of ground reflectance, `ro`)
-1          (No atm. correction, `irapp`)
4           (Ground surface not polarized, `irop`)
```

The corresponding template string should be:

```
'''
0                   (User defined)
{solzen} {solaz} {viewzen} {viewaz} {month} {day}       (SZA, SAZ, VZA, VAZ, month, day)
8                   (User defined H2O, O3)
{H2OSTR}, {O3}      (Values of H2O, O3)
{aermodel}          (Aerosol model, `iaer`; 0  no aerosols)
{AOT550}            (Aerosol optical depth, `v`)
{elev:.2f}          (Target elevation, `xps`)
-{alt:.2f}          (Sensor altitude, `xpp`)
-2                  (Select custom wavelength, `iwave`)
{wlinf} {wlsup}     (Wavelengths for start, `wlinf`, and end, `wlsup`)
0                   (Homogeneous surface, `inhomo`)
0                   (No directional effects, `idirec`)
0                   (Constant ground reflectance, `igroun`)
0                   (Value of ground reflectance, `ro`)
-1                  (No atm. corrections selected, `irapp`)
4                   (Ground surface not polarized, `irop`)
'''
```

For reference, the current input template is here: https://github.com/isofit/isofit/blob/e9fe18e11bb6c5e07c819134cfaef80d575fd852/isofit/radiative_transfer/six_s.py#L33-L53

Note that I haven't tested the new template yet -- stay tuned for that.

pgbrodrick, 2020-11-25:
Hey Alexey - do you have a unit example I can try to reproduce (or even just a settings file you're using)?  I used the November 2018 release to develop sRTMnet (just straight from their [website](http://6s.ltdri.org/pages/downloads.html), and the templates should (in theory) match - I've gotten this to work across different systems, and so have some other colleagues who have tested it out.  It's possible though that some particular config parameter doesn't translate well and is leading to the issue.

ashiklom, 2020-11-25:
Here's the config file I was using. Though I should note that while this does run 6S, it produces a bunch of `NaN` values and ultimately crashes because at least one value is printed out as `******` (which gets converted to just whitespace by Isofit's LUT parser).

That said, I made a minor tweak to 6S to get it to compile with gcc > 8.0, but that may have broken something. I just learned about `-std=legacy`, which does allow it to compile, so I'm gonna try with that. The issues seem unrelated, but they might not be.

<details>
```json
{
  "forward_model": {
    "instrument": {
      "integrations": 1,
      "wavelength_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/hypertrace-data/wavelengths/aviris-classic.txt",
      "parametric_noise_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/hypertrace-data/noise/noise_coeff_sbg_cbe0.txt"
    },
    "surface": {
      "surface_category": "surface",
      "surface_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/hypertrace-data/priors/aviris-classic/surface_EMIT.mat",
      "wavelength_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/hypertrace-data/wavelengths/aviris-classic.txt"
    },
    "radiative_transfer": {
      "lut_grid": {
        "AOT550": [0.001, 0.123, 0.6],
        "H2OSTR": [1.0, 2.5, 3.25]
      },
      "statevector": {
        "AOT550": {
          "bounds": [0.001, 0.5],
          "scale": 0.01,
          "prior_mean": 0.05,
          "prior_sigma": 10.0,
          "init": 0.2
        },
        "H2OSTR": {
          "bounds": [0.01, 6],
          "scale": 0.01,
          "prior_mean": 1.5,
          "prior_sigma": 100,
          "init": 1.0
        }
      },
      "unknowns": {"H2O_ABSCO": 0.01},
      "radiative_transfer_engines": {
        "vswir": {
          "engine_name": "simulated_modtran",
          "engine_base_dir": "/Users/ashiklom/projects/models/6sV-2.1",
          "template_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/sRTMnet-h2o.json",
          "modtran_template_path": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/sRTMnet-modtran.json",
          "aerosol_template_path": "/Users/ashiklom/projects/sbg-uncertainty/isofit/data/aerosol_template.json",
          "earth_sun_distance_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/data/earth_sun_distance.txt",
          "irradiance_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/20151026_SantaMonica/data/prism_optimized_irr.dat",
          "emulator_aux_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/hypertrace-data/other/sRTMnet_v100/sRTMnet_v100_aux.npz",
          "emulator_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/hypertrace-data/other/sRTMnet_v100/sRTMnet_v100",
          "lut_names": ["H2OSTR", "AOT550"],
          "statevector_names": ["H2OSTR", "AOT550"],
          "lut_path": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/luts/sRTMnet-example"
        }
      }
    }
  },
  "implementation": {
    "inversion": {
      "windows": [
        [400, 1300],
        [1450, 1780],
        [1950, 2450]
      ],
      "simulation_mode": true
    },
    "mode": "simulation"
  },
  "input": {
    "reflectance_file": "/Users/ashiklom/projects/sbg-uncertainty/hypertrace-data/reflectance/reference/reference_reflectance"
  },
  "output": {
    "simulated_measurement_file": "/Users/ashiklom/projects/sbg-uncertainty/isofit/examples/py-hypertrace/output/sRTMnet-example/toa-radiance"
  }
}
```
</details>

ashiklom, 2020-11-25:
Could you send me one of your 6S template files from a run that worked? I trust that you and everyone else that's tested this knew what they were doing, so it's probably a problem on my end. It may be easier to back out what I'm doing wrong starting from a good input file.

pgbrodrick, 2020-11-25:
Ahh, cool this is helpful.  Nan results from sixs outputs are commonly a result of view / solar geometry issues, and modtran and sixs treat things slightly differently - quick look from above and I think your viewzen should be 180 - what you have.

For reference here's an example inp file that works.

```
0 (User defined)
33.571796227586155 201.4983624306357 17.34528282821151 71.2713073724544 3 22
8  (User defined H2O, O3) 
1.1998, 0.3 
1
0
0.01
-0.03 (target level)
-19.81 (sensor level)
-1.1998, -0.3
0.01
-2 
0.35
2.5
0 Homogeneous surface
0 (no directional effects)
0
0
0
-1 No atm. corrections selected
```

Also, I've worked in the sRTMnet option into apply_oe, so you should be able to use that utility to construct some working examples with populated LUTs that are written based on radiance, obs, and loc files - that might be a quick way to get to the issue.  If this doesn't work let me know and I'll package up a working example to send you offline.

ashiklom, 2020-11-25:
I think I found the issue -- my sensor altitude ("sensor level") was way out of whack. With SBG in mind, I was using the value for Landsat (705 km). But, 6S tries to be clever about this parameter and handles altitudes less than 100 km (typical airborne values; i.e., all of your work) in a special way:

```
c**********************************************************************c
c                                                                      c
c  xpp is the parameter to express the sensor altitude                 c
c                                                                      c
c                                                                      c
c         xpp= -1000  means that the sensor is a board a satellite     c
c         xpp=     0  means that the sensor is at the ground level     c
c                                                                      c
c                                                                      c
c     for aircraft simulations                                         c
c    -100< xpp <0  means you know the altitude of the sensor expressed c
c                  in kilometers units      			       c
c     this altitude is relative to the target altitude                 c
c                                                                      c
c     for aircraft simulations only, you have to give                  c
c	puw,po3   (water vapor content,ozone content between the       c
c                  aircraft and the surface)                           c
c	taerp     (the aerosol optical thickness at 550nm between the  c
c                  aircraft and the surface)                           c
c    if these data are not available, enter negative values for all    c
c    of them, puw,po3 will then be interpolated from the us62 standard c
C    profile according to the values at ground level. Taerp will be    c
c    computed according to a 2km exponential profile for aerosol.      c
c**********************************************************************c
```

When I run with a sensor elevation less than 100 km, it works fine.

pgbrodrick, 2020-11-25:
Nice find - definitely the issue and the proposed solutions make sense - I'll dig into the code a bit more to see if there's any substantial benefit to swapping it into satellite mode and go from there.  Thanks for flagging!

pgbrodrick, 2023-01-10:
This has been resolved, there is a cap at 99km in place since #246.  Closing.

closed 2023-01-10 by pgbrodrick