isofit/isofit #189: Update GDAL import

ashiklom, 2020-11-30:
When recently firing up Isofit from a fresh conda environment (Python 3.8.5; GDAL 3.2.0; see details below for full environment), I noticed that the `import gdal` statement in `apply_oe.py` didn't work (`ModuleNotFoundError: No module named 'gdal'`), but replacing it with `from osgeo import gdal` seemed to work just fine.

The [PyPI GDAL website](https://pypi.org/project/GDAL/) says that `from osgeo import gdal` is the correct way to import and says that the old way (`import gdal`) will be supported "until at least GDAL 2.1". My guess is that they have deprecated the old way in a more recent version (GDAL 3?).

If you want to be robust to older versions, you can apparently do something like:

```python
try:
  from osgeo import gdal
except ImportError:
  import gdal
```

...though it may be safer to specify a minimum GDAL version requirement (>=2.1) and just stick with the more modern usage. Alternatively, you could set an upper limit on the GDAL versions, though my sense is that the OSGEO folks have made a lot of significant improvements to GDAL, PROJ, and their other tools very recently that Isofit may want to be able to use.

<details>

```
# packages in environment at /Users/ashiklom/opt/anaconda3/envs/isofit:
#
# Name                    Version                   Build  Channel
absl-py                   0.11.0                   pypi_0    pypi
aiohttp                   3.7.3                    pypi_0    pypi
astunparse                1.6.3                    pypi_0    pypi
async-timeout             3.0.1                    pypi_0    pypi
attrs                     20.3.0                   pypi_0    pypi
beautifulsoup4            4.9.3                    pypi_0    pypi
ca-certificates           2020.10.14                    0
cachetools                4.1.1                    pypi_0    pypi
certifi                   2020.11.8        py38hecd8cb5_0
chardet                   3.0.4                    pypi_0    pypi
click                     7.1.2                    pypi_0    pypi
colorama                  0.4.4                    pypi_0    pypi
cycler                    0.10.0                   pypi_0    pypi
decorator                 4.4.2                    pypi_0    pypi
filelock                  3.0.12                   pypi_0    pypi
gast                      0.3.3                    pypi_0    pypi
gdal                      3.2.0                    pypi_0    pypi
google                    3.0.0                    pypi_0    pypi
google-auth               1.23.0                   pypi_0    pypi
google-auth-oauthlib      0.4.2                    pypi_0    pypi
google-pasta              0.2.0                    pypi_0    pypi
grpcio                    1.33.2                   pypi_0    pypi
h5py                      2.10.0                   pypi_0    pypi
idna                      2.10                     pypi_0    pypi
imageio                   2.9.0                    pypi_0    pypi
iniconfig                 1.1.1                    pypi_0    pypi
isofit                    2.6.0                    pypi_0    pypi
joblib                    0.17.0                   pypi_0    pypi
jsonschema                3.2.0                    pypi_0    pypi
keras-preprocessing       1.1.2                    pypi_0    pypi
kiwisolver                1.3.1                    pypi_0    pypi
libcxx                    10.0.0                        1
libedit                   3.1.20191231         h1de35cc_1
libffi                    3.3                  hb1e8313_2
markdown                  3.3.3                    pypi_0    pypi
matplotlib                3.3.3                    pypi_0    pypi
msgpack                   0.6.2                    pypi_0    pypi
multidict                 5.0.2                    pypi_0    pypi
ncurses                   6.2                  h0a44026_1
networkx                  2.5                      pypi_0    pypi
numpy                     1.18.5                   pypi_0    pypi
oauthlib                  3.1.0                    pypi_0    pypi
openssl                   1.1.1h               haf1e3a3_0
opt-einsum                3.3.0                    pypi_0    pypi
packaging                 20.7                     pypi_0    pypi
pandas                    1.1.4                    pypi_0    pypi
pep8                      1.7.1                    pypi_0    pypi
pillow                    8.0.1                    pypi_0    pypi
pip                       20.2.4           py38hecd8cb5_0
pluggy                    0.13.1                   pypi_0    pypi
protobuf                  3.14.0                   pypi_0    pypi
py                        1.9.0                    pypi_0    pypi
py-spy                    0.3.3                    pypi_0    pypi
pyasn1                    0.4.8                    pypi_0    pypi
pyasn1-modules            0.2.8                    pypi_0    pypi
pyparsing                 2.4.7                    pypi_0    pypi
pyrsistent                0.17.3                   pypi_0    pypi
pytest                    6.1.2                    pypi_0    pypi
python                    3.8.5                h26836e1_1
python-dateutil           2.8.1                    pypi_0    pypi
pytz                      2020.4                   pypi_0    pypi
pywavelets                1.1.1                    pypi_0    pypi
pyyaml                    5.3.1                    pypi_0    pypi
ray                       0.8.5                    pypi_0    pypi
readline                  8.0                  h1de35cc_0
redis                     3.4.1                    pypi_0    pypi
requests                  2.25.0                   pypi_0    pypi
requests-oauthlib         1.3.0                    pypi_0    pypi
rsa                       4.6                      pypi_0    pypi
scikit-image              0.17.2                   pypi_0    pypi
scikit-learn              0.23.2                   pypi_0    pypi
scipy                     1.5.4                    pypi_0    pypi
setuptools                50.3.1           py38hecd8cb5_1
six                       1.15.0                   pypi_0    pypi
soupsieve                 2.0.1                    pypi_0    pypi
spectral                  0.22.1                   pypi_0    pypi
sqlite                    3.33.0               hffcf06c_0
tensorboard               2.4.0                    pypi_0    pypi
tensorboard-plugin-wit    1.7.0                    pypi_0    pypi
tensorflow                2.3.1                    pypi_0    pypi
tensorflow-estimator      2.3.0                    pypi_0    pypi
termcolor                 1.1.0                    pypi_0    pypi
threadpoolctl             2.1.0                    pypi_0    pypi
tifffile                  2020.11.26               pypi_0    pypi
tk                        8.6.10               hb0a8c7a_0
toml                      0.10.2                   pypi_0    pypi
typing-extensions         3.7.4.3                  pypi_0    pypi
urllib3                   1.26.2                   pypi_0    pypi
werkzeug                  1.0.1                    pypi_0    pypi
wheel                     0.35.1             pyhd3eb1b0_0
wrapt                     1.12.1                   pypi_0    pypi
xxhash                    2.0.0                    pypi_0    pypi
xz                        5.2.5                h1de35cc_0
yarl                      1.6.3                    pypi_0    pypi
zlib                      1.2.11               h1de35cc_3
```

</details>

davidraythompson, 2020-11-30:
Good catch.  Of the options, I think I favor requiring the minimum GDAL version, for simplicity.

pgbrodrick, 2020-11-30:
Agreed on the requirement approach, and gdal 2 predates the initialization of this repository by several years, so we shouldn't be hurting anyone with an updated requirement.  gdal is currently actually outside of our setup requirements, as it's only necessary in the utilities, and this fixes that as well.

I'll submit a PR shortly.

pgbrodrick, 2020-12-11:
Resolved via PR #190 .

closed 2020-12-11 by pgbrodrick