isofit/isofit #195: libRadtran version for hypertrace workflow

pgbrodrick, 2020-12-29:
I've just started getting errors in the hypertrace workflow, which look like they stem from the wget pulling a slightly different libradtran version than is requested.  The error appears as:

```
Run wget -nv http://www.libradtran.org/download/libRadtran-2.0.3.tar.gz
wget -nv http://www.libradtran.org/download/libRadtran-2.0.3.tar.gz
tar -xf libRadtran-2.0.3.tar.gz
cd libRadtran-2.0.3
wget -nv http://www.meteo.physik.uni-muenchen.de/~libradtran/lib/exe/fetch.php?media=download:reptran_2017_all.tar.gz -O reptran-2017-all.tar.gz
tar -xf reptran-2017-all.tar.gz
PYTHON=$(which python2) ./configure --prefix=$(pwd)
# For some reason, uvspecfunction compilation fails...
make --ignore-errors
shell: /bin/bash -e {0}
env:
  pythonLocation: /opt/hostedtoolcache/Python/3.8.6/x64
  LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.8.6/x64/lib
2020-12-29 13:28:15 URL:http://www.libradtran.org/download/libRadtran-2.0.4.tar.gz [71453375/71453375] -> "libRadtran-2.0.3.tar.gz" [1]
/home/runner/work/_temp/06b19355-b3c5-4cfb-8321-99b1fcbb1f74.sh: line 3: cd: libRadtran-2.0.3: No such file or directory
Error: Process completed with exit code 1.
```

It seems that the ibRadtran-2.0.3.tar.gz file actually contains 2.0.4, which then changes the cd.

Changing out line [39 of the workflow](https://github.com/isofit/isofit/blob/1e7c3680d12080b53821c76713de5483a2442900/.github/workflows/py-hypertrace.yml#L39) from 

```
cd libRadtran-2.0.3
```
to
```
cd libRadtran-2.*/
```

would fix the issue, but I'm not sure if they current hypertrace workflow is compatible with different libradtran versions.  @ashiklom , any thoughts?

ashiklom, 2021-01-05:
Thanks for catching this! A safer fix might actually be to force the older version by using this URL (note the `history` part):

http://www.libradtran.org/download/history/libRadtran-2.0.3.tar.gz

That said, I think we can safely upgrade Hypertrace to use libradtran 2.0.4 -- I checked their release notes and it looks like it's just bugfixes and a few new solvers, but no breaking changes. I'll try to pull together a patch for that in the next few days.

FWIW, for my most recent work, I've started experimenting with using Hypertrace with sRTMnet, which is way faster and, if you're to be believed 😉 , more accurate. I have a PR with that forthcoming sometime soon, at which point I think we can make LibRadtran the "experimental" choice.

pgbrodrick, 2021-01-05:
Aha!  That seems safe, and either 2.04 or the alternate URL works for a patch for now.

Re: sRTMnet integration, awesome news!   I won't yet make any explicit statements regarding direct superiority as compared to libradtran, but hopefully it lets you expand to higher dimensional LUTs, which is almost certainly good :-).

pgbrodrick, 2021-01-21:
Resolved with #199.

closed 2021-01-21 by pgbrodrick