isofit/isofit #198: MODTRAN vapor ceiling

davidraythompson, 2020-12-30:
I have been having trouble with the MODTRAN humidity limit on scenes with large elevation gradients; Mt. Shasta climbs several thousand feet, and no single H2O LUT range can capture the vapor at its base and still be physically plausible at the peak.  The simplest solution is to institute H2OOPT="+" in apply_oe.py.  Would we run any risks for doing this?

pgbrodrick, 2020-12-30:
I'm linking to #120, where I expressed similar challenges (along a similarly challenging elevation grid).

Exceeding 100% RH seems the best solution all things considered - I've been somewhat reluctant to implement simply because this could lead to non-physical solutions in some locations (though hopefully the optimization can sort it out - I just haven't made time to test).  Long term, a more advances strategy than a complete LUT intersection is probably the best way forward, but that's more work.

If this is becoming prohibitive, I say we give it a shot.

NimrodCarmon, 2020-12-30:
I would advise to double check the "single scattering" column in the .chn file, and make sure it doesn't have negative values when the H2OOPT option is turned on. Although we use the "Total Radiance" column when we calculate the atmospheric reflectance, I'm almost sure that modtran internally uses the single scattering and multiple scattering products when calculating that radiance.

davidraythompson, 2020-12-30:
Test in progress, stand by.

brentwilder, 2026-08-25:
This is now addressed by using derived MODTRAN-based polynomials (https://github.com/isofit/isofit/pull/689) to do a per-pixel check prior to OE based on the elevation (https://github.com/isofit/isofit/pull/982 , https://github.com/isofit/isofit/pull/976).

closed 2026-08-25 by brentwilder