isofit/isofit #205: 6S read error

adamchlus, 2021-01-29:
When running apply_oe.py an error occurs in [six_s.py](https://github.com/isofit/isofit/blob/master/isofit/radiative_transfer/six_s.py)  when splitting lines from 6S output files, example output at the bottom.

The first few wavelengths have "******" in two columns which causes [this line](https://github.com/isofit/isofit/blob/master/isofit/radiative_transfer/six_s.py#L224):

```python
ln = ln.replace('*', ' ').strip()
```

to drop these values, resulting in 9 values instead of the expected 11.

Quick fix, replace "******" with "NaN" before [line 224](https://github.com/isofit/isofit/blob/master/isofit/radiative_transfer/six_s.py#L224) 
```python
ln = ln.replace('******', 'NaN')            
```
Example 6S output file:
```
* wave   total  total  total  total  atm.   swl    step   sbor   dsol   toar  *
*        gas    scat   scat   spheri intr                                     *
*        trans  down   up     albedo refl                                     *
*0.3500 1.0069 6.8795 0.8654 0.3335 ******  989.8 0.0000 0.5000 0.9687 ****** *
*0.3525 1.0045 6.6597 0.8670 0.3284 ******  997.9 0.0000 1.0000 0.9687 ****** *
*0.3550 1.0029 6.4484 0.8687 0.3234 ****** 1125.8 0.0000 1.0000 0.9687 ****** *
*0.3575 1.0015 6.2453 0.8704 0.3186 ******  863.4 0.0000 1.0000 0.9687 ****** *
*0.3600 1.0010 6.0498 0.8720 0.3139 ****** 1009.7 0.0000 1.0000 0.9687 ****** *
*0.3625 1.0006 5.8618 0.8736 0.3092 ****** 1029.7 0.0000 1.0000 0.9687 ****** *
*0.3650 1.0000 5.6809 0.8752 0.3047 ****** 1113.7 0.0000 1.0000 0.9687 ****** *
*0.3675 1.0000 5.5067 0.8769 0.3002 ****** 1219.5 0.0000 1.0000 0.9687 ****** *
*0.3700 1.0000 5.3390 0.8785 0.2959 ****** 1236.7 0.0000 1.0000 0.9687 ****** *
*0.3725 1.0000 5.1774 0.8801 0.2916 ****** 1095.5 0.0000 1.0000 0.9687 ****** *
*0.3750 1.0000 5.0218 0.8817 0.2875 ******  994.4 0.0000 1.0000 0.9687 ****** *
*0.3775 1.0000 4.8719 0.8832 0.2834 ****** 1264.8 0.0000 1.0000 0.9687 ****** *
*0.3800 1.0000 4.7274 0.8848 0.2794 ****** 1134.7 0.0000 1.0000 0.9687 ****** *
*0.3825 1.0000 4.5880 0.8864 0.2755 ******  844.6 0.0000 1.0000 0.9687 ****** *
*0.3850 1.0000 4.4537 0.8879 0.2717 ****** 1004.3 0.0000 1.0000 0.9687 ****** *
*0.3875 1.0000 4.3241 0.8895 0.2679 ****** 1022.5 0.0000 1.0000 0.9687 ****** *
*0.3900 1.0000 4.1991 0.8910 0.2642 ****** 1192.1 0.0000 1.0000 0.9687 ****** *
*0.3925 1.0000 4.0784 0.8926 0.2606 -.9783  992.9 0.0000 1.0000 0.9687 -.9783 *

```

pgbrodrick, 2021-01-29:
Hi Adam,

This raises the excellent point that we should implement a read-in safeguard, so if anything's wrong with the 6S result we at least complete the read before throwing an error.  The above suggestion, coupled with a raised attribute error on NaN's should do the trick.

That said, from the results above it looks like something's off in the input settings.  Good things to check are the solar / flight geometries (we do a 180 - inside of the 6S wrapper to match MODTRAN).

adamchlus, 2021-02-02:
Thanks Phil, after much troubleshooting it turns out I swapped the latitude and longitude in the location date cube.

closed 2021-07-18 by pgbrodrick

ashiklom, 2022-09-15:
I suggest revisiting this. Yoseline and I encountered a similar error when trying to use `apply_oe.py` to atmospherically correct some NEON data.

We only encountered the error while building the water vapor look up tables, but interestingly, we hit it for only a subset of specific water vapor contents (not all the highest or lowest), and in a subset of very specific wavelengths:

```
# From inside the LUT directory
#> grep -E " \*{6} " H2OSTR-*
              * wave   total  total  total  total  atm.   swl    step   sbor   dsol   toar  *
              *        gas    scat   scat   spheri intr                                     *
              *        trans  down   up     albedo refl                                     *
H2OSTR-0.6700:*1.5025 ****** 1.0095 0.9995 0.0026 -.0004  276.9 0.0000 1.0000 0.9689 -.0004 *
H2OSTR-1.3400:*2.1650 ****** 1.0069 0.9994 0.0005 -.0001   82.2 0.0000 1.0000 0.9689 -.0001 *
H2OSTR-1.3400:*2.1675 ****** 1.0069 0.9994 0.0005 -.0001   82.5 0.0000 1.0000 0.9689 -.0001 *
H2OSTR-2.6700:*2.4275 ****** 1.0059 0.9995 0.0005 -.0001   56.0 0.0000 1.0000 0.9689 -.0001 *
H2OSTR-2.6700:*2.4300 ****** 1.0059 0.9995 0.0005 -.0001   55.8 0.0000 1.0000 0.9689 -.0001 *
H2OSTR-2.6700:*2.4325 ****** 1.0059 0.9995 0.0005 -.0001   55.6 0.0000 1.0000 0.9689 -.0001 *
H2OSTR-5.3300:*0.7400 ****** 1.0595 0.9977 0.0298 -.0035 1233.5 0.0000 1.0000 0.9689 -.0035 *
H2OSTR-5.3300:*1.5300 ****** 1.0091 0.9996 0.0025 -.0004  267.7 0.0000 1.0000 0.9689 -.0004 *
```

The corresponding input files are in the details below.

In addition to @adamchlus 's suggestion of replacing the `******` with `NaN`, it may make sense to add some additional checks into the code and throw a more informative error/warning (e.g., if more than some fraction of inputs is NaN or non-positive). This may well be a configuration error on our part, but if so, it would be useful to have a more informative error message.

----

<details>

## `LUT_H2OSTR-0.6700.inp`

```
0 (User defined)
111.14356560357427 102.79116596010186 8.774122593728976 154.15323019918742 7 24
8  (User defined H2O, O3)
0.67, 0.3
1
0
0.01
-0.01 (target level)
-1.01 (sensor level)
-0.67, -0.3
0.01
-2
0.35
2.5
0 Homogeneous surface
0 (no directional effects)
0
0
0
-1 No atm. corrections selected
```

## `LUT_H2OSTR-1.3400.inp`

```
0 (User defined)
111.14356560357427 102.79116596010186 8.774122593728976 154.15323019918742 7 24
8  (User defined H2O, O3)
1.34, 0.3
1
0
0.01
-0.01 (target level)
-1.01 (sensor level)
-1.34, -0.3
0.01
-2
0.35
2.5
0 Homogeneous surface
0 (no directional effects)
0
0
0
-1 No atm. corrections selected
```

## `LUT_H2OSTR-2.6700.inp`

```
0 (User defined)
111.14356560357427 102.79116596010186 8.774122593728976 154.15323019918742 7 24
8  (User defined H2O, O3)
2.67, 0.3
1
0
0.01
-0.01 (target level)
-1.01 (sensor level)
-2.67, -0.3
0.01
-2
0.35
2.5
0 Homogeneous surface
0 (no directional effects)
0
0
0
-1 No atm. corrections selected
```

## `LUT_H2OSTR-5.3300.inp`

```
0 (User defined)
111.14356560357427 102.79116596010186 8.774122593728976 154.15323019918742 7 24
8  (User defined H2O, O3)
5.33, 0.3
1
0
0.01
-0.01 (target level)
-1.01 (sensor level)
-5.33, -0.3
0.01
-2
0.35
2.5
0 Homogeneous surface
0 (no directional effects)
0
0
0
-1 No atm. corrections selected
```

</details>

pgbrodrick, 2022-09-15:
Thanks for re-opening.  I think safeguarding with an informative error message is the way to go - in every instance where I've encountered this, it's been an issue with the observation geometry passed.  But it is not clear to the user from the current failure mode that this is the issue.

ashiklom, 2022-09-15:
> in every instance where I've encountered this, it's been an issue with the observation geometry passed

Indeed — after investigating further, our specific issue was also because of invalid geometries. In our case, we were passing projected coordinates, not lat-lon coordinates. So issues with latitude/longitude seem to be a recurring theme here (documenting for future users).