isofit/isofit #217: Jupyter notebooks

davidraythompson, 2021-03-15:
ray is giving me problems when I try to build a forward model from within a jupyter notebook.  Any ideas?

---------------------------------------------------------------------------
RayConnectionError                        Traceback (most recent call last)
<ipython-input-3-6cea0112f630> in <module>
      9     cfg = Config(config)
     10     x = np.r_[rfl, 0.1, 1.4, 160]
---> 11     fm = forward.ForwardModel(cfg)

~/src/isofit/isofit/core/forward.py in __init__(self, full_config)
     71 
     72         # Build the radiative transfer model
---> 73         self.RT = RadiativeTransfer(self.full_config)
     74 
     75         # Build the surface model - this is a bit ugly with the conditionals, but the surface config should already be

~/src/isofit/isofit/radiative_transfer/radiative_transfer.py in __init__(self, full_config)
     63                 rte = SixSRT(rte_config, full_config)
     64             elif rte_config.engine_name == 'simulated_modtran':
---> 65                 rte = SimulatedModtranRT(rte_config, full_config)
     66             else:
     67                 # Should never get here, checked in config

~/src/isofit/isofit/radiative_transfer/sRTMnet.py in __init__(self, engine_config, full_config)
    108         sixs_rte = SixSRT(sixs_config, full_config, build_lut_only = False, 
    109                           wavelength_override=simulator_wavelengths, fwhm_override=np.ones(n_simulator_chan)*2.,
--> 110                           modtran_emulation=True)
    111         self.solar_irr = sixs_rte.solar_irr
    112         self.esd = sixs_rte.esd

~/src/isofit/isofit/radiative_transfer/six_s.py in __init__(self, engine_config, full_config, build_lut, build_lut_only, wavelength_override, fwhm_override, modtran_emulation)
    126 
    127         if build_lut:
--> 128             self.build_lut()
    129 
    130     def find_basedir(self, config: RadiativeTransferEngineConfig):

~/src/isofit/isofit/radiative_transfer/six_s.py in build_lut(self, rebuild)
    251     def build_lut(self, rebuild=False):
    252 
--> 253         TabularRT.build_lut(self, rebuild)
    254 
    255         sixs_outputs = []

~/src/isofit/isofit/radiative_transfer/look_up_tables.py in build_lut(self, rebuild)
    194 
    195             # Make the LUT calls (in parallel if specified)
--> 196             results = ray.get([spawn_rt.remote(rebuild_cmd, self.lut_dir) for rebuild_cmd in rebuild_cmds])
    197 
    198 

~/src/isofit/isofit/radiative_transfer/look_up_tables.py in <listcomp>(.0)
    194 
    195             # Make the LUT calls (in parallel if specified)
--> 196             results = ray.get([spawn_rt.remote(rebuild_cmd, self.lut_dir) for rebuild_cmd in rebuild_cmds])
    197 
    198 

~/src/anaconda37/anaconda3/lib/python3.7/site-packages/ray/remote_function.py in _remote_proxy(*args, **kwargs)
     93         @wraps(function)
     94         def _remote_proxy(*args, **kwargs):
---> 95             return self._remote(args=args, kwargs=kwargs)
     96 
     97         self.remote = _remote_proxy

~/src/anaconda37/anaconda3/lib/python3.7/site-packages/ray/remote_function.py in _remote(self, args, kwargs, num_return_vals, is_direct_call, num_cpus, num_gpus, memory, object_store_memory, resources, max_retries)
    152         """Submit the remote function for execution."""
    153         worker = ray.worker.global_worker
--> 154         worker.check_connected()
    155 
    156         # If this function was not exported in this session and job, we need to

~/src/anaconda37/anaconda3/lib/python3.7/site-packages/ray/worker.py in check_connected(self)
    207         """
    208         if not self.connected:
--> 209             raise RayConnectionError("Ray has not been started yet. You can "
    210                                      "start Ray with 'ray.init()'.")
    211 

RayConnectionError: Ray has not been started yet. You can start Ray with 'ray.init()'.

pgbrodrick, 2021-03-15:
From the above, it looks like ray has just not been initialized in the notebook (which normally happens in isofit.py, so it makes sense that if you're just setting up a forward model this might not be done).

If no LUT needs to be built, you may make it through without issue, but otherwise you'll need to initialize first.  If that doesn't solve things, could you post the start of the notebook so I can try and reproduce?

davidraythompson, 2021-03-16:
This worked.

closed 2021-03-16 by davidraythompson