isofit/isofit #220: Zenodo DOI

davidraythompson, 2021-03-17:
To conform to Copernicus manuscript requirements, I'd like to issue a DOI to this repository.  We set this up automatically via [zenodo](https://docs.github.com/en/github/creating-cloning-and-archiving-repositories/referencing-and-citing-content
).  A new DOI will be issued automatically for each release.  
If nobody objects, I'll go ahead.

serbinsh, 2021-03-17:
Yes!  Strong support

-------------------------------------------------------------------------
Shawn P. Serbin
Scientist
Environmental & Climate Sciences Department
Brookhaven National Laboratory
Upton, NY 11973-5000

Phone: +1-631-344-3165
Fax: +1-631-344-2060
Email: ***@***.******@***.***>
Bio:  https://www.bnl.gov/staff/shawnserbin/index.php
TEST Group: https://www.bnl.gov/envsci/testgroup/
ORCiD: http://orcid.org/0000-0003-4136-8971
Twitter: @doctorjackpine, @test_bnl
-------------------------------------------------------------------------

pgbrodrick, 2021-03-17:
Absolutely concur.

davidraythompson, 2021-03-17:
Would anyone object to my declaring release 2.8.0? This will generate the DOI.

serbinsh, 2021-03-17:
No. @ashiklom do we want to consider 2.8.0 the official first release of "Hypertrace" as well? So in other words the 2.8.0 would include or call out hypertrace 1.0 or whatever....just a thought

davidraythompson, 2021-03-17:
[Done!](https://zenodo.org/record/4614338#.YFJhMpNKgrE) Thanks, all.

closed 2021-03-17 by davidraythompson

serbinsh, 2021-03-17:
we should add "the badge" to the main readme too perhaps