isofit/isofit #237: Compatibility with MacBooks using M1 chip

kelvinmleung, 2021-06-17:
I've been trying to install isofit on a macbook that uses the Apple silicon M1 chip, but there seem to be some compatibility issues. Specifically, there's an error for a package called llvmlite that seems to be because of the M1 chip according to eg. https://github.com/numba/llvmlite/issues/693 (error pasted below)

Does anyone know if this might be a compatibility issue and how to deal with it?
I do have another system I can run this on, but @davidraythompson suggested I bring up this issue here.

Using pip install isofit, I get the error:
```
Building wheels for collected packages: llvmlite
  Building wheel for llvmlite (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/kmleung/miniforge3/bin/python3.8 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/rt/h312492j1w32dwflc7yy6tsm0000gn/T/pip-install-mcvdpsc3/llvmlite_7f994fcd2525439aae766be428886554/setup.py'"'"'; __file__='"'"'/private/var/folders/rt/h312492j1w32dwflc7yy6tsm0000gn/T/pip-install-mcvdpsc3/llvmlite_7f994fcd2525439aae766be428886554/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/rt/h312492j1w32dwflc7yy6tsm0000gn/T/pip-wheel-ufhq_lyh
       cwd: /private/var/folders/rt/h312492j1w32dwflc7yy6tsm0000gn/T/pip-install-mcvdpsc3/llvmlite_7f994fcd2525439aae766be428886554/
  Complete output (11 lines):
  running bdist_wheel
  /Users/kmleung/miniforge3/bin/python3.8 /private/var/folders/rt/h312492j1w32dwflc7yy6tsm0000gn/T/pip-install-mcvdpsc3/llvmlite_7f994fcd2525439aae766be428886554/ffi/build.py
  LLVM version... Traceback (most recent call last):
    File "/private/var/folders/rt/h312492j1w32dwflc7yy6tsm0000gn/T/pip-install-mcvdpsc3/llvmlite_7f994fcd2525439aae766be428886554/ffi/build.py", line 220, in <module>
      main()
    File "/private/var/folders/rt/h312492j1w32dwflc7yy6tsm0000gn/T/pip-install-mcvdpsc3/llvmlite_7f994fcd2525439aae766be428886554/ffi/build.py", line 214, in main
      main_posix('osx', '.dylib')
    File "/private/var/folders/rt/h312492j1w32dwflc7yy6tsm0000gn/T/pip-install-mcvdpsc3/llvmlite_7f994fcd2525439aae766be428886554/ffi/build.py", line 134, in main_posix
      raise RuntimeError(msg) from None
  RuntimeError: Could not find a `llvm-config` binary. There are a number of reasons this could occur, please see: https://llvmlite.readthedocs.io/en/latest/admin-guide/install.html#using-pip for help.
  error: command '/Users/kmleung/miniforge3/bin/python3.8' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for llvmlite
  Running setup.py clean for llvmlite
Failed to build llvmlite
```

pgbrodrick, 2021-06-17:
Hey, thanks for posting.

This is a bit tricky.  We have no direct numba dependencies here (they were purged some time ago), meaning it must be a package sub-dependency.  If possible, I'm happy to consider altering the package structure to mitigate this, but we need to ID the location first.  I can try to hunt down the dependency, but if you want to go through the packages listed in setup.py, and try pip installing them one-by-one, and repost with which package(s) fail(s), that would help enormously (otherwise I can try and set up an M1-based VM, but that may take a bit).

kelvinmleung, 2021-06-18:
I tried pip installing all the packages and the only one that doesn't work is ray. I tried a few different ways of installing it but it seems like there's no support yet for non-intel Mac OS, although I could just be missing something. Please let me know if there's have any work-around.

pgbrodrick, 2021-06-21:
Can you do a quick try install ray 1.4.0?  Reading through some closed Ray issues, it looks like it may have been addressed (https://github.com/ray-project/ray/pull/14299).  If that does solve the issue, I'll update the setup.py file to 1.4.

kelvinmleung, 2021-06-22:
When I do pip install ray or pip install ray==1.4.0, I get the following error
```
ERROR: Could not find a version that satisfies the requirement ray (from versions: none)
ERROR: No matching distribution found for ray
```
I tried following the steps here https://github.com/ray-project/ray/issues/12742, but building bazel from source also gave some errors that I'm not sure how to fix.

pgbrodrick, 2021-06-22:
I'd suggest updating your pip, or starting a new scratch environment for testing - ray 1.4.0 definitely exists on pypi (https://pypi.org/project/ray/), and I just confirmed with a local environment install.  So if a pip instal ray==1.4.0 isn't working, there may be broader issues :-).

pgbrodrick, 2022-04-13:
Looks like ray>=1.8 has support for the M1 chip, (https://discuss.ray.io/t/ray-with-apple-m1/624/14), so I'm going to go ahead and mark this solved.  Please let me know if it continues to be an issue.

closed 2022-04-13 by pgbrodrick