isofit/isofit #239: RayOutOfMemoryError - ray workers using too much memory

winstonolson, 2021-06-22:
It looks like the ray workers are using roughly 2G of memory each, which is much more than should be needed to process a single spectrum.  This is causing ray to run out of memory in some cases and throw the error below. 

```
(pid=2960) bash /data/work/jobs/2021/06/17/20/44/job-isofit_ubuntu__sister-dev-20210617T204428.349550Z/lut_h2o/LUT_H2OSTR-0.0100.sh
Traceback (most recent call last):
 File “/app/isofit/isofit/utils/apply_oe.py”, line 1387, in <module>
  main()
 File “/app/isofit/isofit/utils/apply_oe.py”, line 293, in main
  retrieval_h2o.run()
 File “/app/isofit/isofit/core/isofit.py”, line 159, in run
  indices_to_run))
 File “/opt/conda/lib/python3.7/site-packages/ray/util/actor_pool.py”, line 91, in map_unordered
  yield self.get_next_unordered()
 File “/opt/conda/lib/python3.7/site-packages/ray/util/actor_pool.py”, line 210, in get_next_unordered
  return ray.get(future)
 File “/opt/conda/lib/python3.7/site-packages/ray/_private/client_mode_hook.py”, line 47, in wrapper
  return func(*args, **kwargs)
 File “/opt/conda/lib/python3.7/site-packages/ray/worker.py”, line 1456, in get
  raise value.as_instanceof_cause()
ray.exceptions.RayTaskError(RayOutOfMemoryError): ray::Worker.run_set_of_spectra() (pid=2904, ip=172.17.0.4)
 File “python/ray/_raylet.pyx”, line 440, in ray._raylet.execute_task
 File “/opt/conda/lib/python3.7/site-packages/ray/memory_monitor.py”, line 132, in raise_if_low_memory
  self.error_threshold))
ray.memory_monitor.RayOutOfMemoryError: More than 95% of the memory on node b351cf893d79 is used (65.98 / 68.54 GB). The top 10 memory consumers are:
PID	MEM	COMMAND
2962	2.08GiB	ray::Worker.run_set_of_spectra()
2961	2.08GiB	ray::Worker.run_set_of_spectra()
2946	2.07GiB	ray::Worker.run_set_of_spectra()
2960	2.07GiB	ray::Worker.run_set_of_spectra()
2955	2.07GiB	ray::Worker.run_set_of_spectra()
2932	2.06GiB	ray::Worker.run_set_of_spectra()
2963	2.06GiB	ray::Worker.run_set_of_spectra()
2928	2.05GiB	ray::Worker.run_set_of_spectra()
2925	2.05GiB	ray::Worker.run_set_of_spectra()
2949	2.03GiB	ray::Worker.run_set_of_spectra()
In addition, up to 0.02 GiB of shared memory is currently being used by the Ray object store.
---
--- Tip: Use the `ray memory` command to list active objects in the cluster.
```

pgbrodrick, 2021-06-22:
Thanks Winston.  There are two components here, the broader issue of perhaps accepting a memory requirement and limiting the number of active cores based on this, and the fact that each core should probably use less memory than this.  I'll focus on the latter here (the former is good, but for down the line).

I'm guessing that this is coming from holding the LUT in memory, as these can get increasingly large, but I'll look into trying to run some benchmarks to confirm, and if so, looking into sharing the LUT across ray actors.  If anyone in the community happens to have memory benchmarks, posts would be welcome!

pgbrodrick, 2021-06-25:
Update:

Thanks to a bit of profiling, there are multiple factors in play here:

- First, and this is just a general note - there are definitely different caches in play, including the hashtable in inverse (regulated by a hardcoded max in common.py), along with the flushrate set in fileio (there's actually a switch for this in the config, but it's currently not used, a global hardcoded value is - I'll include in a patch a switch to use the real value).
- The output file is built to flush, but the input files are actually not.  This has become an issue once we navigated to ray actors and are not rebuilding the io components repeatedly.  A simple flushing of input files resolves this (implemented and tested, will include in the coming PR).
- Even if this is handled, the LUT is still held within each forward model (one per actor), which is a big chunk of reserved memory.  I'm trying out a few options to see if I can't share a forward model between actors (this may not be possible, but I'm giving it a shot).

pgbrodrick, 2021-07-18:
Solved with #240.

closed 2021-07-18 by pgbrodrick