isofit/isofit #251: Ray and aiohttp have changed

quarkpt, 2021-11-06:
In my recent attempts to use package management services to install the dependencies for ISOFIT, I found that at the moment only ray 1.7.1 works with ISOFIT. All earlier versions and the most recent version (1.8.0) do not. 

Versions 1.2.0 through 1.7.0 of ray don't specify a version of aiohttp, so when you go to install these versions of ray it picks up the most recent version of aiohttp. That version is not compatible with ray because ray uses the module aiohttp.signals, which no longer exists in aiohttp as of 3.8. 

Versions 1.7.1 and 1.8.0 of ray specify aiohttp 3.7, so in these cases aiohttp doesn't break ray. However, ray 1.8.0 breaks ISOFIT. ISOFIT relies on the module ray.services, which no longer exists as of ray 1.8.0. 

Thus at the moment only ray 1.7.1 simultaneously doesn't break ISOFIT and is not broken by aiohttp.

pgbrodrick, 2021-11-06:
Thanks for this!

I've built a scratch installation and confirmed....looks like this is a rather recent development, but we'll get on a fix.  The use of ray.services can (and should) be removed now and revised (post ray 1.4) strategies implemented for VPN workarounds.  I'll self-assign, and start with a short fix of removing ray services, and then go for a broader fix of trying to look into versions more broadly (including revisiting the M1 chip, though that may be out of scope).

davidraythompson, 2021-11-22:
Solved by #252

closed 2021-11-22 by davidraythompson