isofit/isofit #257: Crash with masked input data

adamchlus, 2022-01-21:
I recently installed the latest master branch of isofit and when running the ‘apply_oe.py’ script isofit crashes when input images are masked. I have images that include clouds and prior to correction I set cloud pixels to the no data value, the previous version of isofit I was using ran without issue (here is the version: d0df47ff356a2827ebe931ef5a9dbb517536162a). I ran number of images and only the ones that were masked crashed, the rest ran successfully. This is the error I get:
```
  File "/src/isofit//isofit/utils/apply_oe.py", line 1406, in <module>
    main()
  File "/src/isofit//isofit/utils/apply_oe.py", line 351, in main
    retrieval_full.run()
  File "/src/miniconda3/lib/python3.8/site-packages/isofit-2.9.1-py3.8.egg/isofit/core/isofit.py", line 159, in run
    res = list(self.workers.map_unordered(lambda a, b: a.run_set_of_spectra.remote(b),
  File "/src/miniconda3/lib/python3.8/site-packages/ray/util/actor_pool.py", line 93, in map_unordered
    yield self.get_next_unordered()
  File "/src/miniconda3/lib/python3.8/site-packages/ray/util/actor_pool.py", line 216, in get_next_unordered
    return ray.get(future)
  File "/src/miniconda3/lib/python3.8/site-packages/ray/_private/client_mode_hook.py", line 105, in wrapper
    return func(*args, **kwargs)
  File "/src/miniconda3/lib/python3.8/site-packages/ray/worker.py", line 1713, in get
    raise value.as_instanceof_cause()
ray.exceptions.RayTaskError(IndexError): [36mray::Worker.run_set_of_spectra()[39m (pid=14880, ip=172.17.0.2, repr=<isofit.core.isofit.Worker object at 0x7f31071233d0>)
  File "/src/miniconda3/lib/python3.8/site-packages/isofit-2.9.1-py3.8.egg/isofit/core/isofit.py", line 223, in run_set_of_spectra
    states = self.iv.invert(input_data.meas, input_data.geom)
  File "/src/miniconda3/lib/python3.8/site-packages/isofit-2.9.1-py3.8.egg/isofit/inversion/inverse.py", line 298, in invert
    x0 = invert_simple(self.fm, meas, geom)
  File "/src/miniconda3/lib/python3.8/site-packages/isofit-2.9.1-py3.8.egg/isofit/inversion/inverse_simple.py", line 163, in invert_simple
    x[forward.idx_RT] = heuristic_atmosphere(RT, instrument,
  File "/src/miniconda3/lib/python3.8/site-packages/isofit-2.9.1-py3.8.egg/isofit/inversion/inverse_simple.py", line 79, in heuristic_atmosphere
    rhi = RT.get_shared_rtm_quantities(x_RT_2, geom)
  File "/src/miniconda3/lib/python3.8/site-packages/isofit-2.9.1-py3.8.egg/isofit/radiative_transfer/radiative_transfer.py", line 123, in get_shared_rtm_quantities
    ret.append(RT.get(x_RT, geom))
  File "/src/miniconda3/lib/python3.8/site-packages/isofit-2.9.1-py3.8.egg/isofit/radiative_transfer/sRTMnet.py", line 234, in get
    return self._lookup_lut(point)
  File "/src/miniconda3/lib/python3.8/site-packages/isofit-2.9.1-py3.8.egg/isofit/radiative_transfer/sRTMnet.py", line 199, in _lookup_lut
    ret[key] = np.array(lut(point)).ravel()
  File "/src/miniconda3/lib/python3.8/site-packages/isofit-2.9.1-py3.8.egg/isofit/core/common.py", line 138, in __call__
    if self.lut_interp_types[i] == 'n':
```
 
These runs are done using sRTMnet and the empirical line interpolation. Reverting back to the previous version isofit I was using resolved this errors.
 
As a side note I ran into a number of ray related issues after installing the current master of isofit, on multiple machines I kept getting core crashes and initialization errors. Getting it to work involved changing how ray was initialized in isofit and finding a ray version that worked (1.9.2 worked but still ran into the error above).

pgbrodrick, 2022-01-21:
Hmm, the middle of that if statement is an odd place to get a failure - just to double check, there's nothing further down in this error statement?  

Assuming there's nothing further, could you do me a favor and see if it runs successfully here: ea48b3fb36f30dd428a3eb5a10f90b27da471990 ?   Trying to isolate potential causes.  

Also how big is the scene you're running?

evan-greenbrg, 2026-09-08:
Closing for lack of traffic. @adamchlus, please reopen if this is still a persistent error.

closed 2026-09-08 by evan-greenbrg