isofit/isofit #265: template file

jenniferbuz, 2022-03-10:
Looking for some clarification- when running the forward model, will sRTMnet.py only load values from the template file? The geometry and days are not correct from our input.

line 86 : # Populate the sixs-values from the modtran template file
        with open(sixs_config.template_file, 'r') as infile:
            modtran_input = yaml.safe_load(infile)['MODTRAN'][0]['MODTRANINPUT']

pgbrodrick, 2022-03-14:
Hi, thanks for the post.

sRTMnet (and the other RTMs for that matter) will all look to read in from the template file, and then overwrite particular elements during the generation of the LUT.  Values are are overwritten are specified under lut_grid in the radiative_transfer section of the config file (see [here](https://github.com/isofit/isofit/blob/master/examples/20171108_Pasadena/configs/ang20171108t184227_beckmanlawn.json) for an example).

To help with your specific issue....where are you seeing that the geometry and days are not correct?  Also, are you using apply_oe to do your run?  If so, are you using one of the pre-specified sensors or are you using the NA- option?

kevinbrodie, 2022-03-15:
This is for calling the forward model only. It appears to just read in the template file before writing anything from the configs.

pgbrodrick, 2022-03-19:
Would you be willing to provide an example configuration file and particular function you're executing?  Dummy or incorrect pathnames are fine, I can reconstruct from there.

From the above, I take it that you are not in fact using apply_oe, but to the first question - where are you seeing that the geometry and days are not correct?

jenniferbuz, 2022-04-11:
It was a paths issue so this is resolved.  thank you!

closed 2022-04-11 by jenniferbuz