isofit/isofit #267: Hypertrace defaults

araiho, 2022-04-12:
@davidraythompson and I were discussing some of the configuration defaults in Hypertrace today including:

- H2OSTR lut_grid: should be finer scale (e.g., ```"H2OSTR": [0.9, 1.0, 1.1, 1.2, 1.3, 1.4, 1.5, 1.6, 1.7, 1.8, 1.9]```) and bounds should match max/min of grid.
- instrument model configuration may need to include unkowns: ```    "instrument": {
      "wavelength_file": "/path/to/wavelength.txt",
      "SNR":800,
      "integrations":1,
      "unknowns": {
        "channelized_radiometric_uncertainty_file": "path/to/isofit/data/avirisng_systematic_error.txt",
        "uncorrelated_radiometric_uncertainty": 0.01
   }```
- the modtran_template.json file defaults are fixed including lat/long, GMTIME, H1ALT, IDAY, etc. The hypertrace user could change these in the hypertrace config but we might want to highlight that they might affect the retrieval if they are not changed. 

A larger question might be and that I think we've discussed before: do we want to switch the hypertrace workflow to match up better with the isofit updates? Configuration best practices may change over time and we may not want to just keep updating the config files / we may forget to update them as things change. What's the easiest way to do that? I know there is `apply_oe.py` Should hypertrace migrate to depending more directly on that?

davidraythompson, 2022-04-12:
Thanks @araiho ! I should clarify that I don't know that the grid points listed above are the ideal solution, just that the current ones are unrealistically coarse.   The above may be okay if hypertrace is managing both the forward and inverse models and we can control the range of simulated "true" H2O values.  

I'm not sure what the best solution is for keeping hypertrace in sync with ISOFIT updates, but I agree we should think about this...

pgbrodrick, 2022-04-13:
@araiho, great points and questions.

To the larger question of config interfaces, this is a tricky one.  Our approach so far has been to leverage the defaults as much as possible, but any config update is going to trickle through apply_oe, as well as all of the examples (including all of the hypertrace ones).

I try and review each PR to make sure that config changes trickle all of the way through, but your point about best practices is well made (and much harder to ensure), and I don't have a good solution, other than for us to collectively try and keep things maintained.  I don't think apply_oe is particularly well adapted for the hypertrace-style runs as a whole (just because it has a bunch of file dependencies that hypertrace may not want to also require), but we could consider utilizing some of the config writing functions that are there to try and at least keep things consistent.

What do you think?  We're all ears for suggestions and opinions!

araiho, 2022-04-14:
@pgbrodrick and @davidraythompson thanks for the feedback!

I was first thinking that some sort of master set of configs should be all that is maintained and drawn from for various example runs. But, as I'm learning more about spectroscopy, it seems to me that each scenario can have its own nuances. When either of you looks at a spectra, it seems you know what might be the issue. So, I'm wondering if an optional set of tests at the end of an atmospheric correction could help? Are there tests that you run now that I might not know about?

pgbrodrick, 2022-04-15:
I think you're right on regarding the application specificity and the nuances that arise in different instances.  To the very first order, we've attempted to populate the default parameter options directly in the code (see isofit/configs/sections) - both so that they show up in the documentation, and so that you don't have to re-specify within each config file.  So the ones that make it through into the configuration file really are those highly dynamic ones.  Whether apply_oe and hypertrace should try and standardize is still a good question, but I think I'm with you that as their goals are different, having different heuristics is okay.

Your idea for a post-hoc diagnostic is **excellent**, though not easy.  Some things are easy to catch - water vapor residuals or other atmospheric effects can be obvious candidates for checks.  Other effects are harder to diagnose, and rely a lot on context, but also generally can't be absolutely determined without a time series.  We don't have any attempts to handle the former (atmospheric residual) set built in yet, but that would be an excellent addition to the code.  I imagine we'll look into something for EMIT along these lines eventually, but if you (or others) are keen, I think everyone would welcome a contribution in this area.

We do have plans to develop a much more extensive suite of tests over the summer and next fall, and any diagnostic work could synergize nicely with that effort.

davidraythompson, 2022-04-15:
A post hoc diagnostic is a great idea.  I propose we develop an approach based on off-the-shelf probability density estimation and write a paper.  We could use this for EMIT!